FreeCAD/FreeCAD #15996: SVG import to Sketch broken

boboxx, 2024-08-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I had a Inkskape SVG that I
[Logo.zip](https://github.com/user-attachments/files/16700940/Logo.zip)
'm trying to import in FreeCAD. I'm able to import and convert to a sketch via the draft 2 sketch. The shape gets converted and I can extrude the sketch, but if i try to open the sketch to make modification, the sketch is empty.

I'm a few days old build as the latest few build release have been broken so I can't see if this has been fixed or not.

I have attached the FreeCAD file and SVG

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38334 (Git)
Build type: Release
Branch: main
Hash: 61c964caa9a7e5bbc7a8d5f7d8914505d365113e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenTheme 2024.7.3
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-22:
I can't confirm on Windows:

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

luzpaz, 2024-08-22:
@JohnOCFII would you mind testing on macOS ?

Roy-043, 2024-08-22:
I can confirm using the same version as @FEA-eng on Windows, with a reset configuration. Which is interesting...

I have create the sketch by selecting "path1", "path002" and "path003" and then pressing the Draft_Draft2Sketch button.

@FEA-eng can your attach your file?

My sketch has the same number of DoFs, the same number of constraints, and the same number of elements (all B-splines) as the sketch in the provided file.

Some B-splines are very short:
0.0005199999999945248
0.020498524505898126
0.021727681246914786
7.071067813507013e-05 (this one may be to blame?)

```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

JohnOCFII, 2024-08-22:
> @JohnOCFII would you mind testing on macOS ?

Yes, I see the same behavior as the OP using the last working macOS weekly that I have: 
```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.8.17
```

FEA-eng, 2024-08-24:
> @FEA-eng can your attach your file?

Sure. It's not with a clean config though.

[Logo.FCStd.zip](https://github.com/user-attachments/files/16736775/Logo.FCStd.zip)

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

boboxx, 2024-08-24:
Interesting, I have the same issue with your file. So it looks like the issue is on the rendering side and not the generation.

FEA-eng, 2025-10-21:
Still cannot reproduce on Windows, even in safe mode.

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/14 15:14:51
Build type: Release
Branch: main
Hash: 41ce37233d64c81c2f105de085635d4dcd669943
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

JohnOCFII, 2025-10-21:
Well -- I don't get a crash. I'm not sure I'm following the exact same steps as the OP, as I didn't see exact steps.  But -- I select the three items (path1, path002, path003) and run the draft2sketch process.  The sketch is created.  By default, it is way under-constrained.  But no crash, and the sketch is editable for me.

[image]

```
OS: macOS 15.7.1
Architecture: arm64
Version: 1.1.0dev.44027 (Git)
Build date: 2025/10/21 12:07:13
Build type: Unknown
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.0, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * Telemetry 1.0.2
```

FEA-eng, 2026-10-25:
[This](https://github.com/FreeCAD/FreeCAD/issues/15996#issuecomment-3427056522) is still relevant

```
OS: Windows 11 build 26300
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
OpenGL version/vendor/renderer: 4.6.0 - Build 27.20.100.9664/Intel/Intel(R) UHD Graphics 620
```

luzpaz, 2026-10-25:
This needs a diagnosis and a proposed solution