FreeCAD/FreeCAD #15997: Duplicated Import options shown

krushia, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Recently when importing file using `File > Import...`, a dialog started showing with duplicate import methods to choose from. I'm not sure what causes it. It happens with STEP. IGES, and BREP. These also show twice in the file picker dialog.

[image]
[image]
[image]

### Full version info

```shell
OS: Gentoo Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.Unknown
Build type: RelWithDebInfo
Python 3.11.9, Qt 5.15.14, Coin 4.0.2, Vtk 9.2.5, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * parts_library
  * Curves 0.6.42
  * A2plus 0.4.65
  * DynamicData 2.62.0
  * Assembly4 0.50.15
  * Defeaturing 1.2.2
  * dodo 1.0.1 (Disabled)
  * boltsfc 2022.11.5
  * Alternate_OpenSCAD 1.0.0 (Disabled)
  * 3DfindIT 1.2.0
  * CurvedShapes 1.0.9
  * Design456 0.0.1
  * fasteners 0.5.25
  * SearchBar 1.0.2
  * DesignSPHysics 0.7.0(15-09-2023)
  * cadquery_module
  * ose-piping (Disabled)
  * Nodes 0.1.36
  * 3D_Printing_Tools
  * Pyramids-and-Polyhedrons
  * workfeature
  * MnesarcoUtils 0.2.5
  * Manipulator 1.5.7
  * Assembly3 0.12.2
  * lattice2 1.0.0
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * CommandPanel
  * CubeMenu
  * MeshRemodel 1.9.23
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-22:
> Version: 0.22.0dev.Unknown
> Build type: RelWithDebInfo

Is it your own build? Compiled when?

krushia, 2024-08-22:
In the last 2 weeks.

FEA-eng, 2024-08-22:
> I suppose this means I have a config issue.

Then can you try clearing it (deleting the configuration files - you can make a backup first, of course)?

luzpaz, 2024-08-23:
> Would it be possible (and reasonable) to specify what workbench **_defined_** an import/export filter in the disambiguation dialog? This seems like a case of the `Design456` WB masquerading as `Import` and `Part`.

We keep bumping into these name clash issue with 3rd party addons. CC @chennes 
Not sure there is a ticket for this specific issue yet.

luzpaz, 2024-08-23:
Downstream bug has been fixed. Closing this ticket. Thanks @Krushia for reporting and diagnosing :+1:

closed 2024-08-23 by luzpaz