FreeCAD/FreeCAD #15998: constrain symmetric shows error "Recompute failed!"

NathanLKing, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I click two points on a rectangle, click the line, click Constrain Symmentric and it shows the error "Recompute failed!"

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
[/code]
```

### Subproject(s) affected?

Sketcher

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

NathanLKing, 2024-08-22:
> Can you try in 0.22 dev? Do you have "auto remove redundants" enabled? Can you share the file?

Howdy. Thanks for the quick reply.
I ticked on "auto remove redundants" and it worked. 
This means that I had incorrectly put another constraint that made this somehow fail?

FEA-eng, 2024-08-22:
> This means that I had incorrectly put another constraint that made this somehow fail?

Many power users would likely disagree with me but IMO symmetric constraint is almost unusable (at least in a way you would use it in other CAD software) without "auto remove redundants". Even with this option enabled, it's often causing overconstraints (I'd rather expect those than "recompute failed" in most cases).

NathanLKing, 2024-08-22:
> Many power users would likely disagree with me but IMO symmetric constraint is almost unusable (at least in a way you would use it in other CAD software) without "auto remove redundants". Even with [...]

I agree, it makes sense. Especially because there's not icons on the sketches showing the applied constraints - e.g. like Fusion

maxwxyz, 2024-08-24:
I'll close this issue then.

closed 2024-08-24 by maxwxyz