FreeCAD/FreeCAD #15999: there are a issue with curretn version of freecad with boost

Tarot67, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

/compile/freecad/src/Mod/Part/App/Geometry.cpp: In member function ‘void Part::Geometry::createNewTag()’:
/compile/freecad/src/Mod/Part/App/Geometry.cpp:462:19: error: ‘mt19937’ in namespace ‘boost’ does not name a type
  462 |     static boost::mt19937 ran;
      |                   ^~~~~~~
/compile/freecad/src/Mod/Part/App/Geometry.cpp:469:9: error: ‘ran’ was not declared in this scope; did you mean ‘tan’?
  469 |         ran.seed(static_cast<unsigned int>(std::time(nullptr)));
      |         ^~~
      |         tan
/compile/freecad/src/Mod/Part/App/Geometry.cpp:472:56: error: ‘mt19937’ is not a member of ‘boost’; did you mean ‘std::mt19937’?
  472 |     static boost::uuids::basic_random_generator<boost::mt19937> gen(&ran);
      |                                                        ^~~~~~~
In file included from /usr/include/c++/14.2.0/random:48,
                 from /usr/local/include/opencascade/NCollection_UBTreeFiller.hxx:22,
                 from /usr/local/include/opencascade/Extrema_HUBTreeOfSphere.hxx:19,
                 from /usr/local/include/opencascade/Extrema_GenExtPS.hxx:23,
                 from /usr/local/include/opencascade/Extrema_ExtPS.hxx:25,
                 from /usr/local/include/opencascade/GeomPlate_BuildPlateSurface.hxx:30,
                 from /compile/freecad/src/Mod/Part/App/Geometry.h:49,
                 from /compile/freecad/src/Mod/Part/App/Geometry.cpp:119:
/usr/include/c++/14.2.0/bits/random.h:1717:37: note: ‘std::mt19937’ declared here
 1717 |     0xefc60000UL, 18, 1812433253UL> mt19937;
      |                                     ^~~~~~~
/compile/freecad/src/Mod/Part/App/Geometry.cpp:472:63: error: template argument 1 is invalid
  472 |     static boost::uuids::basic_random_generator<boost::mt19937> gen(&ran);
      |                                                               ^
/compile/freecad/src/Mod/Part/App/Geometry.cpp:472:70: error: ‘ran’ was not declared in this scope; did you mean ‘tan’?
  472 |     static boost::uuids::basic_random_generator<boost::mt19937> gen(&ran);
      |                                                                      ^~~
      |                                                                      tan
/compile/freecad/src/Mod/Part/App/Geometry.cpp:474:14: error: ‘gen’ cannot be used as a function
  474 |     tag = gen();
      |           ~~~^~
make[2]: *** [src/Mod/Part/App/CMakeFiles/Part.dir/build.make:3185: src/Mod/Part/App/CMakeFiles/Part.dir/Geometry.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:7140: src/Mod/Part/App/CMakeFiles/Part.dir/all] Error 2
make: *** [Makefile:146: all] Error 2

### Full version info

```shell
I fixed the problem. I just apply the proposal of gcc ;)

i just replace boost::mt19937 by std::mt19937 and freecad build correctly.

Regards,

Jean-Sebastien Tarot
```

### Subproject(s) affected?

Part

### Anything else?

This the patch! it is a diff

462c462
<     static std::mt19937 ran;
---
>     static boost::mt19937 ran;
472c472
<     static boost::uuids::basic_random_generator<std::mt19937> gen(&ran);
---
>     static boost::uuids::basic_random_generator<boost::mt19937> gen(&ran);

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Tarot67, 2024-08-22:
i forget line 65 and 75 of this file has same issue
freecad/src/Mod/Sketcher/App/Constraint.cpp 

I have just replace boost::mt19937 by std::mt19937

This file has also this problem line 1403 and 1412
freecad/src/Mod/TechDraw/App/Geometry.cpp

And this file line 242, 254
freecad/src/Mod/TechDraw/App/CosmeticVertex.cpp

And this file line 319,329,495,505
freecad/src/Mod/TechDraw/App/Cosmetic.cpp

And this file line 1079, 1089
freecad/src/Mod/TechDraw/App/CenterLine.cpp

**I m not a expert i just listen gcc ;)**

PS:I have boost-1.86.0 and i found this on other project:
https://github.com/PointCloudLibrary/pcl/issues/2803

Regards,

Jean-Sebastien Tarot

Tarot67, 2024-08-22:
Now freecad build and run.

Tarot67, 2024-08-22:
https://www.boost.org/doc/libs/1_79_0/doc/html/boost_random.html

I found this interesting documentation!

luzpaz, 2024-08-22:
Thanks for reporting @Tarot67
Do you know how to make a Pull Request ?

Tarot67, 2024-08-22:
You have a documentation for make a pull request ?

Because i use just my own repositories and i never do a pull request.

generally i use git init, git add <myfile>,  git commit -v (add comments)
and git push.

sometime i use git switch to switch branch and that's it.

But if you have a documentation. Yes i can do

luzpaz, 2024-08-22:
@Tarot67 Thanks. Here's documentation on contributing via Pull Request:  
https://wiki.freecad.org/Source_code_management#Remote_repositories

luzpaz, 2024-08-22:
@wwmayer could you weigh-in and see if this is the correct solution ?

Tarot67, 2024-08-22:
Indeed because there are several solution.

keep boost a make a pragma like
#if BOOST_VERSION > 1.xx
boost::random::mt19937
#else
boost::mt19937

or use standard library

std::mt19937

mt19937 it is a 32 random fonction. But i don't know if std::mt19937 and
boost::random::mt19937 are really compatible ?

wwmayer, 2024-08-22:
As far as I can see boost::mt19937 is still part of the boost library `(https://github.com/boostorg/random/blob/29ac88879985418c6284587485d0ad8c36e33cbc/include/boost/random/mersenne_twister.hpp)`

I wonder whether a needed header file is missing. Does it solve the build failure if you include `#include <boost/random.hpp>`?

Tarot67, 2024-08-22:
OK i try this if it resolve issue

Tarot67, 2024-08-22:
Ok @wwmayer you have right ;) the include "#include <boost/random.hpp>" is missing in this files:

- freecad/src/Mod/Part/App/Geometry.cpp
- freecad/src/Mod/Sketcher/App/Constraint.cpp
- freecad/src/Mod/TechDraw/App/Geometry.cpp
- freecad/src/Mod/TechDraw/App/CosmeticVertex.cpp
- freecad/src/Mod/TechDraw/App/Cosmetic.cpp
- freecad/src/Mod/TechDraw/App/CenterLine.cpp

I add missing include and now freecad build correctly without issue.

luzpaz, 2024-08-22:
@Tarot67 would you you still like to try learning how to make a PR so you can easily contribute in the future :wink:

Tarot67, 2024-08-22:
Yes i am interesting. But i am very a newbee!. But if do step by step
normally it is ok. You have a test repository for no breaking something ?
for the first pull request ?

luzpaz, 2024-08-22:
@Tarot67 looks like wwmayer made the PR already :point_up_2:

PR #16004 (2024-08-22): Building: Boost dependency errors
Fixes #15999
commit c9a586f3 (2024-08-22): Building: Boost dependency errors

Fixes #15999
closed 2024-08-22 by chennes