FreeCAD/FreeCAD #16000: DXF import stopped working altogether in the last two dev versions

dominik-devops, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I'm going through Yorik's Barcelona tutorial on youtube where he provides a starting DXF file. While the (C++) import didn't work correctly before, it stopped working after tonight update. Please see the screenshot to see how it looks. 
Two versions back the import imported the objects as in issue #13597.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38500 (Git) Snap 1112
Build type: Release
Branch: main
Hash: a75003449050e5d5d10398a522c6a3648eb097a2
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/FreeCAD/FreeCAD/issues/13597
https://github.com
[pabellon de barcelona.zip](https://github.com/user-attachments/files/16709013/pabellon.de.barcelona.zip)
/FreeCAD/FreeCAD/issues/13621
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-08-22:
I don't have access to 0.22.0dev.38500 or 0.22.0dev.38499. But I would not expect the code to reach line 2857 when run from the GUI.

https://github.com/FreeCAD/FreeCAD/blob/a75003449050e5d5d10398a522c6a3648eb097a2/src/Mod/Draft/importDXF.py#L2851-L2859

@dominik-devops 
Can you paste this in the Python console:
`import ImportGui`

dominik-devops, 2024-08-22:
> @dominik-devops Can you paste this in the Python console?: `import ImportGui`

I did and nothing happened, I've tried to import a DFX and no changes.

paullee0, 2024-08-23:
@dominik-devops  It seems the dxf file itself maybe somhow corrupted or not fit for the importer that it crashes it (in fact, the importer should be designed to prevent this if not so).

Could you test importing in other application, e.g. libreCad etc.

paullee0, 2024-08-23:
@Roy-043 Better to put the import action into separate Try: to prevent this module from breaking at this point, agreeable ?

dominik-devops, 2024-08-23:
@paullee0 In LibreCAD, it opens alright, actually that is the first application I use to process the files. And the exact files worked fine before the updates. 
I downloaded the file new from Yorik's dropbox and it doesn't work the same way. There were no changes on my end other than the updates of FreeCAD.

Roy-043, 2024-08-25:
Can confirm there is an issue with the version listed below. FC seems to hang.
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

dominik-devops, 2024-08-26:
I confirm that the import works again. Thank you! Could you @paullee0 provide reference for the change that was made, I couldn't find any related PR?

OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38516 (Git) Snap 1116
Build type: Release
Branch: main
Hash: 46a84da571d51b135e16ac4731baaf67c552d791
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)

paullee0, 2024-08-26:
Seems Roy_043 found it was not working and not sure if anyone had submiteed a PR :)

dominik-devops, 2024-08-26:
Thanks @paullee0! @Roy-043, I'd be interested to see the code for the change if it is possible, please. Also, who and when should close the issue?

closed 2024-08-26 by luzpaz