FreeCAD/FreeCAD #16001: Cant run Freecad from last nvidia driver update

deddroid, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

this is the log from terminal:

❯ flatpak run org.freecadweb.FreeCAD
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

(qt.glx) qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
(qt.glx) qglx_findConfig: Failed to finding matching FBConfig for QSurfaceFormat(version 2.0, options QFlags<QSurfaceFormat::FormatOption>(), depthBufferSize -1, redBufferSize 1, greenBufferSize 1, blueBufferSize 1, alphaBufferSize -1, stencilBufferSize -1, samples -1, swapBehavior QSurfaceFormat::SingleBuffer, swapInterval 1, colorSpace QSurfaceFormat::DefaultColorSpace, profile  QSurfaceFormat::NoProfile)
Could not initialize GLX

~

### Full version info

```shell
Latest

in on cachyos, kde wayland
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-22:
@deddroid Please try this solution: https://forum.manjaro.org/t/some-flatpak-qt-apps-fail-to-start-after-latest-update/99973/2

luzpaz, 2024-08-22:
I know some distros (Manjaro) have downgraded their nividia drivers, maybe for this reason ?

luzpaz, 2024-08-22:
@deddroid Wait, you tried the flatpak beta branch, u need to try the development branch

deddroid, 2024-08-23:
> @deddroid Wait, you tried the flatpak beta branch, u need to try the development branch

I have tryed the commands from this documentation:   https://wiki.freecad.org/Flatpak#Development_versions

1- flatpak remote-add --if-not-exists flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo
flatpak install flathub-beta org.freecadweb.FreeCAD

2- flatpak run --branch=beta org.freecadweb.FreeCAD

isn't right? I have to run in other way?

luzpaz, 2024-08-23:
@deddroid wow.. my mistake. I didn't realize the flatpak didnt have a nightly or weekly release of our development builds. Just 0.21.x builds

Quick workarounds: 
* Downgrade your nividia drivers

OR 

* Consider using the development builds: [AppImage](https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds), [Snap](https://github.com/FreeCAD/FreeCAD-snap) or [building from source](https://wiki.freecad.org/Compile_on_Linux).

deddroid, 2024-08-23:
> @deddroid wow.. my mistake. I didn't realize the flatpak didnt have a nightly or weekly release of our development builds. Just 0.21.x builds
>
> Quick workarounds:
>
> * Downgrade your nividia drivers [...]

Thanks for your time @luzpaz. I tryed the Appimage, but it won't start. Like evry time nvidia create a bugged update :smile: 

I'll try to downgrade or wait for a new fixed release. Thanks again!

luzpaz, 2024-08-23:
> Like evry time nvidia create a bugged update 😄

Yea, that seems right. At least from what I've seen in Manjaro Linux

alezura, 2024-11-01:
same happens on fedora NVIDIA drivers AppImage