FreeCAD/FreeCAD #16003: Addon Manager dumps updated workbench files on Windows desktop

FEA-eng, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I've noticed that recently, when installing add-on workbench updates, the Addon Manager dumps workbench files (at least for CfdOF and SheetMetal since those are frequently updated) on the Windows desktop for some reason (even though the workbenches are installed in the default location). Those are .FCMacro files to be more specific and there's quite a lot of them, especially for CfdOF. Can it be due to #15953?

I have 0.21.2 installed on the same machine but can it be a problem here?

What's even worse, it seems that the workbenches aren't really updated this way because restarting FreeCAD shows their available updates again. I've tried refreshing Addon Manager cache too.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-08-22:
@FEA-eng CfdOF and Sheetmetal Wb stand out for me as they are members of a small club, Addons that have Unit Tests.

chennes, 2024-08-22:
The addon manager has a somewhat odd feature in that it is supposed to copy files with the FCMacro extension into your macros folder. I don't see how my recent update (which simply deletes your existing installation when using the ZIP installation method) could affect that. I'll take a look at those two addons and see if I can reproduce this issue locally. Are you using ZIP or git installation method?

FEA-eng, 2024-08-24:
> Are you using ZIP or git installation method?

I'm not sure. I haven't changed any default settings for Addon Manager but I also haven't installed any special libraries for it. I'm using Windows.

Syres916, 2024-08-24:
@chennes possibly some new security setting on Windows redirecting the files to the Desktop? I've just tested both Git install of CfdOF and Zip install of Sheetmetal on Windows 7 and no issues, the macros all deposited in the correct Macro location.

```
OS: Windows 7 build 7601
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38507 (Git)
Build type: Release
Branch: main
Hash: baec1037d8ce9c02f6dc67685926fce9dbaf42cf
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CfdOF 1.27.5
  * Curves 0.6.40
  * Dracula 0.0.5
  * FreeCAD-themes 2024.7.24
  * sheetmetal 0.4.24
```

chennes, 2024-08-24:
I can replicate the problem with a ZIP install on Windows 11 -- the git works fine, which is sort of interesting, since the FCMacro-copying code should be the same between the two.

chennes, 2024-08-24:
OK, it looks like this isn't really a bug in the Addon Manager, at least not directly -- MacroPath (in preferences group "User parameter:BaseApp/Preferences/Macro") is getting set to the Desktop someplace. There may still be a bug in AM (that the git installation is *not* doing this is actually a problem) but we need to expand our search for the culprit to the MacroPath getting set to the desktop.

chennes, 2024-08-24:
After the single replication, I can't repeat it -- I've deleted by ~/AppData/Roaming/FreeCAD and restarted a number of times, playing with launching Macro preferences, deleting the MacroPath key, installing Addons via both git and Zip, and everything is working fine. I've also tried applying our three built-in preference packs on the chance that it is encoded into one of them, but nothing I did repeated the bad setting of that key.

FEA-eng, 2024-08-26:
@chennes After checking, I can confirm that I have the macros location set to Desktop but I just wanted it for macros that I recorded in FreeCAD, I didn't expect it to affect the Addon Manager and external workbenches as well. Also, I had this setting for a while and nothing happened until the recent change so I'm not sure if that's the culprit.

[image]

chennes, 2024-08-26:
FreeCAD is not very good with multiple macro locations: you basically only get one. So if an Addon needs macros for something, those have to go in the same path as your other macros.

FEA-eng, 2024-08-26:
I see, so it doesn't seem to be a bug but maybe it could be turned into a feature request for the future because it can be confusing that changing the macro recording directory affects some add-on workbenches.

chennes, 2024-08-31:
@maxwxyz this might be a duplicate now that the problem is narrowed down to FreeCAD's handling of macro directories -- I don't see an obvious candidate, but I'm sure we've talked about this.