FreeCAD/FreeCAD #16005: [CAM Workbench] Sub-objects not hidden when Job hidden in "Model" outliner

alexfacciorusso, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When hiding a Job created with the CAM Workbench, its sub-objects remain shown.

### Full version info

```shell
OS: Windows 11 build 22635
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
```

### Subproject(s) affected?

CAM

### Anything else?

**Screenshots**

https://github.com/user-attachments/assets/669197ad-5cc5-4883-97d3-fe0e359aa2ac

**Sample project**

[hide_job_example.zip](https://github.com/user-attachments/files/16713717/hide_job_example.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

alexfacciorusso, 2024-08-22:
The only way to hide all the operation paths is to hide them one by one (**extremely** bothersome for jobs with many ops + many jobs

maxwxyz, 2024-08-23:
@alexfacciorusso is it the same behavior in the stable version?

alexfacciorusso, 2024-08-23:
@maxwxyz I haven't tried the stable version in a while, to be honest. Nevertheless, I would strongly suggest this to be fixed by 1.0 as it makes the CAM workbench feel unpolished and almost unusable for big jobs, because you can imagine the vast amount of operation a complex piece may take, and having to manually hide each and every of them, even if possible by multi-selecting, is very tedious.

In addition, the fact the body clone is also not hidden, makes it very hard to understand what's going on to the new user coming from other software

luzpaz, 2024-08-23:
@alexfacciorusso we ask because we try to ascertain if a bug is a regression.

fanfragma, 2024-08-23:
Can confirm the same behavior in 21.2 and also 22.0

luzpaz, 2024-08-24:
@fanfragma can you please paste the [About](https://wiki.freecad.org/About) information of each version you've tested on? TIA

fanfragma, 2024-08-24:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38314 (Git)
Build type: Release
Branch: main
Hash: 3fa339e9eb71d34542e463b3f4581857a9e696fb
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Belgium (nl_BE)
Installed mods: 
  * parts_library

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Dutch/Belgium (nl_BE)
Installed mods: 
  * parts_library

luzpaz, 2024-08-24:
@fanfragma your dev version isn't up to date (we're at r38507) can you retest with the latest dev and report back ?

fanfragma, 2024-08-24:
So i download:
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Belgium (nl_BE)
Installed mods: 
  * parts_library

1. Created a new file
2. made a part
3. created a job
4. added adaptive operation
5. then toggled the job and only the stock switched visibillity
[image]

luzpaz, 2024-08-24:
CC @sliptonic

tarman3, 2025-02-06:
Workaround by macro
https://forum.freecad.org/viewtopic.php?p=809400#p809400