FreeCAD/FreeCAD #16006: [Sketcher] Overconstrain dimensions should be automatically reference

vernzimm, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have a simple proposal for Sketcher when adding dimensions... if the dimension to be placed will cause the sketch to be over constrained, make that dimension reference automatically.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Behave-Dark-Colors 0.1.1
  * ProDarkThemePreferencePack 1.0.0
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-08-22:
I would rather keep both dimensions normal and make the relevant geometry overconstrained, but your workflow can be valid. An option could be provided in the settings. 

Currently there is an inconsistency with sketch constrains: redundant dimensions are always removed automatically, whether or not the option to do so for constrains is active or not.

vernzimm, 2024-08-22:
Yes, I get what you are saying. Here is why I often run into the issue... I use sketcher a lot with existing models to pull geometry and dimensional information about a part that I'm working on. I might fully dimension a line to create an intersection, then want to know how far that intersection is from something else for instance. The dimension I want to know about is fully constrained, but I need the information about it as a reference.

I would be fine having it as an option, but also it would seem to be a solution to the inconsistency with the behavior to have it automatically flip to reference, don't you think?

kaiwas, 2024-08-22:
Both options would be a great solution instead of the current one.
I really hope that the discussion in this thread will be noticed and implemented someday.
Losing the "entered data" is just awful!

pierreporte, 2024-08-23:
@maxwxyz I wanted to open an issue similar to this one to request that redundant dimensions shouldn’t be deleted automatically, and I feel that it is way too similar. What about renaming this one so that we can discuss the two following questions:

1. Should redundant dimensions be kept and not deleted, in compliance with geometric constrains?
2. What to do with redundant dimensions?

maxwxyz, 2024-08-23:
@pierreporte 
1. there is a setting for that in the Sketcher preferences
2. redundant are automatically added as driven dimensions IIRC

pierreporte, 2024-08-23:
The settings only works for geometric constrains. This very issue, that I can reproduce, show that redundant dimensions are not driven by default.

My version:

```
OS: Ubuntu 24.04 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38467 (Git) AppImage
Build type: Release
Branch: main
Hash: 87e73967da5734e100d5ac4b2fd5af1648a02572
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * CosmeticThread3D
  * Curves 0.6.39
  * CurvedShapes 1.0.9
  * sheetmetal 0.4.19
```

vernzimm, 2024-08-23:
@maxwxyz While I agree that redundant geometric constrains can be deleted automatically, I am hoping that for redundant dimensions we simply bump them into reference but not delete them.

@pierreporte 
> This very issue, that I can reproduce, show that redundant dimensions are not driven by default.

I'm a bit confused by this... the problem is that adding redundant dimensions _are_ driven by default, and then potentially deleted. Did you mean to say that they are not deleted by default regardless of the Sketcher preference setting?

@kaiwas
I would pose the trim to construction geometry as a separate issue. It is an interesting concept, but outside the scope of this issue.

pierreporte, 2024-08-23:
@vernzimm Please provide a screenshot of what you get. I understand that your redundant dimensions turn blue, and that the option in settings makes it disappear automatically or stay in the sketch depending on the state of the sketchbox. In my case, dimensions are always deleted and never turn blue.

kaiwas, 2024-08-23:
> I would pose the trim to construction geometry as a separate issue. It is an interesting concept, but outside the scope of this issue.

Okay. I'll make some tickets next week.
Plans:
"non-destructive trim".
"Dividing a segment into equal segments."
"Array from point to point"

pierreporte, 2024-08-26:
@maxwxyz This issue looks like a duplicate #14150.

maxwxyz, 2024-08-27:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/14150

closed 2024-08-27 by maxwxyz