FreeCAD/FreeCAD #16011: Displacement mesh is not generated in FEM workbench after click on FEM_FemMesh2Mesh button

ss4workonly, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Open FEMExample
2. Click CCX_Results_Mesh in a tree view
3. Click a FEM_FemMesh2Mesh button
4. See Mesh created

The issue is that a normal mesh is created instead of displacement one as described [here](https://wiki.freecad.org/FEM_FemMesh2Mesh).

 It looks like the root cause of the issue is in this [line](https://github.com/FreeCAD/FreeCAD/blob/40d23c26390ddb50b03fa597242135ce9de6edab/src/Mod/Fem/femmesh/femmesh2mesh.py#L171) as there is no `Name` property in myResults.

`        FreeCAD.Console.PrintMessage(f"{myResults.Name}\n")
`

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git) AppImage
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
```

### Subproject(s) affected?

FEM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-22:
CC @FEA-eng

FEA-eng, 2024-08-22:
Deformed mesh can be saved only using Python (example script is included on the wiki).

ss4workonly, 2024-08-23:
If this is by design than the issue can be closed. Thank you.

FEA-eng, 2024-08-23:
Yes, but there is a PR that will change it: #15858. Hopefully soon.

FEA-eng, 2024-08-26:
@maxwxyz Can you change it to feature request or just close as a duplicate of #11301?

maxwxyz, 2024-08-26:
Duplicate of #11301

closed 2024-08-26 by maxwxyz