FreeCAD/FreeCAD #16012: Expression source value change incorrectly errors 'Invalid constraint index'

samohd, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The following is in reference to the attached image, and the project is also attached. I have two constraints defined with Expressions, referencing a line. If I double click the 3.5mm in the drawing, which pops-up the requester to change this numeric value, it will error with 'Invalid constraint index'. Though it is the correct index. If however, I access that constraint via Model > Sketch > Constraint# to change it, this works. Oddly, after I change it the alternate way, and then go back and double-click to change it, it will work once... the next change fails, and I have to go back to the alternate way. The attached document has 1 DoF, cuz I can't figure out how to constrain it the way I want, but the issue exists irrelevant of this.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38495 (Git)
Build type: Release
Branch: main
Hash: 131956e201dc5033de69ed09c28db332feb081c1
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

[ExpressionSourceBug.zip](https://github.com/user-attachments/files/16719244/ExpressionSourceBug.zip)
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

samohd, 2024-08-22:
I made a modification to my initial project that works around the bug... I deleted the diameter constraint, and instead constrained the arc's center point to the nearby construction line. The end result is only one Expression is needed. With this change, double-clicking the 3.5mm constraint and changing the value now works. So it seems, with my very limited experience with thus far, that two Expressions referencing the same source is potentially problematic.

[image]

Pendelf, 2024-08-25:
Everything seems to work fine, you need to either remove the radius or remove the equality then everything works

https://github.com/user-attachments/assets/b6c0d534-e1ef-48ea-9751-e3addbaec09d

OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.8.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.24

samohd, 2024-08-25:
"Everything seems to work fine, you need to either remove the radius or remove the equality then everything works"

Umm well, yeah, but it's still a bug. :-)

Pendelf, 2024-08-26:
> "Everything seems to work fine, you need to either remove the radius or remove the equality then everything works"
> 
> Umm well, yeah, but it's still a bug. :-)

Ok :)

Mathias-Boulay, 2024-12-01:
The same kind of issue with both the release 1.0 of FreeCad and the weekly build `FreeCAD_weekly-builds-39380-conda-Linux-x86_64-py311.AppImage`.

[link to sketch with the issue](https://github.com/user-attachments/files/17970174/nanoleaf_proto_one.zip)
Increasing the constraint `size_cut` creates the issue, whereas decreasing the size doesn't have it.
Changing from the source like indicated in the issue works around it.

luzpaz, 2025-07-29:
* Perhaps has some relation to #22808

FEA-eng, 2025-09-29:
Still relevant

[image]

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/23 18:47:15
Build type: Release
Branch: main
Hash: e9ce06155d8d47aad7d6c079bfa8ba54df2b6081
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```

FEA-eng, 2025-12-14:
There's "solver failed to converge" status when recreating this dimension. @maxwxyz Can you reopen it?

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.2.0dev.20251209 (Git shallow)
Build date: 2025/12/09 13:53:17
Build type: Release
Branch: grafted,grafted
Hash: 5ad6a1ba58c665c23f1d217c23e74379d8c1a5a4
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/81.5973/1
```