FreeCAD/FreeCAD #16013: Curves workbench: Mapping circle to cone and setting thickness doesn't work right

theosib, 2024-08-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I realize that the Curves workbench is an external plugin, so this might be their fault. But someone on Discord asked me if I had filed an issue, so that seemed to be a suggestion that I should file one.

Steps:
- Install the Curves workbench
- Go to Part workbench
- Create a cone; leave its values at default
- Select the side of the cone
- Go to Curves
- Select Sketch on surface
- Open the sketch
- Draw a circle in the middle of it
- Go back to the Sketch_On_Surface object
- Set the thickness to 1

Expected:
I should see something cylinder-like sticking out of the cone.

Actual results
It's all twisted up in ways that only an image could convey.

Please also see the FCStd file attached.

[image]

[curve sketch on cone.FCStd.zip](https://github.com/user-attachments/files/16719452/curve.sketch.on.cone.FCStd.zip)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-08-22:
Setting the offset to something other than zero makes it work right.

theosib, 2024-08-22:
Someone on Discord can get this to work with an offset of zero. But I can't. He and I are both using dev version 38459. I'm betting he and I are on different platforms (I'm on Mac), and compiler optimizer differences are resulting different floating point rounding errors.

luzpaz, 2024-08-22:
Hi @tomate44, is this a curves wb issue or FreeCAD issue ?

theosib, 2024-08-22:
I'm adding other examples of failed extrusions (or whatever you call it when you set the thickness to nonzero) to the other bug report.

tomate44, 2024-08-24:
I think this is a FC bug in Part.makeRuledSurface (probably TNP related).
The automatic orientation of the ruled surface is no longer reliable.
I can reproduce with a Part.RuledSurface feature.
In the attached file, setting the thickness of SketchOnSurface to 3 is OK, but at 2, it is wrong.
[ruled_surf_bug.zip](https://github.com/user-attachments/files/16735823/ruled_surf_bug.zip)

luzpaz, 2024-08-24:
@bgbsww please weigh-in (https://github.com/FreeCAD/FreeCAD/issues/16013#issuecomment-2308178082), TIA!

bgbsww, 2024-08-24:
Thanks for the tag.  Bug does not show in LS3.  If RuledSurface::execute is switched to the non-TNP code, then the problem PERSISTS.  So I don't think this is directly TNP related.

Note that the issue manifests at  https://github.com/bgbsww/FreeCAD/blob/ebf02483af225b502fccfa7111499dfde33999fd/src/Mod/Part/App/PartFeatures.cpp#L256-L258  or the equivalent at https://github.com/bgbsww/FreeCAD/blob/ebf02483af225b502fccfa7111499dfde33999fd/src/Mod/Part/App/TopoShapeExpansion.cpp#L2327-L2329   and in the failure case, we are detecting negative and doing the reverse leading to the strange result.

Why the code has become incompatible in 0.22 I do not know.

tomate44, 2024-08-24:
I confirm that the RuledSurface feature also fails automatic orientation in FC 0.21.2
There is one thing missing in the TNP_FIX version. It does not change the comparison points for closed curves like the old version did here : 
https://github.com/bgbsww/FreeCAD/blob/ebf02483af225b502fccfa7111499dfde33999fd/src/Mod/Part/App/PartFeatures.cpp#L224-L226
https://github.com/bgbsww/FreeCAD/blob/ebf02483af225b502fccfa7111499dfde33999fd/src/Mod/Part/App/PartFeatures.cpp#L236-L238

tomate44, 2024-08-24:
For automatic orientation, I would suggest to take points slightly away from end points.
I think this would work for both open and closed curves : 
```
            // get end points of 1st curve
            gp_Pnt p1 = a1->Value(0.9*a1->FirstParameter() + 0.1*a1->LastParameter());
            gp_Pnt p2 = a1->Value(0.1*a1->FirstParameter() + 0.9*a1->LastParameter());
            if (S1.getShape().Orientation() == TopAbs_REVERSED) {
                std::swap(p1, p2);
            }

            // get end points of 2nd curve
            gp_Pnt p3 = a2->Value(0.9*a2->FirstParameter() + 0.1*a2->LastParameter());
            gp_Pnt p4 = a2->Value(0.1*a2->FirstParameter() + 0.9*a2->LastParameter());
            if (S2.getShape().Orientation() == TopAbs_REVERSED) {
                std::swap(p3, p4);
            }

```

bgbsww, 2024-08-25:
Interesting how specific this is - if you change the slope of the face on the cone by altering its height from the initial 2 um you can also toggle the error with various single digit values.  In any case, the proposed changes from @tomate44 make sense, appear to work, and to pass all current tests.

PR #16090 (2024-09-02): Fix #16013
Fix #16013 and fix the Ruled Surface element map tests while I'm at it.
commit 3a6f7094 (2024-09-02): Fix #16013 (#16090)

* Fix #16013

* Fix #16013
closed 2024-09-02 by yorikvanhavre