FreeCAD/FreeCAD #16020: Regressen: Desktop integration KDE

berberic2, 2024-08-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With the current FreeCAD the desktop integration with KDE is broken. Some icons in the KDE-filedialog are replaced by wrong icons from another theme.
The cause are commits from https://github.com/FreeCAD/FreeCAD/pull/15588

[image]
Left: KDE-filedialog as it should look like and as it looked iin FreeCAD in the past.
Right: KDE-filedialog as it looks now. Red arrows point to wrong icons.

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 +3 (Git)
Build type: Release
Branch: Meins
Hash: 5f17684668a110f3978b52785abc8efee74a2548
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly4 0.50.15
  * sheetmetal 0.4.24
  * Silk 0.1.5
  * CurvedShapes 1.0.9
  * DynamicData 2.62.0
  * FreeCAD_assembly3 0.11.3
  * fasteners 0.5.26
  * QuickMeasure 2022.10.28
  * ExplodedAssembly
  * Curves 0.6.44
  * Pyramids-and-Polyhedrons
  * freecad.gears 1.2.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2024-08-23:
Yes, and it would be good for @berberic2 to test it first.

berberic2, 2024-08-24:
> Yes, and it would be good for @berberic2 to test it first.

Yes (for me) https://github.com/FreeCAD/FreeCAD/pull/16018, fixes the problem.

luzpaz, 2024-08-24:
KDE related: #8351 
Could someone test this other ticket and see if it's still relevant ?

maxwxyz, 2024-12-07:
Closing as fixed by #16018

closed 2024-12-07 by maxwxyz