FreeCAD/FreeCAD #16022: Gui::Application::Instance should be used instead of getGuiApplication()

aiksiongkoh, 2024-08-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

getGuiApplication() is used 61 times
Gui::Application::Instance is used 669 times
Best to remove getGuiApplication()

### Full version info

```shell
https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/freecad_source.tar.gz
still has getGuiApplication() as of today.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-23:
@aiksiongkoh btw, the most up-to-date source is at https://github.com/FreeCAD/FreeCAD

luzpaz, 2024-08-23:
`getGuiApplication()` :arrow_right:  https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD%20getGuiApplication()%20&type=code

`Gui::Application::Instance` :arrow_right:  https://github.com/search?q=repo%3AFreeCAD%2FFreeCAD+Gui%3A%3AApplication%3A%3AInstance&type=code