FreeCAD/FreeCAD #16023: ActiveDocument should be used instead of activeDocument()

aiksiongkoh, 2024-08-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

...ActiveDocument is used 3477 times
...activeDocument() is used 639 times
activeDocument() should be removed

### Full version info

```shell
https://github.com/FreeCAD/FreeCAD-Bundle/releases/download/weekly-builds/freecad_source.tar.gz
still has activeDocument() as of today.
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-24:
@aiksiongkoh again, 
`FreeCAD-Bundle/releases/download/`**weekly-builds**`freecad_source.tar.gz  
lags behind[ FreeCAD/FreeCAD ](https://github.com/FreeCAD/FreeCAD)

`git clone https://github.com/FreeCAD/FreeCAD` is what you're looking for

redPanther, 2024-08-25:
just to update the stats with latest git version
...ActiveDocument is used 3342 times
...activeDocument() is used 924 times

redPanther, 2024-08-25:
just to update the stats with latest git version:
...ActiveDocument is used 2933 times
...activeDocument() is used 920 times

s-quirin, 2024-08-27:
Has the recommendation changed towards ActiveDocument? Because the recommendation used to be activeDocument().
[Forum discussion](https://forum.freecad.org/viewtopic.php?t=44133)

aiksiongkoh, 2024-08-27:
I think ActiveDocument is very explicit and unambiguous. There is always uncertainty if activeDocument() or getActiveDocument() does so something extra, especially for newcomers like me. If the functions are identical to ActiveDocument, then they should not exist. If the do something extra, the function name should say so.

wwmayer, 2024-08-28:
Removing a method that has existed for many years is a bad idea because this will break existing Python addons or macro code.

luzpaz, 2024-08-28:
Perhaps we can start to deprecate them over time ?