FreeCAD/FreeCAD #16024: specifying equal sides to a rectangle not available from keyboard

iplayfast, 2024-08-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a square, you can type in the dimension of one side, but there is no way of saying that the other side is equal. Normally you would do that with the 'e' key, but using the 'e' key on the 2nd dimension causes the square to be aborted.

### Full version info

```shell
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Unknown
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/Canada (en_CA)
Installed mods: 
  * freecad.gears 1.0.0
  * 3D_Printing_Tools
  * CycloidGearBox
  * FeedsAndSpeeds 0.5.0
  * Ondsel-Lens 2024.1.22.01
  * ExplodedAssembly
  * Nodes 0.1.36
  * 3DfindIT 1.2.0
  * A2plus 0.4.63
  * MeshRemodel 1.9.22
  * lattice2 1.0.0
  * fasteners 0.5.10
  * Assembly3 0.12.2
  * Assembly4 0.50.7
  * POV-Ray-Rendering
  * PieMenu 1.2.7
  * CurvedShapes 1.0.5
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-24:
@iplayfast is this a regression from  0.21.x ?

Syres916, 2024-08-24:
@luzpaz it can't be a regression as OVP didn't exist in 0.21.x.

What I can't understand, there's a command for creating a Square but it's even less user intuitive as it's dimensioned based on a peripheral circle like a hexagon. I'll have a look at making the Square command run the Rectangle code but only asking for one dimension and automatically applying the equals constraint but please don't ask for rounded corners etc just yet, small steps and all that.

iplayfast, 2024-08-30:
@luzpaz No, .21.x can't enter in dimensions when creating a rectangle. I guess this is an enhancement request more than a bug, it just seemed like an intuative thing to do, but pressing 'e' aborts the square creation totally which seems like a bug.