FreeCAD/FreeCAD #16025: Freecad crashes won't let me fuse two linked objects together.

noahmartinwilliams, 2024-08-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When I select two linked objects, and press the boolean button both objects disappear and the boolean parameters window thing (I'm not sure what it's called but it's under the tasks tab) shows up. That seems kind of weird to me, and when I press the "Add Body" button and then select the first body in the Model tab I'm able to add that to the list of bodies, but not the second body.

I'm pretty sure this is a bug as it's easy to get it to crash with a segfault from here. If I try selecting one body, press the boolean fusion button, select the other object which is linked to it, and do ctrl+z it crashes with a segfault.

### Full version info

```shell
[code]
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.1, OCC 7.7.2
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

PartDesign

### Anything else?

[zipped.zip](https://github.com/user-attachments/files/16733809/zipped.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-24:
Could you please try this in the weekly builds?
Please provide steps how to reproduce, are you doing this in one body, 3 bodies,...?

noahmartinwilliams, 2024-08-24:
I tried the app image from the link and the two bodies still disappear. Also I did provide the steps I went through to reproduce in the bug report, but here they are again:
* Create an object
* Create another object that's linked to the first one
* Select them both and press the boolean operation button to fuse them together.
* They both disappear for some reason.
* A window(ish) pops up in the tasks tab asking me to select an object.
* I'm only able to select the first object for some reason and they still disappear after I click ok.
* If I press ctrl+z at this point the program crashes with a segfault.

luzpaz, 2024-08-24:
@noahmartinwilliams please paste the full [About](https://wiki.freecad.org/About) info of the development appimage you tested. TIA

noahmartinwilliams, 2024-08-24:
Oh, right. Sorry I forgot to do that.
```
OS: Arch Linux (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git) AppImage
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

Roy-043, 2024-08-25:
If you activate the Link to the Body and then do a PartDesign boolean operation using the original Body as the tool, you are creating a cyclic reference. There is a warning to that effect (on Windows, but Linux may crash here):
13:26:09  <Exception> Cyclic dependencies detected: Search cannot be performed

The bug is that the user is not prevented from doing this.

It is actually quite rare to use boolean operations in the PartDesign WB. Features are either additive or subtractive. In this case a mirror pattern would be the more obvious (and easier) solution.

noahmartinwilliams, 2024-08-25:
Ok. I'll look into mirroring then.

FEA-eng, 2026-11-03:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 27.1.0dev.20261020 (Git shallow)
Build date: 2026/10/20 22:15:52
Build type: Release
Branch: (HEAD detached at 9946038aa)
Hash: 9946038aace97af288eb28f2727a169f9d1158c7
Python 3.13.15, Qt 6.11.2, Coin 4.0.10 (bundled), Pivy 0.6.11 (bundled), Vtk 9.7.0, boost 1_90, Eigen3 5.0.1, PySide 6.11.2
shiboken 6.11.2, xerces-c 3.3.0, Clipper2 2.0.1, IfcOpenShell 0.8.5, OCC 8.0.1
Locale: Polish/Poland (pl_PL)
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
QPA/File dialog/Color dialog: windows/via Win32/via Qt
Navigation Style/Orbit Style/Rotation Mode: Blender/Rounded Arcball/Window center
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
OpenGL version/vendor/renderer: 4.6.0 NVIDIA 474.82/NVIDIA Corporation/Quadro K4000/PCIe/SSE2
```