FreeCAD/FreeCAD #16026: Draft polar array causes reliable crash

theosib, 2024-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Steps:
- Load the attached document
- Select Sketch_On_Surface001
- Go to the Draft workbench
- Create a polar array
- Specify 30 as the number of elements

This results in a crash. The crash report from MacOS is attached.

[knotplate.FCStd.zip](https://github.com/user-attachments/files/16734622/knotplate.FCStd.zip)
[knot crash.txt](https://github.com/user-attachments/files/16734633/knot.crash.txt)

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38459 (Git)
Build type: Release
Branch: main
Hash: 58729b3ab0060a1525fc45c87beabfd4b3d5e22e
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * Curves 0.6.44
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

theosib, 2024-08-24:
BTW, I can do up to 25. If I increase it to 26, it crashes. This attachment has it at 25. So all you have to do to get the crash is increase the count to 26.
[knotplate2.FCStd.zip](https://github.com/user-attachments/files/16734640/knotplate2.FCStd.zip)

luzpaz, 2024-08-24:
Roy, assigned to you so you see this. Feel free to unassigned if in error

Roy-043, 2024-08-24:
There are a number of issues.

The most important one is that setting "Fuse" to true for a Link array actually does not make sense, you can't fuse such an array. But the task panel does allow checking both the Link and Fuse checkboxes. What is worse is that apparently the code does not ignore the "Fuse" setting for Link arrays. Both issues need to be addressed and would solve (avoid) the reported crash.

If a fused array is the target then the arrayed shape should be simplified. Currently the top and bottom faces are not planar. which results in unnecessarily long calculation times and unwanted faces. You must then recreate the array and uncheck the Link checkbox in the task panel. Of course the resultant array must have overlapping objects for the "Fuse" property to actually have an effect.

Looking at the model I don't think a fused array is intended, so for now the advice is to set "Fuse" to false for this array.

theosib, 2024-08-24:
I think this highlights one of the controversies over making fuse the default. I didn't set it to true. It just came that way, and I didn't even notice it.

I suspect the faces are not planar as a side-effect of the way the Curves workbench works. This was the only way I could find to warp a sketch around the edge of a disc, but I can't set a cone's height to zero (possibly another bug). 

As for the Link setting, I'm not sure which option that refers to since for me "Link Transform" is already set to false.

Finally, thank you for the analysis since unchecking "fuse" does mitigate the crash.

theosib, 2024-08-24:
Incidentally, this inability to fuse does cause problems down the line. Notice how it's impossible to fuse this array with anything else. That's okay since what I really want is probably a collection so that I can color the pattern differently in the slicer if I want. But this might be a bug  to look into.

[knotplate3.FCStd.zip](https://github.com/user-attachments/files/16736665/knotplate3.FCStd.zip)

theosib, 2024-08-24:
This, BTW, is another of numerous cases where exporting as STEP produces a wrong result, while exporting as STL works fine.

Roy-043, 2024-08-25:
> I didn't set it to true. It just came that way, and I didn't even notice it.

That is not correct. You must have accidentally checked the Fuse checkbox at some point.

> Incidentally, this inability to fuse does cause problems down the line.

There is a mix up here. The Fuse property of Draft arrays only pertains to the fusion of objects within the array. The problem here seems to be related to this particular model. See the attached file (remove the zip extension): [knotplate3-roy.FCStd.zip](https://github.com/user-attachments/files/16739926/knotplate3-roy.FCStd.zip).

> This, BTW, is another of numerous cases where exporting as STEP produces a wrong result.

I have not checked that. Please create a separate issue, thanks.

theosib, 2024-08-27:
I did not explicitly enable fuse when creating the array. However, if there's a global setting, I think I may have changed it from the default based on advice to work around a problem with an earlier version of FreeCAD.

theosib, 2024-08-27:
It looks like you intersected the extrusion with a cube in order to flatten it. That's a fantastic solution. Thank you!

theosib, 2024-08-27:
I can't find a global setting for this. But by default, whenever I create a Draft Array, Fuse and Link Array are both set by default.

Roy-043, 2024-12-11:
I have to correct myself. The Fuse property does work for a Link array. But in the 3D view the unfused array is shown. See the attached file (remove the .zip extension). If you use Part_CheckGeometry on the Link array and the non-Link array, 17 faces are reported in both cases. And the Shape of the Link array is a Compound containing a singe Solid.

[Star.FCStd.zip](https://github.com/user-attachments/files/18094227/Star.FCStd.zip)

Roy-043, 2026-10-10:
I can't reproduce the issue with the knotplate2.FCStd file. Increasing NumberPolar to 26 works fine. I am therefore going to close this one. Note that what I said about the Fuse option for Link arrays is not correct. They can be fused, but this is just not correctly shown in the 3D View.

closed 2026-10-10 by Roy-043