FreeCAD/FreeCAD #16028: Mesh: Quasi-structured Quad Gmsh algorithm doesn't work with Mesh_FromPartShape but works with Mesh_RemeshGmsh

FEA-eng, 2024-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As pointed out in [this issue](https://github.com/FreeCAD/FreeCAD-Bundle/issues/277) (posted in the wrong repo), the newly added Quasi-structured Quad Gmsh algorithm doesn't work with _Create mesh from shape_ (_Mesh_FromPartShape_). The following messages appear:

```
...
Warning : No vector element found containing point (9.76562,10,0) in PostView field (for norm)
...
Warning : Mesh generation error summary
Warning :  1665 warnings
Warning :     0 errors
Warning : Check the full log for details
Warning : --------------------------
```

However, it works with _Refinement_ (_Mesh_RemeshGmsh_).

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Mesh

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-12-04:
Which do you want to close ?

FEA-eng, 2024-12-05:
> Which do you want to close ?

Maybe let's close this one and add the information from here to the other one.

closed 2025-11-14 by luzpaz