FreeCAD/FreeCAD #16029: Error message when trying to change line witdh in Draft workbench

APOS80, 2024-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Klicking to set line width results in error message.

[image]

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Swedish/Sweden (sv_SE)
[/code]
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-24:
could you try this in the weekly build https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds

Roy-043, 2024-08-25:
Under normal circumstances it should be impossible to start Draft_SetStyle if another task panel is still active. So steps to reproduce are essential here.

APOS80, 2024-08-25:
It’s strange, I don’t get it now. 

Don’t know why.

maxwxyz, 2024-08-25:
I'll close this issue then. Please comment if you are able to reproduce it again and we can reopen

closed 2024-08-25 by maxwxyz