FreeCAD/FreeCAD #16034: Part Design: Body: Shape Appearance colors do not set correctly

samohd, 2024-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is weirdness in setting Body > Shape Appearance colors.

When you expand the Shape Appearance section, the Shape Appearance color and Diffuse Color will set the same, regardless of which you set. If they are actually the same setting then when you expand, it should remove the options in the Shape Appearance row.

Setting either of these colors in the color requester results in the wrong values... For example, RGB=100,100,100 becomes 99,99,99. 102,101,99 becomes 102,102,99. When you go back to the color requester they have the incorrect numbers. So, the requester values are being incorrectly obtained.

[image]

Other colors in the Shape Appearance section have the same problem. I also tried Line Color and Point Color above this section and those seem to work.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Pendelf, 2024-08-25:
Confirmed, the problem is reproducing 

https://github.com/user-attachments/assets/fe4483c8-40e4-4a3a-8c4a-8c4060ff84e5

OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * A2plus.backup1693145886.351258 0.4.60n
  * Assembly3.backup1697878740.3967428 0.12.0
  * Assembly4.backup1697878772.1268227 0.50.3
  * BIM.backup1704622292.3622465 2021.12.0
  * CurvedShapes.backup1704622353.7785401 1.0.5
  * Curves.backup1693145923.4219682 0.6.13
  * Design456.backup1712597574.1363904 0.0.1
  * dodo.backup1712597608.2771313 1.0.1
  * ExplodedAssembly
  * fasteners 0.5.24
  * FEMbyGEN 2.1.0
  * lattice2 1.0.0
  * LCInterlocking
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.8.0
  * Pyramids-and-Polyhedrons
  * sheetmetal 0.4.24

FEA-eng, 2026-10-31:
See [this](https://github.com/FreeCAD/FreeCAD/issues/16034#issuecomment-3394319863)