FreeCAD/FreeCAD #16035: Part Design: Model > Data > Pad: GUI has wrong value editors

samohd, 2024-08-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In Part Design: Model panel > Data > Pad: the Length2, Direction:X/Y/Z, Offset all have the wrong editors for values, and the values can not be changed.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

olvll, 2024-08-25:
Not really a bug as far as I can see. These values become editable when you set "Use Custom Vector" (directly above) to true. So there's some logic and intent behind it, even though it might not be very intuitive. Perhaps FreeCAD should give a message when you try to edit some property if another property blocks this.

samohd, 2024-08-25:
Ok, I see what you are saying... these are showing the value editor, but it's disabled. Where I was thrown off by this is the cursor vertical bar, suggesting it was functional. If the values can't be edited, then the field should be disabled to begin with, thus no popup disabled requester should show. So, I guess this is a feature request rather than a bug.