FreeCAD/FreeCAD #16039: Feature : Tree View reorganization

tibfel, 2024-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I think that it would be a good idea to open each document in a separate MDI tab (a document that was intentionally opened by the user) and hide the objects in memory (that were loaded when a part was inserted into the assembly), or at least hide them into a Linked Objects Folder.  

In the current state, the model tree is cluttered with everything that is open in memory (every file and linked object become part of the tree view) and it is hard to navigate. Every part in the assembly has its tree view and can be edited from there, so there is no need to display it as a separate file in the tree view under the assembly tree view. Those files should be hidden from the user (as are in Inventor, Solidworks, SolidEdge, etc.). 
Also in Part or Techdraw, linked objects could be positioned in the linked object subfolder within the tree view, so I do not accidentally

Also, different object types can be grouped into subfolders (like joints are in the assembly workbench). Dimensions in techdraw can also be grouped in subfolders so they do not clutter the tree view. 

Thank you for any feedback. 

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Slovak/Slovakia (sk_SK)
Installed mods: 
  * OpenTheme 2024.8.15
  * Render 2024.8.4
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2024-08-25:
are you aware of [Std_TreeCollapseDocument](https://wiki.freecad.org/Std_TreeCollapseDocument)?

tibfel, 2024-08-25:
@Roy-043 Thank you very much. I did not know about that setting and I really looked into everything in preferences. So I can close this issue.

closed 2024-08-25 by luzpaz