FreeCAD/FreeCAD #16047: Segfault in StartView::firstStartWidgetDismissed

outpaddling, 2024-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The latest release of FreeCAD is stable on FreeBSD and I use it pretty regularly.

I'm working on a port of the latest  commits now, and I'm getting a segfault right after selecting a theme during startup.  Backtrace using FreeCAD built with debug info is shown below.  I can also build Qt components with debug info if that might help.  Any suggestions would be appreciated.

```
# lldb -c FreeCAD.core /usr/local/bin/FreeCAD
(lldb) target create "/usr/local/bin/FreeCAD" --core "FreeCAD.core"
Core file '/home/bacon/FreeCAD.core' (x86_64) was loaded.
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV
  * frame #0: 0x0000000837822763 libQt5Widgets.so.5`QScrollArea::ensureWidgetVisible(QWidget*, int, int) + 51
    frame #1: 0x0000000837822727 libQt5Widgets.so.5`QScrollArea::focusNextPrevChild(bool) + 55
    frame #2: 0x00000008376d2806 libQt5Widgets.so.5`QWidgetPrivate::hide_helper() + 566
    frame #3: 0x00000008376d2bf1 libQt5Widgets.so.5`QWidgetPrivate::setVisible(bool) + 369
    frame #4: 0x00000008376bbc23 libQt5Widgets.so.5`QStackedLayout::setCurrentIndex(int) + 403
    frame #5: 0x00000008b8710f9c StartGui.so`StartGui::StartView::firstStartWidgetDismissed(this=0x000001a5a9c743c0) at StartView.cpp:460:16
    frame #6: 0x00000008b8718769 StartGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (StartGui::StartView::*)()>::call(f=20 0f 71 b8 08 00 00 00 00 00 00 00 00 00 00 00, o=0x000001a5a9c743c0, arg=0x0000000820f6f2d8) at qobjectdefs_impl.h:152:13
    frame #7: 0x00000008b87186cd StartGui.so`void QtPrivate::FunctionPointer<void (StartGui::StartView::*)()>::call<QtPrivate::List<>, void>(f=20 0f 71 b8 08 00 00 00 00 00 00 00 00 00 00 00, o=0x000001a5a9c743c0, arg=0x0000000820f6f2d8) at qobjectdefs_impl.h:185:13
    frame #8: 0x00000008b8718612 StartGui.so`QtPrivate::QSlotObject<void (StartGui::StartView::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x000001a5aa710e80, r=0x000001a5a9c743c0, a=0x0000000820f6f2d8, ret=0x0000000000000000) at qobjectdefs_impl.h:418:17                                                         
    frame #9: 0x000000083b056672 libQt5Core.so.5`___lldb_unnamed_symbol9682 + 1074
    frame #10: 0x00000008b86d58e2 StartGui.so`StartGui::FirstStartWidget::dismissed(this=0x000001a5aa7267a0) at moc_FirstStartWidget.cpp:132:5
    frame #11: 0x00000008b8709939 StartGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (StartGui::FirstStartWidget::*)()>::call(f=c0 58 6d b8 08 00 00 00 00 00 00 00 00 00 00 00, o=0x000001a5aa7267a0, arg=0x0000000820f6f530) at qobjectdefs_impl.h:152:13
    frame #12: 0x00000008b870989d StartGui.so`void QtPrivate::FunctionPointer<void (StartGui::FirstStartWidget::*)()>::call<QtPrivate::List<>, void>(f=c0 58 6d b8 08 00 00 00 00 00 00 00 00 00 00 00, o=0x000001a5aa7267a0, arg=0x0000000820f6f530) at qobjectdefs_impl.h:185:13
    frame #13: 0x00000008b87097e2 StartGui.so`QtPrivate::QSlotObject<void (StartGui::FirstStartWidget::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x000001a5aa710900, r=0x000001a5aa7267a0, a=0x0000000820f6f530, ret=0x0000000000000000) at qobjectdefs_impl.h:418:17
    frame #14: 0x000000083b056672 libQt5Core.so.5`___lldb_unnamed_symbol9682 + 1074
    frame #15: 0x0000000837775de3 libQt5Widgets.so.5`___lldb_unnamed_symbol14663 + 115
    frame #16: 0x0000000837775c60 libQt5Widgets.so.5`___lldb_unnamed_symbol14661 + 240
    frame #17: 0x0000000837776d93 libQt5Widgets.so.5`QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 291
    frame #18: 0x00000008376d3182 libQt5Widgets.so.5`QWidget::event(QEvent*) + 514
    frame #19: 0x000000083781bf04 libQt5Widgets.so.5`QPushButton::event(QEvent*) + 180
    frame #20: 0x0000000837699b43 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 259
    frame #21: 0x000000083769ca20 libQt5Widgets.so.5`QApplication::notify(QObject*, QEvent*) + 7488
    frame #22: 0x0000000828cefa34 libFreeCADGui.so`Gui::GUIApplication::notify(this=0x0000000820f70f00, receiver=0x000001a5a9cd63b0, event=0x0000000820f6fcf8) at GuiApplication.cpp:83:34
    frame #23: 0x000000083b022df1 libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 177
    frame #24: 0x000000083769a4a5 libQt5Widgets.so.5`QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 901
    frame #25: 0x00000008376f02f7 libQt5Widgets.so.5`___lldb_unnamed_symbol14087 + 2039
    frame #26: 0x00000008376ef1f4 libQt5Widgets.so.5`___lldb_unnamed_symbol14081 + 212
    frame #27: 0x0000000837699b43 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 259
    frame #28: 0x000000083769aeee libQt5Widgets.so.5`QApplication::notify(QObject*, QEvent*) + 526
    frame #29: 0x0000000828cefa34 libFreeCADGui.so`Gui::GUIApplication::notify(this=0x0000000820f70f00, receiver=0x000001a5a9b6e700, event=0x0000000820f70430) at GuiApplication.cpp:83:34
    frame #30: 0x000000083b022df1 libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 177
    frame #31: 0x00000008357a6edc libQt5Gui.so.5`QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 3308
    frame #32: 0x000000083578adcb libQt5Gui.so.5`QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 219
    frame #33: 0x000000085ff55c2f libQt5XcbQpa.so.5`___lldb_unnamed_symbol3550 + 31
    frame #34: 0x000000084e066f71 libglib-2.0.so.0`___lldb_unnamed_symbol2541 + 369
    frame #35: 0x000000084e06731f libglib-2.0.so.0`___lldb_unnamed_symbol2542 + 591
    frame #36: 0x000000084e0673b6 libglib-2.0.so.0`g_main_context_iteration + 102
    frame #37: 0x000000083b075915 libQt5Core.so.5`QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 101
    frame #38: 0x000000083b01f786 libQt5Core.so.5`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 550
    frame #39: 0x000000083b023471 libQt5Core.so.5`QCoreApplication::exec() + 145
    frame #40: 0x0000000828aca3ce libFreeCADGui.so`(anonymous namespace)::tryRunEventLoop(mainApp=0x0000000820f70f00) at Application.cpp:2062:5
    frame #41: 0x0000000828ac7743 libFreeCADGui.so`(anonymous namespace)::runEventLoop(mainApp=0x0000000820f70f00) at Application.cpp:2082:9
    frame #42: 0x0000000828ac6e1e libFreeCADGui.so`Gui::Application::runApplication() at Application.cpp:2159:5
    frame #43: 0x000000000022cd94 FreeCAD`main(argc=1, argv=0x0000000820f71978) at MainGui.cpp:300:13
    frame #44: 0x000000083bda2afa libc.so.7`__libc_start1 + 298
    frame #45: 0x000000000022acb0 FreeCAD`_start at crt1_s.S:83
```

### Full version info

```shell
Can't get that far since it crashes on startup.

# git describe --tags
0.21rc1-4839-gbaec1037d8
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

outpaddling, 2024-08-25:
Actually, I *can* get to the About screen while the startup config dialog is open:

Version 0.22.0dev
Revision number 14555 (Git shallow)
Release date 2024/08/24 23:38:29
Operating system FreeBSD 14.0-RELEASE-p8
Word size 64-bit
Branch master
Hash 44fa4c5

luzpaz, 2024-08-25:
There is no master branch and that revision (14555) is waay off. We're currently at revision 38513

outpaddling, 2024-08-25:
I have no idea how it's coming up with that revision number and branch, but the working 0.21.2 port shows the same thing:

FreeBSD coral.acadix  bacon ~ 1001: FreeCAD
FreeCAD 0.21.2, Libs: 0.21.2R14555 (Git shallow)
...

I'm using the latest commit from main branch as of this morning.

outpaddling, 2024-08-25:
The 14555 is hard-coded into src/Tools/SubWCRev.py:

```
    def revisionNumber(self, srcdir, origin=None):
        """sets the revision number
        for main and release branches all commits are counted
        for other branches. The version number is split in to two parts:
        The first number reflects the number of commits in common with the
        blessed main repository. The second part (separated by " +") reflects
        the number of commits that are different from the main repository"""
        referencecommit = "7d8e53aaab17961d85c5009de34f69f2af084e8b"
        referencerevision = 14555
```

luzpaz, 2024-08-25:
https://github.com/FreeCAD/FreeCAD/blob/7c797b5a10c19ae55e317c4ff9a02823a4b4dc0b/src/Tools/SubWCRev.py#L294-L302

Curiously it does. Oh ok so 7d8e53aaab17961d85c5009de34f69f2af084e8b is the 0.21.2 stable tag AKA current stable. Can you test building FreeCAD from HEAD (instead of a tag) ?

outpaddling, 2024-08-25:
I believe I am using head (do you mean main?).  My last attempt used commit baec1037d8ce9c02f6dc67685926fce9dbaf42cf.  Same results as two other recent commits.

Is FreeCAD stable with Qt6 at this point?  If so, I don't think it's worth debugging problems involving Qt5 at this stage.  I'd rather put my time into moving the port forward.

luzpaz, 2024-08-26:
Yes, I mean HEAD of 'main' branch. 

Yes, we're Qt6 stable. Moving the port to Qt6 makes sense.

luzpaz, 2024-08-27:
@outpaddling can we close this ticket in the meantime ?

outpaddling, 2024-08-27:
Sure.  It will always be here if we want to revisit.

luzpaz, 2024-08-27:
Thanks for your work on the FreeBSD port :heart:

outpaddling, 2024-08-28:
So, getting the same error with Qt6:

```
lldb -c FreeCAD.core /usr/local/bin/FreeCAD
(lldb) target create "/usr/local/bin/FreeCAD" --core "FreeCAD.core"
Core file '/home/bacon/FreeCAD.core' (x86_64) was loaded.
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV
  * frame #0: 0x00000008380db313 libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) + 51
    frame #1: 0x00000008380db2d7 libQt6Widgets.so.6`QScrollArea::focusNextPrevChild(bool) + 55
    frame #2: 0x0000000837f4f516 libQt6Widgets.so.6`QWidgetPrivate::hide_helper() + 566
    frame #3: 0x0000000837f503e9 libQt6Widgets.so.6`QWidgetPrivate::setVisible(bool) + 969
    frame #4: 0x0000000837f4fd96 libQt6Widgets.so.6`QWidget::setVisible(bool) + 662
    frame #5: 0x0000000837f35140 libQt6Widgets.so.6`QStackedLayout::setCurrentIndex(int) + 384
    frame #6: 0x00000008d359022c StartGui.so`StartGui::StartView::firstStartWidgetDismissed(this=0x0000022f79729a00) at StartView.cpp:469:16
    frame #7: 0x00000008d3597882 StartGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (StartGui::StartView::*)()>::call(f=b0 01 59 d3 08 00 00 00 00 00 00 00 00 00 00 00, o=0x0000022f79729a00, arg=0x0000000820784768) at qobjectdefs_impl.h:145:13
    frame #8: 0x00000008d35977cd StartGui.so`void QtPrivate::FunctionPointer<void (StartGui::StartView::*)()>::call<QtPrivate::List<>, void>(f=b0 01 59 d3 08 00 00 00 00 00 00 00 00 00 00 00, o=0x0000022f79729a00, arg=0x0000000820784768) at qobjectdefs_impl.h:182:13
    frame #9: 0x00000008d35976f5 StartGui.so`QtPrivate::QCallableObject<void (StartGui::StartView::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x0000022f75f6a0e0, r=0x0000022f79729a00, a=0x0000000820784768, ret=0x0000000000000000) at qobjectdefs_impl.h:553:21
    frame #10: 0x000000083cb7459a libQt6Core.so.6`___lldb_unnamed_symbol12370 + 1354
    frame #11: 0x00000008d3551602 StartGui.so`StartGui::FirstStartWidget::dismissed(this=0x0000022f768b4550) at moc_FirstStartWidget.cpp:139:5
    frame #12: 0x00000008d3586652 StartGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (StartGui::FirstStartWidget::*)()>::call(f=e0 15 55 d3 08 00 00 00 00 00 00 00 00 00 00 00, o=0x0000022f768b4550, arg=0x00000008207849d0) at qobjectdefs_impl.h:145:13
    frame #13: 0x00000008d358659d StartGui.so`void QtPrivate::FunctionPointer<void (StartGui::FirstStartWidget::*)()>::call<QtPrivate::List<>, void>(f=e0 15 55 d3 08 00 00 00 00 00 00 00 00 00 00 00, o=0x0000022f768b4550, arg=0x00000008207849d0) at qobjectdefs_impl.h:182:13
    frame #14: 0x00000008d35864c5 StartGui.so`QtPrivate::QCallableObject<void (StartGui::FirstStartWidget::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x0000022f761a0660, r=0x0000022f768b4550, a=0x00000008207849d0, ret=0x0000000000000000) at qobjectdefs_impl.h:553:21
    frame #15: 0x000000083cb7459a libQt6Core.so.6`___lldb_unnamed_symbol12370 + 1354
    frame #16: 0x0000000838014453 libQt6Widgets.so.6`___lldb_unnamed_symbol18148 + 115
    frame #17: 0x00000008380142e3 libQt6Widgets.so.6`___lldb_unnamed_symbol18146 + 243
    frame #18: 0x0000000838015340 libQt6Widgets.so.6`QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 208
    frame #19: 0x0000000837f50cd5 libQt6Widgets.so.6`QWidget::event(QEvent*) + 469
    frame #20: 0x00000008380d1ea9 libQt6Widgets.so.6`QPushButton::event(QEvent*) + 265
    frame #21: 0x0000000837efee88 libQt6Widgets.so.6`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 248
    frame #22: 0x0000000837f0132f libQt6Widgets.so.6`QApplication::notify(QObject*, QEvent*) + 6015
    frame #23: 0x0000000825ed348f libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00000008207863a0, receiver=0x0000022f7974dab0, event=0x0000000820785280) at GuiApplication.cpp:83:34
    frame #24: 0x000000083cb20654 libQt6Core.so.6`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 180
    frame #25: 0x0000000837eff4e0 libQt6Widgets.so.6`QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 960
    frame #26: 0x0000000837f688b5 libQt6Widgets.so.6`___lldb_unnamed_symbol17064 + 1669
    frame #27: 0x0000000837f67a39 libQt6Widgets.so.6`___lldb_unnamed_symbol17060 + 121
    frame #28: 0x0000000837efee88 libQt6Widgets.so.6`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 248
    frame #29: 0x0000000837effdd9 libQt6Widgets.so.6`QApplication::notify(QObject*, QEvent*) + 553
    frame #30: 0x0000000825ed348f libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00000008207863a0, receiver=0x0000022f79924aa0, event=0x00000008207858e0) at GuiApplication.cpp:83:34
    frame #31: 0x000000083cb20654 libQt6Core.so.6`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 180
    frame #32: 0x000000083a7192b8 libQt6Gui.so.6`QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 2264
    frame #33: 0x000000083a77a86b libQt6Gui.so.6`QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 395
    frame #34: 0x00000008654c88e9 libQt6XcbQpa.so.6`___lldb_unnamed_symbol2727 + 41
    frame #35: 0x000000083cb29b07 libQt6Core.so.6`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 519
    frame #36: 0x000000083cb20e44 libQt6Core.so.6`QCoreApplication::exec() + 148
    frame #37: 0x0000000825c914db libFreeCADGui.so`(anonymous namespace)::tryRunEventLoop(mainApp=0x00000008207863a0) at Application.cpp:2062:5
    frame #38: 0x0000000825c8e763 libFreeCADGui.so`(anonymous namespace)::runEventLoop(mainApp=0x00000008207863a0) at Application.cpp:2082:9
    frame #39: 0x0000000825c8ddfe libFreeCADGui.so`Gui::Application::runApplication() at Application.cpp:2159:5
    frame #40: 0x000000000022cc8a FreeCAD`main(argc=1, argv=0x0000000820787128) at MainGui.cpp:300:13
    frame #41: 0x000000083d9b2afa libc.so.7`__libc_start1 + 298
    frame #42: 0x000000000022aa40 FreeCAD`_start at crt1_s.S:83
```

outpaddling, 2024-08-28:
> Thanks for your work on the FreeBSD port ❤️

Ditto * 1000 to everyone developing FreeCAD!

luzpaz, 2024-08-28:
@outpadding, you're still getting incorrect about info after building main ?

thyssentishman, 2024-09-06:
I'm having the same issue with the [OpenBSD port of FreeCAD ](https://forum.freecad.org/viewtopic.php?t=80792)I'm working on. I'm also building from HEAD (fe40748).

<details>
<summary>Backtrace</summary><pre>
Reading symbols from /usr/local/lib/freecad/bin/FreeCAD...
[New process 329929]
[New process 541059]
[New process 182906]
[New process 324312]
[New process 287037]
[New process 168999]
[New process 314166]
[New process 507709]
[New process 208595]
[New process 111048]
[New process 573217]
[New process 143001]
[New process 486576]
[New process 248880]
[New process 548918]
[New process 268263]
[New process 336193]
[New process 511449]
[New process 159470]
[New process 359197]
Core was generated by `FreeCAD'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00000e8c54e2b566 in QScrollArea::ensureWidgetVisible(QWidget*, int, int) () from /usr/local/lib/libQt6Widgets.so.2.1
[Current thread is 1 (process 329929)]
(gdb) bt
#0  0x00000e8c54e2b566 in QScrollArea::ensureWidgetVisible(QWidget*, int, int) () from /usr/local/lib/libQt6Widgets.so.2.1
#1  0x00000e8c54e2b509 in QScrollArea::focusNextPrevChild(bool) () from /usr/local/lib/libQt6Widgets.so.2.1
#2  0x00000e8c54c89dc6 in QWidgetPrivate::hide_helper() () from /usr/local/lib/libQt6Widgets.so.2.1
#3  0x00000e8c54c8a1d2 in QWidgetPrivate::setVisible(bool) () from /usr/local/lib/libQt6Widgets.so.2.1
#4  0x00000e8c54c6dfa5 in QStackedLayout::setCurrentIndex(int) () from /usr/local/lib/libQt6Widgets.so.2.1
#5  0x00000e8d00d41dfb in StartGui::StartView::firstStartWidgetDismissed() () from /usr/local/lib/freecad/lib/StartGui.so
#6  0x00000e8cf4c11d30 in void doActivate<false>(QObject*, int, void**) () from /usr/local/lib/libQt6Core.so.4.1
#7  0x00000e8cf4c11d30 in void doActivate<false>(QObject*, int, void**) () from /usr/local/lib/libQt6Core.so.4.1
#8  0x00000e8c54d58e85 in QAbstractButtonPrivate::emitClicked() () from /usr/local/lib/libQt6Widgets.so.2.1
#9  0x00000e8c54d58cc5 in QAbstractButtonPrivate::click() () from /usr/local/lib/libQt6Widgets.so.2.1
#10 0x00000e8c54d5a128 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#11 0x00000e8c54c8af59 in QWidget::event(QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#12 0x00000e8c54e212bb in QPushButton::event(QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#13 0x00000e8c54c344ba in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#14 0x00000e8c54c369fd in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#15 0x00000e8d1cd8a84d in Gui::GUIApplication::notify(QObject*, QEvent*) () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#16 0x00000e8cf4bb6e76 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/libQt6Core.so.4.1
#17 0x00000e8c54c34b66 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) ()
   from /usr/local/lib/libQt6Widgets.so.2.1
#18 0x00000e8c54ca3673 in QWidgetWindow::handleMouseEvent(QMouseEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#19 0x00000e8c54ca27a2 in QWidgetWindow::event(QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#20 0x00000e8c54c344ba in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#21 0x00000e8c54c35414 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#22 0x00000e8d1cd8a84d in Gui::GUIApplication::notify(QObject*, QEvent*) () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#23 0x00000e8cf4bb6e76 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/libQt6Core.so.4.1
#24 0x00000e8c81246d4f in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () from /usr/local/lib/libQt6Gui.so.6.0
#25 0x00000e8c812b0b5b in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Gui.so.6.0
#26 0x00000e8d38308fc3 in xcbSourceDispatch(_GSource*, int (*)(void*), void*) () from /usr/local/lib/qt6/plugins/platforms/../../../libQt6XcbQpa.so.3.0
#27 0x00000e8cbf8f011f in g_main_context_dispatch_unlocked () from /usr/local/lib/libglib-2.0.so.4201.12
#28 0x00000e8cbf8f050d in g_main_context_iterate_unlocked () from /usr/local/lib/libglib-2.0.so.4201.12
#29 0x00000e8cbf8f05bb in g_main_context_iteration () from /usr/local/lib/libglib-2.0.so.4201.12
#30 0x00000e8cf4e3f7e2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Core.so.4.1
#31 0x00000e8cf4bc1385 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Core.so.4.1
#32 0x00000e8cf4bb7613 in QCoreApplication::exec() () from /usr/local/lib/libQt6Core.so.4.1
#33 0x00000e8d1cca95d4 in Gui::Application::runApplication() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#34 0x00000e8a4e98b6bc in main ()
(gdb) quit
</pre>
</details>

[image]

This only seems to happen when I use the _Done_ (2) button from the first start widget. If I close the _Start_ tab (1) I'm able to use FreeCAD just fine (based on my tests so far) and change the same settings (Language, Theme, Units, etc.) from Edit  -> Preferences without a problem. Since it might be related, I tried to reproduce #16116 but so far I haven't been able to. It seems to work just fine on my side.

```
OS: OpenBSD 7.6
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.Unknown
Build type: Release
Python 3.11.9, Qt 6.6.3, Coin 4.0.2, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
```

luzpaz, 2024-09-06:
@wwmayer could you weigh-in on this when you get a moment ?

wwmayer, 2024-09-07:
I don't know what exactly is causing the crash. I tried the described workflow on Linux but there it doesn't crash.
But according to the call stack it looks like a focus problem and there it often leads to problems if the widget with the focus is made hidden by user interactions.

Those you get the crash could try to explicitly use a queued connection in `StartView::StartView`:
```
    connect(firstStartWidget,
            &FirstStartWidget::dismissed,
            this,
            &StartView::firstStartWidgetDismissed,
            Qt::QueuedConnection);
```

thyssentishman, 2024-09-08:
> I don't know what exactly is causing the crash. I tried the described workflow on Linux but there it doesn't crash. But according to the call stack it looks like a focus problem and there it often [...]

Assuming that the following diff is what you meant, then I can confirm that this didn't make a difference on OpenBSD (same behavior as before).

```diff
Index: src/Mod/Start/Gui/StartView.cpp
--- src/Mod/Start/Gui/StartView.cpp.orig
+++ src/Mod/Start/Gui/StartView.cpp
@@ -123,7 +123,8 @@ StartView::StartView(QWidget* parent)
     connect(firstStartWidget,
             &FirstStartWidget::dismissed,
             this,
-            &StartView::firstStartWidgetDismissed);
+            &StartView::firstStartWidgetDismissed,
+            Qt::QueuedConnection);
     firstStartRegion->addWidget(firstStartWidget);
     firstStartRegion->addStretch();
     _contents->addWidget(firstStartScrollArea);

```

Here the backtrace using this patch: <details>
<summary>Backtrace</summary><pre>
Reading symbols from /usr/local/lib/freecad/bin/FreeCAD...
[New process 523370]
[New process 146189]
[New process 304548]
[New process 269861]
[New process 410608]
[New process 269506]
[New process 222502]
[New process 263211]
[New process 396116]
[New process 169331]
[New process 553731]
[New process 505727]
[New process 151919]
[New process 351037]
[New process 204834]
[New process 150289]
[New process 268619]
[New process 116587]
[New process 598793]
Core was generated by `FreeCAD'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000008ad9deed566 in QScrollArea::ensureWidgetVisible(QWidget*, int, int) ()
   from /usr/local/lib/libQt6Widgets.so.2.1
[Current thread is 1 (process 523370)]
(gdb) bt
#0  0x000008ad9deed566 in QScrollArea::ensureWidgetVisible(QWidget*, int, int) () from /usr/local/lib/libQt6Widgets.so.2.1
#1  0x000008ad9deed509 in QScrollArea::focusNextPrevChild(bool) () from /usr/local/lib/libQt6Widgets.so.2.1
#2  0x000008ad9dd4bdc6 in QWidgetPrivate::hide_helper() () from /usr/local/lib/libQt6Widgets.so.2.1
#3  0x000008ad9dd4c1d2 in QWidgetPrivate::setVisible(bool) () from /usr/local/lib/libQt6Widgets.so.2.1
#4  0x000008ad9dd2ffa5 in QStackedLayout::setCurrentIndex(int) () from /usr/local/lib/libQt6Widgets.so.2.1
#5  0x000008ade6617dfb in StartGui::StartView::firstStartWidgetDismissed() () from /usr/local/lib/freecad/lib/StartGui.so
#6  0x000008ad3000ff9b in QObject::event(QEvent*) () from /usr/local/lib/libQt6Core.so.4.1
#7  0x000008ad9dd4d5d6 in QWidget::event(QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#8  0x000008ad9de95b93 in QMainWindow::event(QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#9  0x000008ad9dcf64ba in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#10 0x000008ad9dcf7414 in QApplication::notify(QObject*, QEvent*) () from /usr/local/lib/libQt6Widgets.so.2.1
#11 0x000008ad23674e5d in Gui::GUIApplication::notify(QObject*, QEvent*) () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#12 0x000008ad2ffc0e76 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /usr/local/lib/libQt6Core.so.4.1
#13 0x000008ad2ffc1f80 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/local/lib/libQt6Core.so.4.1
#14 0x000008ad30249f19 in postEventSourceDispatch(_GSource*, int (*)(void*), void*) () from /usr/local/lib/libQt6Core.so.4.1
#15 0x000008ad05c6311f in g_main_context_dispatch_unlocked () from /usr/local/lib/libglib-2.0.so.4201.12
#16 0x000008ad05c6350d in g_main_context_iterate_unlocked () from /usr/local/lib/libglib-2.0.so.4201.12
#17 0x000008ad05c635bb in g_main_context_iteration () from /usr/local/lib/libglib-2.0.so.4201.12
#18 0x000008ad302497e2 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Core.so.4.1
#19 0x000008ad2ffcb385 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/local/lib/libQt6Core.so.4.1
#20 0x000008ad2ffc1613 in QCoreApplication::exec() () from /usr/local/lib/libQt6Core.so.4.1
#21 0x000008ad23593be4 in Gui::Application::runApplication() () from /usr/local/lib/freecad/lib/libFreeCADGui.so
#22 0x000008aaff1c06bc in main ()
(gdb) quit
</pre>
</details>

outpaddling, 2024-09-09:
> I don't know what exactly is causing the crash. I tried the described workflow on Linux but there it doesn't crash. But according to the call stack it looks like a focus problem and there it often [...]

Didn't help on FreeBSD either.

<details>
<summary>Backtrace</summary>
FreeBSD coral.acadix  bacon ~ 1001: lldb -c FreeCAD.core /usr/local/bin/FreeCAD
(lldb) target create "/usr/local/bin/FreeCAD" --core "FreeCAD.core"
Core file '/home/bacon/FreeCAD.core' (x86_64) was loaded.
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV
  * frame #0: 0x0000000837adb313 libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) + 51
    frame #1: 0x0000000837adb2d7 libQt6Widgets.so.6`QScrollArea::focusNextPrevChild(bool) + 55
    frame #2: 0x000000083794f516 libQt6Widgets.so.6`QWidgetPrivate::hide_helper() + 566
    frame #3: 0x00000008379503e9 libQt6Widgets.so.6`QWidgetPrivate::setVisible(bool) + 969
    frame #4: 0x000000083794fd96 libQt6Widgets.so.6`QWidget::setVisible(bool) + 662
    frame #5: 0x0000000837935140 libQt6Widgets.so.6`QStackedLayout::setCurrentIndex(int) + 384
    frame #6: 0x00000008e3cdf70c StartGui.so`StartGui::StartView::firstStartWidgetDismissed(this=0x00003e8c539c5880) at StartView.cpp:470:16
    frame #7: 0x00000008e3ce6d62 StartGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (StartGui::StartView::*)()>::call(f=90 f6 cd e3 08 00 00 00 00 00 00 00 00 00 00 00, o=0x00003e8c539c5880, arg=0x00003e8c4fc731c8) at qobjectdefs_impl.h:145:13
    frame #8: 0x00000008e3ce6cad StartGui.so`void QtPrivate::FunctionPointer<void (StartGui::StartView::*)()>::call<QtPrivate::List<>, void>(f=90 f6 cd e3 08 00 00 00 00 00 00 00 00 00 00 00, o=0x00003e8c539c5880, arg=0x00003e8c4fc731c8) at qobjectdefs_impl.h:182:13
    frame #9: 0x00000008e3ce6bd5 StartGui.so`QtPrivate::QCallableObject<void (StartGui::StartView::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x00003e8c4ef21520, r=0x00003e8c539c5880, a=0x00003e8c4fc731c8, ret=0x0000000000000000) at qobjectdefs_impl.h:553:21
    frame #10: 0x000000083a168553 libQt6Core.so.6`QObject::event(QEvent*) + 611
    frame #11: 0x0000000837951911 libQt6Widgets.so.6`QWidget::event(QEvent*) + 3601
    frame #12: 0x0000000837a88f12 libQt6Widgets.so.6`QMainWindow::event(QEvent*) + 786
    frame #13: 0x00000008378fee88 libQt6Widgets.so.6`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 248
    frame #14: 0x00000008378ffdd9 libQt6Widgets.so.6`QApplication::notify(QObject*, QEvent*) + 553
    frame #15: 0x00000008264c3e3f libFreeCADGui.so`Gui::GUIApplication::notify(this=0x0000000820919d60, receiver=0x00003e8c539c5880, event=0x00003e8c4fc73180) at GuiApplication.cpp:83:34
    frame #16: 0x000000083a120654 libQt6Core.so.6`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 180
    frame #17: 0x000000083a121827 libQt6Core.so.6`QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 503
    frame #18: 0x000000083a2a8b88 libQt6Core.so.6`QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 72
    frame #19: 0x0000000861eae8d4 libQt6XcbQpa.so.6`___lldb_unnamed_symbol2727 + 20
    frame #20: 0x000000083a129b07 libQt6Core.so.6`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 519
    frame #21: 0x000000083a120e44 libQt6Core.so.6`QCoreApplication::exec() + 148
    frame #22: 0x000000082628b8ab libFreeCADGui.so`(anonymous namespace)::tryRunEventLoop(mainApp=0x0000000820919d60) at Application.cpp:2062:5
    frame #23: 0x0000000826288b33 libFreeCADGui.so`(anonymous namespace)::runEventLoop(mainApp=0x0000000820919d60) at Application.cpp:2082:9
    frame #24: 0x00000008262881ce libFreeCADGui.so`Gui::Application::runApplication() at Application.cpp:2159:5
    frame #25: 0x000000000022cfda FreeCAD`main(argc=1, argv=0x000000082091aae8) at MainGui.cpp:300:13
    frame #26: 0x000000083ef37afa libc.so.7`__libc_start1 + 298
    frame #27: 0x000000000022ad90 FreeCAD`_start at crt1_s.S:83
</details>

xtemp09, 2024-09-09:
Nothing in the code can cause this problem. Compilation with some options, like _ASAN_, _LSAN_, _TSAN_, `-fstack-protector-strong`, `-fsanitize=safe-stack` and all other ones can.

@outpaddling, @thyssentishman, how do you compile _Qt_ and _FreeCAD_? I read this:[^1]

> These features make program bugs easier to detect and harder to exploit: instead of memory being corrupted or an invalid access being ignored, they often result in a [segmentation fault](https://en.wikipedia.org/wiki/SIGSEGV) and abortion of the process.

[^1]: https://en.wikipedia.org/wiki/OpenBSD_security_features#Memory_protection

outpaddling, 2024-09-09:
The backtrace points to StartView.cpp:470, where FreeCAD calls `setCurrentIndex(1)`.  Just asking the obvious as a first step: Can we be sure that '1' is a valid index at this time?  Maybe something has not been allocated or initialized.

```
void StartView::firstStartWidgetDismissed()
{
    auto hGrp = App::GetApplication().GetParameterGroupByPath(
        "User parameter:BaseApp/Preferences/Mod/Start");
    hGrp->SetBool("FirstStart2024", false);
    _contents->setCurrentIndex(1);
}
```

xtemp09, 2024-09-10:
> Is this not being used in the Linux build?

Stack protection is not used.

> Just asking the obvious as a first step: Can we be sure that '1' is a valid index at this time? Maybe something has not been allocated or initialized.

_Linux_, _Windows_ and _MacOS_ users don't get crash, so, it is valid.

Can you test _FreeCAD_ without stack protection? I noticed that you have compiled using `-g`, while not using, for example, `-Og` or `-O2` option. Can you compile _FreeCAD_ `-fstack-protector-strong`lessly with `-g -Og` options?

outpaddling, 2024-09-10:
> > Is this not being used in the Linux build?
> 
> Stack protection is not used.
> 
> > Just asking the obvious as a first step: Can we be sure that '1' is a valid index at this time? Maybe something has not been allocated or initialized.
> 
> _Linux_, _Windows_ and _MacOS_ users don't get crash, so, it is valid.

That doesn't necessarily follow.  If something is not initialized properly, there could be a stray pointer or subscript that lands on a permitted memory address on some platforms, while causing a segfault on others.  Depending where the pointer lands, it could corrupt other variables, or have no consequence at all (e.g. if it lands on an address that will be overwritten with valid data before it is read again).   I see this all the time when testing new code on various platforms.

> Can you test _FreeCAD_ without stack protection? I noticed that you have compiled using `-g`, while not using, for example, `-Og` or `-O2` option. Can you compile _FreeCAD_ `-fstack-protector-strong`lessly with `-g -Og` options?

I originally was not using -g.  I added it in order to get more info out of the backtrace.  Adding debug info did not change the behavior.  

-Og appears to be a GCC thing.  It's not described in the clang reference: [https://clang.llvm.org/docs/ClangCommandLineReference.html](https://clang.llvm.org/docs/ClangCommandLineReference.html).

I can try a build without stack protection, but disabling it in a committed FreeBSD port if not considered a valid fix, so it would only be to collect a data point.

Can you try a build on Linux with stack protection to see if you can reproduce the crash?

thyssentishman, 2024-09-10:
> Nothing in the code can cause this problem. Compilation with some options, like _ASAN_, _LSAN_, _TSAN_, `-fstack-protector-strong`, `-fsanitize=safe-stack` and all other ones can.
> 
> @outpaddling, @thyssentishman, how do you compile _Qt_ and _FreeCAD_? 

The flags  `-fstack-protector-strong`, `-fsanitize=safe-stack` don't seem to be used on OpenBSD by default:
```
[824/6238] /usr/ports/pobj/freecad-1.0/bin/c++ -DBOOST_PP_VARIADICS=1 -DDOCDIR=\"/usr/local/share/freecad/doc\" -DFMT_SHARED -DFreeCADApp_EXPORTS -DHAVE_CONFIG_H -DHAVE_FREETYPE -DHAVE_OPENGL_EXT -DHAVE_TK -DHAVE_XLIB -DLIBRARYDIR=\"/usr/local/lib/freecad/lib\" -DOCC_CONVERT_SIGNALS -DPYCXX_6_2_COMPATIBILITY -DQT_CORE_LIB -DQT_NO_DEBUG -DQT_NO_KEYWORDS -DQT_XML_LIB -DRESOURCEDIR=\"/usr/local/share/freecad\" -D_OCC64 -I/usr/ports/pobj/freecad-1.0/build-amd64/src/App/FreeCADApp_autogen/include -I/usr/ports/pobj/freecad-1.0/build-amd64 -I/usr/ports/pobj/freecad-1.0/build-amd64/src -I/usr/ports/pobj/freecad-1.0/FreeCAD-fe40748c10503eed367840239db3e0f81873e359/src -I/usr/ports/pobj/freecad-1.0/build-amd64/src/App -I/usr/local/include/python3.11 -isystem /usr/local/include -isystem /usr/local/include/X11/qt6/QtCore -isystem /usr/local/include/X11/qt6 -isystem /usr/local/include/X11/qt6/QtXml -isystem /usr/local/lib/qt6/mkspecs/openbsd-clang -Wall -Wextra -Wpedantic -Wno-write-strings -O2 -pipe -Wno-undefined-var-template -DNDEBUG -std=gnu++17 -fPIC -fno-direct-access-external-data -MD -MT src/App/CMakeFiles/FreeCADApp.dir/Application.cpp.o -MF src/App/CMakeFiles/FreeCADApp.dir/Application.cpp.o.d -o src/App/CMakeFiles/FreeCADApp.dir/Application.cpp.o -c /usr/ports/pobj/freecad-1.0/FreeCAD-fe40748c10503eed367840239db3e0f81873e359/src/App/Application.cpp
```
From what I can see in the Qt6 ports on OpenBSD, they also don't add any more compilation flags than the default. See [here](https://github.com/openbsd/ports/blob/ae55e627ce438db2c7e77b68096a49daee30445e/x11/qt6/Makefile.inc#L38).

> I read this:[1](#user-content-fn-1-cb7ddd11201baa54a76c8d1d3c500499)
> 
> > These features make program bugs easier to detect and harder to exploit: instead of memory being corrupted or an invalid access being ignored, they often result in a [segmentation fault](https://en.wikipedia.org/wiki/SIGSEGV) and abortion of the process.

If this was the cause for the segmentation fault we are having, then the code **IS** what's causing the problem.

xtemp09, 2024-09-10:
> -Og appears to be a GCC thing. It's not described in the clang reference:

https://clang.llvm.org/docs/CommandGuide/clang.html#code-generation-options

> Can you try a build on Linux with stack protection to see if you can reproduce the crash?

I'll try, but unreproducibility of the crash will not indicate anything.

----

> If this was the cause for the segmentation fault we are having, then the code IS what's causing the problem.

I'd like to point out again that users of the supported platforms cannot reproduce this crash. Maybe the problem is caused by the code, maybe not.

xtemp09, 2024-09-10:
I cannot reproduce the crash.

<details>

```
$ clang-19 --version
Ubuntu clang version 19.1.0 (++20240903024228+0c641568515a-1~exp1~20240903144250.30)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/lib/llvm-19/bin
```

```
$ cmake ../freecad-source -G Ninja \
-D CMAKE_BUILD_TYPE=Debug \
-D CMAKE_INSTALL_PREFIX=/home/xtemp09/fc-test/ \
\
-D CMAKE_C_COMPILER=clang-19 -D CMAKE_CXX_COMPILER=clang++-19      \
-D CMAKE_C_FLAGS_DEBUG="-g -Og -fstack-protector-strong -fno-omit-frame-pointer -Wno-enum-constexpr-conversion"   \
-D CMAKE_CXX_FLAGS_DEBUG="-g -Og -fstack-protector-strong -fno-omit-frame-pointer -Wno-enum-constexpr-conversion" \
\
-D CMAKE_EXE_LINKER_FLAGS="-fuse-ld=lld"    \
-D CMAKE_SHARED_LINKER_FLAGS="-fuse-ld=lld" \
\
-D CMAKE_AR=/usr/bin/llvm-ar-19               \
-D CMAKE_ADDR2LINE=/usr/bin/llvm-addr2line-19 \
-D CMAKE_MT=/usr/bin/llvm-mt-19               \
-D CMAKE_NM=/usr/bin/llvm-nm-19               \
-D CMAKE_OBJCOPY=/usr/bin/llvm-objcopy-19     \
-D CMAKE_OBJDUMP=/usr/bin/llvm-objdump-19     \
-D CMAKE_RANLIB=/usr/bin/llvm-ranlib-19       \
-D CMAKE_READELF=/usr/bin/llvm-readelf-19     \
-D CMAKE_RC_COMPILER=/usr/bin/llvm-rc-19      \
-D CMAKE_STRIP=/usr/bin/llvm-strip-19
```

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38689 (Git)
Build type: Debug
Branch: main
Hash: 3503d677a91c98672b80c2b312593440fb3da0db
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: //
```

</details>

outpaddling, 2024-09-10:
> I cannot reproduce the crash.

It was worth a shot, thanks for trying.  I'm a little swamped right now, but I'll try and find time soon to dig into the code, maybe get some better trace info out of Qt.  In theory it could also be a Qt issue, but that seems unlikely given that the same problem occurs with 3 stable Qt builds (OpenBSD ports, FreeBSD ports Qt5 and FreeBSD ports Qt6).  We'll solve this eventually.

luzpaz, 2024-09-10:
Also, are we the only project encountering this issue? (i wonder)

outpaddling, 2024-09-11:
> Also, are we the only project encountering this issue? (i wonder)

I would say "probably" based on FreeBSD problem reports: [https://bugs.freebsd.org/bugzilla/query.cgi?format=advanced](https://bugs.freebsd.org/bugzilla/query.cgi?format=advanced)

There are very few open PRs for Qt packages, and none that I could find blaming Qt for a crash.  That with ~1300 ports that use Qt.

Looking at the limited info from Qt in the tail end of the backtrace, it seems like the only thing that could be causing a stray memory reference is the QWidget * passed to ensureWidgetVisible(), or the QScrollArea object itself, unless there's something else being referenced by ensureWidgetVisible() that's not in the argument list.  All other arguments are ints and bools passed by value.  I would guess there's something corrupt the widget or the QScrollArea object.

```
  * frame #0: 0x0000000837adb313 libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) + 51
    frame #1: 0x0000000837adb2d7 libQt6Widgets.so.6`QScrollArea::focusNextPrevChild(bool) + 55
    frame #2: 0x000000083794f516 libQt6Widgets.so.6`QWidgetPrivate::hide_helper() + 566
    frame #3: 0x00000008379503e9 libQt6Widgets.so.6`QWidgetPrivate::setVisible(bool) + 969
    frame #4: 0x000000083794fd96 libQt6Widgets.so.6`QWidget::setVisible(bool) + 662
    frame #5: 0x0000000837935140 libQt6Widgets.so.6`QStackedLayout::setCurrentIndex(int) + 384
    frame #6: 0x00000008e3cdf70c StartGui.so`StartGui::StartView::firstStartWidgetDismissed(this=0x00003e8c539c5880) at StartView.cpp:470:16
    frame #7: 0x00000008e3ce6d62 StartGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (StartGui::StartView::*)()>::call(f=90 f6 cd e3 08 00 00 00 00 00 00 00 00 00 00 00, o=0x00003e8c539c5880, arg=0x00003e8c4fc731c8) at qobjectdefs_impl.h:145:13
    frame #8: 0x00000008e3ce6cad StartGui.so`void QtPrivate::FunctionPointer<void (StartGui::StartView::*)()>::call<QtPrivate::List<>, void>(f=90 f6 cd e3 08 00 00 00 00 00 00 00 00 00 00 00, o=0x00003e8c539c5880, arg=0x00003e8c4fc731c8) at qobjectdefs_impl.h:182:13
```

outpaddling, 2024-09-11:
I rebuilt qt6-base with debug info and got a more informative backtrace:

<details>
<sumary>Backtrace with debug info from Qt</summary>

```
(lldb) target create "/usr/local/bin/FreeCAD" --core "FreeCAD.core"
Core file '/home/bacon/FreeCAD.core' (x86_64) was loaded.
(lldb) bt
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV
  * frame #0: 0x0000000839331e13 libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) [inlined] int std::__1::__cxx_atomic_load[abi:v160006]<int>(__a=0x0000000000000004, __order=memory_order_relaxed) at atomic:963:12
    frame #1: 0x0000000839331e13 libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) [inlined] std::__1::__atomic_base<int, false>::load[abi:v160006](this=0x0000000000000004, __m=memory_order_relaxed) const at atomic:1551:17                                                                            
    frame #2: 0x0000000839331e13 libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) [inlined] int QAtomicOps<int>::loadRelaxed<int>(_q_value=0x0000000000000004) at qatomic_cxx11.h:202:25
    frame #3: 0x0000000839331e13 libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) [inlined] QBasicAtomicInteger<int>::loadRelaxed(this=0x0000000000000004) const at qbasicatomic.h:36:45
    frame #4: 0x0000000839331e13 libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) [inlined] QWeakPointer<QObject>::internalData(this=0x0000361051e818e0) const at qsharedpointer_impl.h:752:45
    frame #5: 0x0000000839331e0c libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) [inlined] QPointer<QWidget>::data(this=0x0000361051e818e0) const at qpointer.h:74:33
    frame #6: 0x0000000839331e0c libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(QWidget*, int, int) [inlined] QPointer<QWidget>::operator->(this=0x0000361051e818e0) const at qpointer.h:78:14
    frame #7: 0x0000000839331e0c libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(this=0x000036104f80add0, childWidget=0x00003610533ce8f0, xmargin=50, ymargin=50) at qscrollarea.cpp:450:10
    frame #8: 0x0000000839331dd7 libQt6Widgets.so.6`QScrollArea::focusNextPrevChild(this=0x000036104f80add0, next=<unavailable>) at qscrollarea.cpp:404:13
    frame #9: 0x000000083917dea6 libQt6Widgets.so.6`QWidgetPrivate::hide_helper(this=0x0000361051e81600) at qwidget.cpp:8280:20
    frame #10: 0x000000083917ea34 libQt6Widgets.so.6`QWidgetPrivate::setVisible(this=0x0000361051e81600, visible=false) at qwidget.cpp:8463:13
    frame #11: 0x000000083917e584 libQt6Widgets.so.6`QWidget::setVisible(this=0x000036104f80add0, visible=false) at qwidget.cpp:8371:8
    frame #12: 0x0000000839162525 libQt6Widgets.so.6`QStackedLayout::setCurrentIndex(this=0x000036104eb35fe0, index=1) at qstackedlayout.cpp:284:19
    frame #13: 0x00000008e267494c StartGui.so`StartGui::StartView::firstStartWidgetDismissed(this=0x0000361053364e00) at StartView.cpp:469:16
    frame #14: 0x00000008e267bfa2 StartGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (StartGui::StartView::*)()>::call(f=d0 48 67 e2 08 00 00 00 00 00 00 00 00 00 00 00, o=0x0000361053364e00, arg=0x0000000820c15358) at qobjectdefs_impl.h:145:13
    frame #15: 0x00000008e267beed StartGui.so`void QtPrivate::FunctionPointer<void (StartGui::StartView::*)()>::call<QtPrivate::List<>, void>(f=d0 48 67 e2 08 00 00 00 00 00 00 00 00 00 00 00, o=0x0000361053364e00, arg=0x0000000820c15358) at qobjectdefs_impl.h:182:13
    frame #16: 0x00000008e267be15 StartGui.so`QtPrivate::QCallableObject<void (StartGui::StartView::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x000036104eb3a4c0, r=0x0000361053364e00, a=0x0000000820c15358, ret=0x0000000000000000) at qobjectdefs_impl.h:553:21
    frame #17: 0x000000083dfb5f84 libQt6Core.so.6`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(this=0x000036104eb3a4c0, r=0x0000361053364e00, a=0x0000000820c15358) at qobjectdefs_impl.h:469:51
    frame #18: 0x000000083dfb5f6e libQt6Core.so.6`void doActivate<false>(sender=0x0000361053358fb0, signal_index=10, argv=0x0000000820c15358) at qobject.cpp:4086:26                                                                            
    frame #19: 0x00000008e2635bc2 StartGui.so`StartGui::FirstStartWidget::dismissed(this=0x0000361053358fb0) at moc_FirstStartWidget.cpp:139:5
    frame #20: 0x00000008e266ad72 StartGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (StartGui::FirstStartWidget::*)()>::call(f=a0 5b 63 e2 08 00 00 00 00 00 00 00 00 00 00 00, o=0x0000361053358fb0, arg=0x0000000820c155c0) at qobjectdefs_impl.h:145:13
    frame #21: 0x00000008e266acbd StartGui.so`void QtPrivate::FunctionPointer<void (StartGui::FirstStartWidget::*)()>::call<QtPrivate::List<>, void>(f=a0 5b 63 e2 08 00 00 00 00 00 00 00 00 00 00 00, o=0x0000361053358fb0, arg=0x0000000820c155c0) at qobjectdefs_impl.h:182:13
    frame #22: 0x00000008e266abe5 StartGui.so`QtPrivate::QCallableObject<void (StartGui::FirstStartWidget::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x000036104eb6d8a0, r=0x0000361053358fb0, a=0x0000000820c155c0, ret=0x0000000000000000) at qobjectdefs_impl.h:553:21
    frame #23: 0x000000083dfb5f84 libQt6Core.so.6`void doActivate<false>(QObject*, int, void**) [inlined] QtPrivate::QSlotObjectBase::call(this=0x000036104eb6d8a0, r=0x0000361053358fb0, a=0x0000000820c155c0) at qobjectdefs_impl.h:469:51
[... cut ...]
    frame #47: 0x00000008277220fb libFreeCADGui.so`(anonymous namespace)::tryRunEventLoop(mainApp=0x0000000820c16ef0) at Application.cpp:2172:5
    frame #48: 0x000000082771f383 libFreeCADGui.so`(anonymous namespace)::runEventLoop(mainApp=0x0000000820c16ef0) at Application.cpp:2192:9
    frame #49: 0x000000082771ea1e libFreeCADGui.so`Gui::Application::runApplication() at Application.cpp:2269:5
    frame #50: 0x000000000022d086 FreeCAD`main(argc=1, argv=0x0000000820c17ca0) at MainGui.cpp:301:13
    frame #51: 0x00000008413f3afa libc.so.7`__libc_start1 + 298
    frame #52: 0x000000000022adb0 FreeCAD`_start at crt1_s.S:83
```
</details>

wwmayer, 2024-09-11:
`frame #7: 0x0000000839331e0c libQt6Widgets.so.6`QScrollArea::ensureWidgetVisible(this=0x000036104f80add0, childWidget=0x00003610533ce8f0, xmargin=50, ymargin=50) at qscrollarea.cpp:450:10
`
At the line 450 at qscrollarea.cpp we have this code:
```
void QScrollArea::ensureWidgetVisible(QWidget *childWidget, int xmargin, int ymargin)
{
    Q_D(QScrollArea);
    if (!d->widget->isAncestorOf(childWidget))
        return;
...
}
```
where `d->widget` which is of type `QPointer<QWidget>` is de-referenced. When looking through the code of qscrollarea.cpp then `d->widget` is only set if `QScrollArea::setWidget` is called.

In `StartView::StartView` you will find the variable `firstStartScrollArea` for which `setWidget` is never used so that the call of `ensureWidgetVisible` leads to the crash. But it's also a bit weird that Qt doesn't check for a possible null pointer.

Anyway try to replace the corresponding block inside `StartView::StartView` with:
```
    // First start page
    auto firstStartScrollArea = gsl::owner<QScrollArea*>(new QScrollArea());
    auto firstStartScrollWidget = gsl::owner<QWidget*>(new QWidget(firstStartScrollArea));
    firstStartScrollArea->setWidget(firstStartScrollWidget);
    firstStartScrollArea->setWidgetResizable(true);

    auto firstStartRegion = gsl::owner<QHBoxLayout*>(new QHBoxLayout(firstStartScrollWidget));
    firstStartRegion->addStretch();
    auto firstStartWidget = gsl::owner<FirstStartWidget*>(new FirstStartWidget(this));
    connect(firstStartWidget,
            &FirstStartWidget::dismissed,
            this,
            &StartView::firstStartWidgetDismissed);
    firstStartRegion->addWidget(firstStartWidget);
    firstStartRegion->addStretch();
    _contents->addWidget(firstStartScrollArea);
```

wwmayer, 2024-09-11:
When debugging `QScrollArea::ensureWidgetVisible` under Windows then there `d->widget` is also a null pointer but de-referencing it and calling `isAncestorOf` doesn't cause a crash because the implementation of this method doesn't access a member variable:
```
bool QWidget::isAncestorOf(const QWidget *child) const
{
    while (child) {
        if (child == this)
            return true;
        if (child->isWindow())
            return false;
        child = child->parentWidget();
    }
    return false;
} 
```
This explains why on many systems it doesn't  crash but nevertheless it's UB.

thyssentishman, 2024-09-11:
Nice find @wwmayer ! I can confirm that with the suggested changes to `src/Mod/Start/Gui/StartView.cpp`, FreeCAD doesn't crash anymore after closing the first-start widget. Thank you very much!

outpaddling, 2024-09-11:
That seems to have fixed the issue on FreeBSD as well.  Nice work!
I also see we're on RC1 now, so good timing...
I think this issue can be closed once the patch is committed.

closed 2024-09-12 by wwmayer

outpaddling, 2024-09-12:
Incidentally, if I build Qt with debug info and no optimizations, FreeCAD doesn't crash.  I had to build with `-g` and `-O2` in order to get the detailed backtrace.  Just goes to show how sensitive bugs like this can be to the build env.

PR #16453 (2024-09-12): StartPage: Make sure to pass a container widget to the QScrollArea
Fixes #16047: FreeBSD: Segfault in StartView::firstStartWidgetDismissed
commit 0946c688 (2024-09-12): StartPage: Make sure to pass a container widget to the QScrollArea

Fixes #16047: FreeBSD: Segfault in StartView::firstStartWidgetDismissed
commit e7902003 (2024-09-12): StartPage: Make sure to pass a container widget to the QScrollArea

Fixes #16047: FreeBSD: Segfault in StartView::firstStartWidgetDismissed