FreeCAD/FreeCAD #16052: CAM: SimulatorGL helix error only half the pocket is produced

sparker0594, 2024-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When Simulating a helix pocket, the SimulatorGL tool only produced half the helix in the animation.
The 'old' Simulator tool produces the correct full helix pocket. 
Example file attached.

Example of SimulatorGL:
[image]

Example of 'old' Simulator:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * CurvedShapes 1.0.9
  * Curves 0.6.45
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

CAM

### Anything else?

[Pull Start Adapter Plate.zip](https://github.com/user-attachments/files/16740613/Pull.Start.Adapter.Plate.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-25:
@shaise FYI

shaise, 2024-08-25:
Thanks! I will investigate.
@luzpaz , how do I assign this to me?

shaise, 2024-08-28:
This issue should be fixed once pr #16118 is merged

closed 2024-09-02 by sliptonic