FreeCAD/FreeCAD #16053: Std_Measure not working for Draft_ArrayTools (Draft workbench)

johndoe2323, 2024-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Preparation:
1. Insert cube in part workbench, default cube has 10mm width/length/height
2. Create a linear array in Draft workbench, e.g. z-intervall of 11mm intervall thus leaving 1mm gap

Try to measure two edges of different cubes in mode Distance, should be 1mm but gives 10mm.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Installed mods: 
  * Assembly4 0.50.4
  * OpenDark 2023.10.30
  * ProDarkThemePreferencePack 1.0.0
  * Woodworking 0.21.30486
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-25:
Ok, I can confirm this in the newest weekly build. The position of measurement lines is wrong too.

[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

furgo16, 2024-09-21:
https://github.com/FreeCAD/FreeCAD/pull/16251 was merged, but this is still reproduceable on:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38794 (Git) Snap 1144
Build type: Release
Branch: main
Hash: a47800d9ccd7d7a7cdb1779d56de03d3c64cd76e
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Stylesheet/Theme/QtStyle: OpenLight.qss/OpenLight/Qt default
Installed mods: 
  * OpenTheme 2024.9.1
  * ezydxf
```

johndoe2323, 2024-09-22:
Can confirm, still reproducable in 1.1.0dev.38794

Roy-043, 2026-02-13:
This has indeed been fixed. Thanks for reporting.