FreeCAD/FreeCAD #16054: Std_measure not updating when positions are changing

johndoe2323, 2024-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Insert cube in Part Workbench (e.g. standard width/length/height of 10mm)
2. Insert another cube in Part Workbench, position it so that there is a gap between the cubes (e.g. z-position of 12mm)
3. With Std_measure, measure the distance of two edges of the two cubes (e.g. 2mm)
4. Change the position of the cubes

Issue: the measurement auto-updates but not the measurement geometry/lines.

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38261 (Git)
Build type: Release
Branch: main
Hash: 51789ae8a1a0b1ed20317eaedcd190baac734e3d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Austria (de_AT)
Installed mods: 
  * Assembly4 0.50.4
  * OpenDark 2023.10.30
  * ProDarkThemePreferencePack 1.0.0
  * Woodworking 0.21.30486
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

johndoe2323, 2024-08-25:
[image]

FEA-eng, 2024-08-25:
Please try with the newest weekly build. This was fixed recently.

johndoe2323, 2024-08-25:
Gosh you are right - I was using the wrong build! Works in 38506!

closed 2024-08-25 by johndoe2323