FreeCAD/FreeCAD #16055: Assembly: Cylindrical joints break when moving (dragging) parts

sparker0594, 2024-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Cylindrical joints in the assembly can break when moving parts of the assembly.
The attached example file contains several cylindrical joints on three assembly parts. Initially, movement of the wheel centre will be normal and the concentric alignment will be maintained. Though moving the wheel centre further distances, or moving it rapidly will beak the control alignment and the parts will rotate. If movement is continued after the alignment break, some errors can be generated and also the fixed part can end up moving/rotating.

Example:
[image]

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * CurvedShapes 1.0.9
  * Curves 0.6.45
  * sheetmetal 0.4.24
```

### Subproject(s) affected?

Assembly

### Anything else?

Some of the error messages (may not be helpful):
16:07:14  Unhandled std exception in ViewProvider::eventCallback: invalid vector subscript
(Event type: SoLocation2Event, object type: AssemblyGui::ViewProviderAssembly)
16:07:15  Unhandled std exception in ViewProvider::eventCallback: invalid vector subscript
(Event type: SoMouseButtonEvent, object type: AssemblyGui::ViewProviderAssembly)

FreeCAD File: 
[LawnMower Wheel V2 (Assembly Git Issue).zip](https://github.com/user-attachments/files/16740970/LawnMower.Wheel.V2.Assembly.Git.Issue.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-08-29:
@aiksiongkoh I think this may be on the solver side. Can you please check the sample file provided by the user and see if you can find the issue?

sparker0594, 2024-11-09:
Hi All, 

This is still an issue in 39141, very easy to break assemblies with cylindrical joints. Or for cylindrical joints to fail to create if the mating features a some distance away. (using above example)

Is this a blocker?

To me the assembly workbench/stability of assemblies is not ready yet.

OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.39141 (Git)
Build type: Release
Branch: main
Hash: 4bd80bdcbb97d6038bc7030aed9ecebf57f4e428
Python 3.11.10, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
  * Curves 0.6.47
  * sheetmetal 0.4.25

kwahoo2, 2025-07-16:
Hello,

I noticed there also instability when a pair of cylindrical joints is used, like in the attached example (two linear bearing connected with two rails):

[plasma-cutter-gh.zip](https://github.com/user-attachments/files/21264463/plasma-cutter-gh.zip)

https://github.com/user-attachments/assets/cb91c777-7847-4d30-bc00-ab05a01741a9

Interestingly, the small frame flips too, even if  I move the big one.

Behaves the same way in 1.0.1 and weekly:

OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.1.39285 (Git) Conda AppImage
/
OS: Fedora Linux 42 (KDE Plasma Desktop Edition) (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.42531 (Git) AppImage
Build date: 2025/07/11 16:00:50

Note: In 1.1dev I had to move subassembly with the transformation tool, otherwise dragging had been failing with:

```
21:33:14  260.437 <Assembly> AssemblyObject.cpp(160): Solve failed: vector::_M_range_check: __n (which is 69) >= this->size() (which is 69)
21:33:14  Unhandled std exception in ViewProvider::eventCallback: vector::_M_range_check: __n (which is 69) >= this->size() (which is 69)
(Event type: SoLocation2Event, object type: AssemblyGui::ViewProviderAssembly)
21:33:14  Unhandled std exception in ViewProvider::eventCallback: vector::_M_range_check: __n (which is 69) >= this->size() (which is 69)
(Event type: SoMouseButtonEvent, object type: AssemblyGui::ViewProviderAssembly)
```

aiksiongkoh, 2025-07-16:
@kwahoo2 Can we setup a Google Meet appointment for me to see this on your computer? I am in Mountain Time USA.

kwahoo2, 2026-02-22:
Reversing the reference order - first joint reference on the rail (fixed part) does not change much for me.
[double-cylinder-test.zip](https://github.com/user-attachments/files/25471246/double-cylinder-test.zip)

https://github.com/user-attachments/assets/cb569ad3-ea72-4fc7-abdf-eab556eda6b8

luzpaz, 2026-05-02:
@PaddleStroke any idea how to proceed ?