FreeCAD/FreeCAD #16059: Application Freezes When Using Box Selection

AnudaWewalage, 2024-08-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I was following the "Creating a Simple Part with Draft and Part Workbench" tutorial. However, every time I attempt to use the Box Selection tool during the process, the application stops responding. The issue occurs consistently and makes it impossible to proceed with the tutorial.

### Full version info

```shell
[code]
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2024-08-25:
Please try with 0.22 dev version (weekly builds: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds). And attach the file in which it occurs, if possible.

AnudaWewalage, 2024-09-01:
I am a beginner for this software. yesterday, when I opened freeCAD, some updates were run, and now the issue is solved. Thanks for your support.

luzpaz, 2024-09-02:
@AnudaWewalage we are very close to releasing v1.0  
Please the development version 0.22.x as we are not fixing bugs in 0.21.x anymore.

closed 2024-09-02 by luzpaz

AnudaWewalage, 2024-11-05:
Thank you for the update. 
Just one concern—the site still lists 0.21 as the stable version. It might be helpful to update that as well.

luzpaz, 2024-11-06:
0.21 is technically stable version. RC stands for Release Client. Due to FreeCAD's current unorthodox release cycle we're in this protracted release cycle. After 1.0 hopefully we'll have shorter release cycles.