FreeCAD/FreeCAD #16068: DXF importer (C++) in 0.22 does not import layer names, in 0.21 does

dominik-devops, 2024-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In the development version all objects/layers are imported as "Compound/Compound001..." when "Group layers into blocks" checked. Unlike in stable version 0.21, where it imports with the layer name, e. g. Flooring, Doors, Columns...

Related issue [13597](https://github.com/FreeCAD/FreeCAD/issues/13597) though I am not exactly sure what problem was being solved there. The symptoms of this issue are visible there, too.

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38516 (Git) Snap 1116
Build type: Release
Branch: main
Hash: 46a84da571d51b135e16ac4731baaf67c552d791
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Anything else?

[pabellon de barcelona.zip](https://github.com/user-attachments/files/16747982/pabellon.de.barcelona.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-26:
@Roy-043 FYI

yorikvanhavre, 2024-09-04:
An additional problem here is the options are used both by the old Python and the new C++ importers, with sometimes different results.

There are a few things to check here then:

* What makes it fail on first import but succeed on second one. Is it something that is present in the document, a module that wasn't loaded (Draft most probably), or something in the importer that stayed initialized?
* Why grouping does not work if "use layers" is disabled. grouping should then implicitly always enable "use layers" (even if not shown to the user)

I can look into this after 1.0 but pinging @kpemartin here too

kpemartin, 2024-09-05:
Regarding the "fails the first time" issue, I'm not sure in what manner the first try "fails" here.
I'm working on #16131 right now and I'm seeing a similar fails-at-first-then-works behaviour; in my case the "failure" is that none of the entities import (appear in the drawing) but the import completes without errors or messages. See that issue for more info.

kpemartin, 2024-09-11:
The error message mentioned should not occur either.
`Layer.execute` is an empty Python method consisting of a single `pass` statement, and should never fail.
I suspect this is a failure in the lazy importer while trying to load the "real" Layer.py file when `execute` is first called.

The `SystemError` exception is the same one that occurs in #16131 (though unrelated to that actual Issue) within the C++ call to `doc->addObject("Part::FeaturePython", ...)` while trying to load the `Part` module.

kpemartin, 2024-09-12:
Found it, nothing to do with my setup. details and fix to follow.