FreeCAD/FreeCAD #16072: Pad: Issue when changing pad from dimenson to two dimensions

maxwxyz, 2024-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When creating a pad, checking "symmetrical" and then changing type from dimension to two dimension, changing one of both dimensions still changes them both symmetrical.
You have to go back to dimension, uncheck symmetrical and then go back to two lengths.

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-26:
@FlachyJoe FYI

FlachyJoe, 2024-08-26:
@maxwxyz it's an UI issue, I can't help.

maxwxyz, 2024-08-26:
@kadet1090 maybe?

Syres916, 2024-08-26:
@maxwxyz I can't believe this is too difficult but what I need to test is whether Pocket, etc need fixing as well.
it should be as simple as adding `ui->checkBoxMidplane>setChecked(false);` to each case mode where appropriate. It'll be tomorrow before I can do some proper testing and submit a PR.

maxwxyz, 2024-10-08:
Added the blocker label as this is a regression and could break old models as mentioned in
https://github.com/FreeCAD/FreeCAD/issues/16868

maxwxyz, 2024-10-14:
@chennes who could help out here?

maxwxyz, 2024-10-17:
@FlachyJoe do you want to take a look at this?

kadet1090, 2024-10-19:
I'll tackle that, should be done this week

closed 2024-10-25 by chennes

chennes, 2024-11-06:
Noting that @kadet1090 is eligible to claim the bounty from the FPA. To claim, please send an email to fpa@freecad.org referencing this issue, and your preferred method of payment.

PR #17435 (2024-10-25): PD: Fix TwoLengths and Midplane both enabled
This fixes regression introduced with TNP fix. It removes code responsible for centering the extrusion when symmetry was checked, this is exactly how it was done in 0.21. 

No UI changes are necessary since UI was already in a good condition. Migration code is also not necessary since it only aligns toponaming aware code with one that is not.

Fixes: #16072