FreeCAD/FreeCAD #16073: New CAM Simulator doesn't support Touchpad/trackpad inputs

JohnOCFII, 2024-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As discovered in https://github.com/FreeCAD/FreeCAD/issues/15665#issuecomment-2309327041 the new CAM simulator doesn't support the built-in FreeCAD Touchpad settings.  While I can zoom with the existing support, I can't PAN or rotate using a Trackpad/Touchpad on a 2021 MacBook Pro.

### Full version info

```shell
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38507 (Git)
Build type: Release
Branch: main
Hash: baec1037d8ce9c02f6dc67685926fce9dbaf42cf
Python 3.11.8, Qt 5.15.8, Coin 4.0.2, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.8.17
  * fasteners 0.5.27
```
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-08-26:
@shaise @Rexbas FYI

shaise, 2024-08-26:
@maxwxyz , thanks I asked for this issue. Can you assign it to me?

closed 2024-09-02 by sliptonic

JohnOCFII, 2024-09-04:
Works great!  Thanks!