FreeCAD/FreeCAD #16078: Second selected face not higlighted for Links in Std_Parts and Assemblies

Roy-043, 2024-08-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The second selected face is not higlighted for Links in Std_Parts and Assemblies, which is very confusing.

1. Unzip [assy_021_test.zip](https://github.com/user-attachments/files/16752430/assy_021_test.zip) in a single directory.
2. Open assy_021.FCStd. The file contains a Std_Part.
3. Recompute the file.
4. Select the cylindrical face.
5. Hold down Ctrl and add a planar face from the other Link to the selection.
6. Result: the face is selected but not highlighted.

The same test can be done with an Assembly: [assy_test.zip](https://github.com/user-attachments/files/16752432/assy_test.zip).

The issue has been confirmed on the forum (with assy_test.zip):
https://forum.freecad.org/viewtopic.php?t=90083

Last version where the issue does not occur:
OK: Version: 0.22.0dev.37730 (Git)
First version where the issue occurs:
BAD: Version: 0.22.0dev.37841 (Git)

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2025-07-17:
I followed steps and cannot reproduce. @Roy-043 please advice if this is fixed

Roy-043, 2025-07-17:
This has indeed been fixed. Thanks.
```
OS: Ubuntu 22.04.5 LTS (ubuntu:GNOME/ubuntu/wayland)
Architecture: x86_64
Version: 1.1.0dev.42564 (Git)
Build date: 2025/07/14 18:51:42
Build type: Release
Branch: main
Hash: dc2f2374b6ab3e3381653fad125b1b619bb49f7c
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Dutch/Netherlands (nl_NL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Free Turntable/Drag at cursor
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/fusion
Logical DPI/Physical DPI/Pixel Ratio: 96/86.1283/1
Installed mods:
```