FreeCAD/FreeCAD #16081: [Part Design] Malformed Helixes

leoheck, 2024-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I am experiencing malformed addictive Helices. I found some issues reported but I am not sure if they are related so I am creating the new one.

I made a file with examples to test and show what I am experiencing. It is attached here.
[additive_helix_tests.zip](https://github.com/user-attachments/files/16755105/additive_helix_tests.zip)

There is a cylinder with diameter 50 mm (radius 25 mm) and a Sketch of a triangle with the thread profile on ZX axis.

When the thread profile is placed on the external face of a cylinder, the thread is malformed and does not appear. 

[image]

In the past it could be fixed by placing this thread on the `radius - 0.01`. But now, this results on a broken operation with the message `Removing splitter failed`.

[image]

Modifying the Sketch a bit, making the triangle on Z=1 instead of Z=0, (while the keeping the Attachment offset Z=0), results on a malformed operation which kind of works, but it creates the helix upwards, AND a bit downwards.

[image]

The only way to make a valid additive thread is to change the attachment offset to z > 0 (in this example of course, since the cylinder grows from Z=0 to Z=20) and also place the thread profile on the distance <= external wall radius.

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git) AppImage
Build type: Release
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-08-27:
However here, in this second example, where I have 2 stacked cylinders (lets say a bottle with a neck), I could not make it work at all. When the height of the profile is set within the Sketch, the operation does not render anything but works. When the Attachment offset is set and the profile is sketch on Z=0. The operation fails with the `Removing splitter failed.` issue. 

[additive_thread_neck_failed.zip](https://github.com/user-attachments/files/16755271/additive_thread_neck_failed.zip)

leoheck, 2024-08-27:
Ah, there is one new weekly from today. Well, the same issues persist on this one too. 
```
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

maxwxyz, 2024-08-27:
@FlachyJoe another HELLix
maybe related https://github.com/FreeCAD/FreeCAD/issues/15353

FlachyJoe, 2024-08-27:
@leoheck do you know if it worked a day? Have you test with 0.21?

leoheck, 2024-08-27:
@FlachyJoe I ported the fcstd file with a  existing macro to 0.21 and the first file behaves equal with the same issues. However the second file where the Sketch is not close to Z=0 it works fine on 0.21 while on 0.22 it doesn't.

leoheck, 2024-08-27:
Even the old way of creating a thread with a Thread profile and a Part Helix does not work.
[thread_part_helix.zip](https://github.com/user-attachments/files/16767430/thread_part_helix.zip)

[image]

leoheck, 2024-08-27:
In this example I made the thread in a Body, and using the Clone I tried to add a cylinder.
The operation fails if the Cylinder grows up from z=0, but it works if it grows symmetrically. 

[thread_thread_body.zip](https://github.com/user-attachments/files/16767604/thread_thread_body.zip)

[image]

Here the thread and the external face of the cylinder are placed at the same distance (25 mm)

leoheck, 2024-08-27:
Another try.

Creating the Thread as a Body, and then fusing it with the Neck as another Body with the Part Design Boolean.
This does not work either. 

[thread_part_design_workbench_boolean.zip](https://github.com/user-attachments/files/16767753/thread_part_design_workbench_boolean.zip)

[image]

leoheck, 2024-08-28:
I could find a way to work around this issue. 

I could make a Cylinder with the Thread (following the known tricks - overlapping thread profile with the cylinder, and also moving it off the z=0) an then, using this to make a Part Design Boolean fusion with another Body, this way you can place the Thread where you want. 

[thread_workaround.zip](https://github.com/user-attachments/files/16786803/thread_workaround.zip)

alfrix, 2024-09-12:
if i revert 74e5c3fafba7b0307c6b75e57ef648baae32f971
the "broken" model from OP starts working again
calling the autor @FlachyJoe to see if he can help

leoheck, 2024-09-12:
I could fix my threads buy intercepting more material with the base cylinder. on my examples, the thread profile and the cylinder has a bit of overlap (0.1). Maybe this is not always enough.

FlachyJoe, 2024-09-12:
Your broken case is fixed by switching its Refine property to False (it fails to remove the single vertex on the cylinder bottom).
The third malformed is fixed by #16226
Aren't the other cases common boolean failure with coincidence with face or seam edge?

leoheck, 2024-09-12:
> Your broken case is fixed by switching its Refine property to False 

Hm, interesting, I did not think about this since now the default is true. 
That is a good observation.

leoheck, 2024-09-13:
Here is an example where I could fix my issues above, but there is two other examples where one is broken and one is malformed

[additive_helix_tests_2.zip](https://github.com/user-attachments/files/16986952/additive_helix_tests_2.zip)

New problems arrise when the thread is put on a bottle neck.
[image]

FlachyJoe, 2024-09-13:
No problem here
[image]

> OS: Debian GNU/Linux trixie/sid (XFCE/xfce)
> Word size of FreeCAD: 64-bit
> Version: **1.0.0RC1.38702** (Git)
> Build type: Release
> Branch: main
> Python 3.12.6, Qt 6.6.2, Coin 4.0.2, Vtk 9.3.0, **OCC 7.8.2.dev**
> Locale: English/United States (en_US) [ OS: French/France (fr_FR) ]
> Stylesheet/Theme/QtStyle: /Classic/

Please give your system information.

mendberg, 2024-09-27:
1.0.0RC2 still has issues w/ additive helix:

1. sketch a centered circle on XY plane, diameter 60 mm
[image]

2. pad it to a height of 70 
[image]

3. sketch a circle on XZ plane, 30 mm on X, diameter 1 mm
[image]

4. take second sketch and make it an additive helix with any parameters, hit [OK] and **FAIL:** "Removing splitter failed"
[image]

FlachyJoe, 2024-09-27:
The **removing splitter** error come from the refinement, please disable the Refine property.

mendberg, 2024-09-27:
I read about that but did only turn off refinement for the pad. I had to disable refinement globally in Preferences/Part Design/General.

Thanks for the hint.

leoheck, 2024-09-27:
> The **removing splitter** error come from the refinement, please disable the Refine property.

Can this feature defaults to `Refine = False`?

FlachyJoe, 2024-09-27:
@leoheck IMHO the auto-refine preference should be removed and the user should choose when he wants to refine.

Refinement could also emit a warning and not an error it could be considered not to be a critical feature.

leoheck, 2024-09-27:
I see. It was like that in the past. They used to say it was due to the performance. But in this case it is about the features not being able to resolve well. I was a bit of surprised they changed this for this 1.0 version.

PR #17119 (2024-10-21): Fix boolean operations
Fix #17159
Fix #17150
Fix #17085
Fix #16081 
Fix #15825
Fix #15615
can alleviate #11083 ( fixes with nonstandard setting for boolean fuzzy > 1000 - but that's too high to cut out very delicate parts from very large blocks )
Fix #9666
Allows workaround for #17338 

OCCT has issues with boolean operations failing when parts are touching but not intersecting. This includes but is not limited to the PartDesign helix , which is notorious for producing results that do not pass BOP tests or even fail completely.

The suspected problem is too narrow precision. Part precision sho [... cut]
closed 2024-10-21 by yorikvanhavre