FreeCAD/FreeCAD #16088: CAM: Profiling a hole with offset yields inaccurate coordinate values

jkvfc, 2024-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Profiling a hole (radius 3 mm) with FreeCAD's default 5 mm endmill **without**  compensation (highlighted circle digitised)
[image]
[TestHoleWideningWithoutCompensation.zip](https://github.com/user-attachments/files/16759011/TestHoleWideningWithoutCompensation.zip)
yields the coordinate values
[image]
The length of the first point from the origin is sqrt(2.121320^2+2.121320^2)=2.999999. That's perfect. The geometry is OK.
The same **with** compensation
[image]
[TestHoleWideningWithCompensation.zip](https://github.com/user-attachments/files/16759068/TestHoleWideningWithCompensation.zip)
yields the coordinate values
[image]
The length of the first point from the origin is sqrt(0.348087^2+0.348087^2)=0.492269. Should be 0.5 ((6-5)/2). The value 0.348087 should be 0.5*sin(45°)=0.353553. The differences of about 0.008 is absolute unacceptable because a milling machine works in the range of 0.001 mm.
Did I completely miss something?

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38506 (Git)
Build type: Unknown
Branch: main
Hash: ebf02483af225b502fccfa7111499dfde33999fd
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United States (en_US)
```

### Subproject(s) affected?

CAM

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

spanner888, 2024-09-26:
FreeCAD preferences - General - Number of decimals default is 2. This is less than the precision you require.

@jkvfc  What is your setting?

You may need 4 decimals or more to get the precision you require. Can you review and retest and report back?

Also, please do not paste images of text, instead insert as text within triple back-ticks. The images you posted are impossible to read for me, and I only have a very minor color impairment.

jkvfc, 2024-09-26:
@spanner888 I overlooked the fact  that in Preferences - Job Preferences the Default Geometry Tolerance and the Default Curve Accuracy can be set. In order to set these values to 0.0001 in Preferences - General the number of decimals must be set to 4. If these parameters are set in this way all calculated coordinates are correct.
```
(Profile)
(Compensated Tool Path. Diameter: 5.0)
G0 Z5.000000
G0 X0.353496 Y0.353495
G0 Z3.000000
G1 X0.353496 Y0.353495 Z-1.999999
G2 I-0.353495 J-0.353495 K0.000000 X0.020258 Y-0.499507 Z-1.999999
G2 I-0.020257 J0.499507 K0.000000 X-0.020341 Y0.499503 Z-1.999999
G2 I0.020342 J-0.499503 K0.000000 X0.353496 Y0.353495 Z-1.999999
G0 F0.000000 Z5.000000
G0 Z5.000000
```
sqrt(0.353496^2+0.353496^2)=0.49992 or sqrt(0.020258^2+0.499507^2)=0.49992 

Thank you very much for your hint and sorry for the  unreadable coordinate images which is because  I am using dark theme. Do you close this issue or should I do so?

closed 2024-09-27 by jkvfc