FreeCAD/FreeCAD #16092: Part Design: Display Properties disables some toolbar buttons of Part Design workbench

leoheck, 2024-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Going to Display Properties (Ctrl+D)  disables 2 buttons on the toolbar.

This is weird, looks like a bug. 
- Create an additive primitive
- Create a subtractive primitive

[image]

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-27:
Related:  
* #14834
* #10027 (although related to Sketcher)

Syres916, 2024-08-28:
The two commands that become disabled when the Display/Appearance dialog opens is due to the logic that only be active if there's an _ActiveDocument_ and there's no _Dialog_ open when actually makes perfect sense to me, see https://github.com/FreeCAD/FreeCAD/blob/main/src/Mod/PartDesign/Gui/CommandPrimitive.cpp#L234.

The question arises then why should most of the other commands be still active. The user shouldn't be trying (or allowed to try) to create a Revolution when they are changing the display properties on an existing object.

leoheck, 2024-12-03:
read the thread robot... just read the thread.

luzpaz, 2025-05-20:
@FreeCAD/design-working-group anyone want to weigh in on this UX issue ?