FreeCAD/FreeCAD #16094: PartDesign - Thickness TNP

leoheck, 2024-08-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

_Thickness McFly!_ 

When Part Design Pad changes (shrinking, for instance), Thickness never updates.

[image]

[image]

This happens when I have an existing project, that has many features in it.
But does not work on a fresh new example with just these 2 features (pad and thickness).

[thickness_mcfly.zip](https://github.com/user-attachments/files/16768360/thickness_mcfly.zip)

### Full version info

```shell
OS: Ubuntu 24.04.1 LTS (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git) AppImage
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-08-27:
Anyway, changing the size of the Sketch of a Pad (shrinking/increasing the area) should not change the name of the faces when they are Base features, at least, right?

leoheck, 2024-08-27:
This issue is bad since when I go to edit the Thickness, I don't immediately have the preview of the old result, to understand which face was used, see:
[image]

However, if I toggle the Select/Preview I can now see the old resulting object to understand which face was selected.
[image]

And then if I select the right face, it gives me exactly the same `Face14` name.
Then I can make it work by removing the Existing `?Face14`, after.

Because the face name is the same it smells a simple bug that can be fixed, right?
[image]

leoheck, 2024-08-27:
Extra:

Pipe is not being updated either.
[image]

leoheck, 2024-08-27:
Attachment is not updating too.
[image]

bgbsww, 2024-08-28:
The attached file has a broken reference and will not recalculate.  Can we get an example that is not a broken file that shows the issue?

leoheck, 2024-08-28:
I am not allowed to share the project. I copied the problematic Body/Feature to another file, doesn't this work? I am afraid I don't have any valid examples since I fixed those issues in the original file but since I work with my files on dropbox I may  have something to recover. 

What is a `broken reference` or a `broken file`? Because, for me it opens fie. The Thickness is that is not recalculating.

leoheck, 2024-08-28:
@bgbsww does this help?

[broken_thickness.zip](https://github.com/user-attachments/files/16775573/broken_thickness.zip)

I isolated the issue in a different way.

bgbsww, 2024-08-28:
Unfortunately, no.  Once that ? makes it into a file, further tests don't mean anything until it is corrected:

[image]

If I go in and make it Face14 instead of ?Face14, then everything starts operating normally again, including if I change the length of Base  ( Pad25 ) - the thickness does not break again.

If you can recreate the process whereby the original thickness reference turns into a missing reference, which would be on some operation, then there's potentially a bug to address.  But we've been swatting those throughout the last few months, so we'd need to see it happen on a current version; otherwise this just looks like legacy files from earlier broken 0.22 releases.

Do you have a sequence of steps to make the missing element happen?

leoheck, 2024-08-28:
The list of steps is this
- Double click the Sketch under the Pad, there is a constraint with 130, then educe it a bit to 120, close it. 
- And see the Thickness failing...  

I could not reproduce this on a newly created Body with 2 features.

I will guess that since my Body has some more features... it is not updating completely to the end.. and then it seems that ist is not updating the face because of that. But this is just a guess since I am not sure how this TPN fix works.
Other features of the same object are failing to update too (Pipe, and The Attachment mode), maybe due to the same reason.

Should all the features of the body be working for the TPN fix to work or it is enough just the features before it?

I will continue working, if I have a new idea I will add here. I did not have time to properly read everything

maxwxyz, 2024-08-29:
So as I understand it, when attempting to create the same model from scratch (in the newest weekly) this bug does not happen anymore? So only files created with an older dev version have this issue and no files created from 0.21.2.
If so, I'd close this issue.

leoheck, 2024-08-29:
> to create the same model 
No, not the same model. A simplified version of a subset of the same model which has 2 features.
The original module has 30 features or more.

> So only files created with an older dev version 
Yes, older as in 1 or 2 weeks older.

I don't use 0.21. 

If you close this, can I open it again if I keep finding the same issue?
If not, can you keep it open a bit longer?

maxwxyz, 2024-08-30:
Yes you can reopen anytime.

closed 2024-08-30 by maxwxyz

leoheck, 2024-08-30:
Except that I can't, 

[image]