FreeCAD/FreeCAD #16103: PartDesign: Split bodies by surface / sketch

maxwxyz, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Related to trimming surfaces: https://github.com/FreeCAD/FreeCAD/issues/5587

It is currently not possible to split a solid body in PartDesign by a surface or sketch. This is possible in the @realthunder Link branch.
Here is a demo of the link branch implementation:

https://youtu.be/qPoFrgk8S2U?si=uDGPZ87rcSlP_Lyr&t=1320

It is also possible in @tomate44 Curves WB or the Part WB

[image]

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Anything else?

@pierreporte FYI

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-08-28:
Not only surfaces and sketches should be supported, but also datum planes.

maxwxyz, 2024-08-28:
yes, planes and general faces / surfaces.

oficinerobotica, 2024-12-01:
Please, just please bring this feature to main. 
It is perhaps one of the most useful PartDesign features of link branch together with tolerances for dress up operations.

luzpaz, 2025-04-06:
Any volunteers ?