FreeCAD/FreeCAD #16106: PartDesign: No parametric translation feature

maxwxyz, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is no feature to add to a body to parametrically translate the body with respect to a chosen reference (origin, point, plane, axis).
This is helpful to transform the body parametrically depending on an internal or external reference.

Either by providing a **direction and distance**, transforming a selected point to a different point (**point to point**), or by providing **coordinates** (x,y,z) and the axis system.

Similar to the translation there could be a combined feature for either translation, rotation, scale or a combination of them.

[image]

[image]

[image]

[image]

Some of the features are present in the Manipulator WB

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Anything else?

@pierreporte FYI

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-10-25:
I'd say that this functionality should be done by the PartDesign LinearPattern no? When number of occurence = 1

pierreporte, 2024-10-25:
I don't think that translation and pattern are similar enough to be the same feature. It can be the same code but as a user it feels very different. I had the same comment for the equivalent in sketcher.

maxwxyz, 2025-01-20:
IMO fixing this issue by creating a feature object from the transfrom dialog could fix (workaround) this issue: https://github.com/FreeCAD/FreeCAD/issues/12470

maxwxyz, 2026-04-25:
https://github.com/user-attachments/assets/6502741f-6f23-4698-91ab-96d9eae99106

Vibe-coded implementation: https://github.com/maxwxyz/FreeCAD/tree/pd-translate

codekitchen, 2026-11-13:
A parametric translation feature would be frequently useful when designing parts for 3D printing. Often, you want to design the parts in their assembled positions, and then separate them and re-orient them to the best positions and rotations for printing. Sometimes you need to perform cuts as well before positioning.

The current workaround is to clone the bodies, re-position the clones, and then export just those clones. This is tedious, especially when going back and making changes. See for example this workaround being used at 17:30 in this video https://youtu.be/LyEaUOZ7rhM?t=1047

If a translation feature is implemented, this workflow would become simpler and more straightforward.