FreeCAD/FreeCAD #16108: PartDesign: No pattern option to pattern on points / user defined pattern

maxwxyz, 2024-08-28:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

There is no option to pattern one or multiple features on a user defined pattern, e.g. defined points.
Related for the hole feature: https://github.com/FreeCAD/FreeCAD/issues/6884

It should be possible to select the features or the entire body and pattern it at given references, including the anchor point for the patterned object (optional).

[image]

[image]

[image]

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38553 (Git)
Build type: Release
Branch: main
Hash: 59c1ccec3e6b70f56eeee8f94d361019b84bd850
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Curves 0.6.42
  * fasteners 0.5.25
  * freecad.gears 1.2.0
  * Manipulator 1.5.7
```

### Subproject(s) affected?

PartDesign

### Anything else?

@pierreporte FYI

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

pierreporte, 2024-08-28:
Seeing the Catia screenshots, I think it's necessary to clarify the desired implementation. When repeating a feature on a pattern, when the feature is set with a “through all” length, the actual length is computed for each instances, and the length of the reference feature is not simply reused. This is painful un Catia.

NomAnor, 2024-08-30:
> when the feature is set with a “through all” length, the actual length is computed for each instances

This is tracked in #5568. This is not currently the case in FreeCAD (https://github.com/FreeCAD/FreeCAD/pull/12589#issuecomment-2018046014).

closed 2026-10-01 by maxwxyz